jsk-ros-pkg/jsk_robot #707: fetch体内のストレージの空きスペースがない

mmurooka, 2016-11-04:
```
murooka@fetch15:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7.8G  4.0K  7.8G   1% /dev
tmpfs           1.6G  1.3M  1.6G   1% /run
/dev/sda2        94G   94G     0 100% /
none            4.0K     0  4.0K   0% /sys/fs/cgroup
none            5.0M     0  5.0M   0% /run/lock
none            7.8G   76K  7.8G   1% /run/shm
none            100M   12K  100M   1% /run/user
/dev/sda1       511M  3.4M  508M   1% /boot/efi
overflow        1.0M  4.0K 1020K   1% /tmp
```

jsk-members, 2016-11-04:
/var/ros/logかな？のsixasis.logを消してください

jsk-members, 2016-11-04:
タイポ sixaxis.log かな
60Gぐらいあるはず。消したらリブートして下さい

恒久的な直し方は調査中

2016年11月4日金曜日、Kei Okada<k-okada@jsk.t.u-tokyo.ac.jp>さんは書きました:

> /var/ros/logかな？のsixasis.logを消してください
>
> 2016年11月4日金曜日、Masaki Murooka<notifications@github.com
>
>> murooka@fetch15:~$ df -h
>> Filesystem      Size  Used Avail Use% Mounted on
>> udev            7.8G  4.0K  7.8G   1% /dev
>> tmpfs           1.6G  1.3M  1.6G   1% /run
>> /dev/sda2        94G   94G     0 100% /
>> none            4.0K     0  4.0K   0% /sys/fs/cgroup
>> none            5.0M     0  5.0M   0% /run/lock
>> none            7.8G   76K  7.8G   1% /run/shm
>> none            100M   12K  100M   1% /run/user
>> /dev/sda1       511M  3.4M  508M   1% /boot/efi
>> overflow        1.0M  4.0K 1020K   1% /tmp
>>
>>
>
>
> ◉ Kei Okada

mmurooka, 2016-11-04:
治りました！

mmurooka, 2016-11-04:
消すのは /var/log/upstart/sixad.log でした．

@Affonso-Gui もしできたらReadMeに書いてPR送ってみましょう．

closed 2016-11-04 by mmurooka

furushchev, 2016-11-04:
これですが、おそらくsixad.log.1とかが一緒のディレクトリにあるでしょうか？
もし無くて、単体ならlogrotateの設定が必要だと思います。
もし有ったら、logrotateしていて、サイズか指定した時間が経つとlogファイルが圧縮されて、何世代か保管されたのちに勝手に削除されるはずです。
まずは`/etc/logrotate.d/sixad`があると勝手に当りを付けます(無ければ`/etc/logrotate.conf`)が、この辺の設定ファイルを調べて、もし、`size XX MB`みたいにログのファイル上限が指定されているか確認が必要だと思います。
もし`size`が指定されていれば、単純にlogrotateがファイルサイズを監視する周期よりも早くログが貯まってしまうということで、今度は`cron`に登録されている`logrotate`の実行周期を上げてみるのが一番簡単な対処方法だと思います。
(`/etc/crontab` or `/etc/cron.weekly`? かその辺に書いてあると思います。)

mmurooka, 2016-11-04:
教えてくれたファイルの内容は以下のようになっていました．

/etc/logrotate.conf の
```
# keep 4 weeks worth of backlogs
rotate 4
```
を書き換えてみればいいでしょうか．

```
$ cat /etc/logrotate.d/sixad 
/var/log/sixad.log {
  rotate 10
  size 50M
  missingok
  compress
}
```
```
$ cat /etc/cron.daily/logrotate
#!/bin/sh

# Clean non existent log file entries from status file
cd /var/lib/logrotate
test -e status || touch status
head -1 status > status.clean
sed 's/"//g' status | while read logfile date
do
    [ -e "$logfile" ] && echo "\"$logfile\" $date"
done >> status.clean
mv status.clean status

test -x /usr/sbin/logrotate || exit 0
/usr/sbin/logrotate /etc/logrotate.conf
```
```
$ cat /etc/logrotate.conf
# see "man logrotate" for details
# rotate log files weekly
weekly

# use the syslog group by default, since this is the owning group
# of /var/log/syslog.
su root syslog

# keep 4 weeks worth of backlogs
rotate 4

# create new (empty) log files after rotating old ones
create

# uncomment this if you want your log files compressed
#compress

# packages drop log rotation information into this directory
include /etc/logrotate.d

# no packages own wtmp, or btmp -- we'll rotate them here
/var/log/wtmp {
    missingok
    monthly
    create 0664 root utmp
    rotate 1
}

/var/log/btmp {
    missingok
    monthly
    create 0660 root utmp
    rotate 1
}

# system-specific logs may be configured here
```

furushchev, 2016-11-04:
基本的に狭いスコープが優先されるので、この場合

```
/var/log/sixad.log {
  rotate 10
  size 50M
  missingok
  compress
}
```

が優先されそうです。
50MBを上限に10世代分の過去ログを圧縮して残すことになっているようです。
今回消したのは`sixad.log`だけでしょうか？それとも過去ログの圧縮ファイルも消していますか？　また消したことでどれくらい容量が増えましたか？
もし`sixad.log`単体で60GBも容量を食っている場合、まずlogを出すプログラムが何かおかしい気がしますが、
短期的には例えば`/etc/cron.daily/logrotate`を`/etc/cron.hourly`に移動してみるなどで、ストレージがなくなって何もできなくなる状況にはならなくなると思います。

wkentaro, 2016-11-07:
@k-okada 
@jsk-members とは誰ですか？

jsk-members, 2016-11-07:
しりません．gakusei-kyoin とかをアサインした？ @yuki-asano

k-okada, 2016-11-07:
以下か．こういうことになるので、やめたほうがいいと思うけど、誰が登録したのかな？

```
GitHub <noreply@github.com>
2015/04/21

To jsk-members
Hey, we want to verify that you are indeed "jsk-members".  Verifying this
address will let you receive notifications and password resets from
GitHub.  If you wish to continue, please follow the link below:

https://github.com/users/jsk-members/emails/14216696/confirm_verification/898f410b4499382f57303e5431317c0ddd25dde6

If you're not jsk-members or didn't request verification, you can ignore
this email.
```

furushchev, 2016-11-07:
@k-okada 完全に勘ですが、

https://github.com/jsk-ros-pkg/jsk_robot/issues/704

の権限の変更の前後でおかしくなった気がします。

jsk-members, 2016-11-08:
なるほど
@jsk-members をグループから外しました
これでOKかな

k-okada, 2016-11-08:
グループとメンバーと2つあるのか

k-okada, 2016-11-08:
1. そもそもsixadがおかしいのは、https://groups.google.com/a/fetchrobotics.com/forum/?utm_medium=email&utm_source=footer#!topic/fetch-users/vIlJ8VDcdYA で質問した．
2. sixadのrestartをするとエラーは出なくなるんではないか、という気もするので、１０M超えたらlogrotateして（gzかかる）、sixadをリスタートする作戦にしてみた．
3. wifiも切れやすいので以下の２つのファイルをcronで回すようにした．
```
fetch@fetch15:~$ crontab -l
# Edit this file to introduce tasks to be run by cron.

* * * * * sudo /home/fetch/reconnect-wifi.sh 
* * * * * sudo /home/fetch/restart-sixad.sh

fetch@fetch15:~$ cat reconnect-wifi.sh 
#!/bin/bash

#ping -c 1 8.8.8.8 || (nmcli dev disconnect iface wlan0; sleep 10; nmcli dev wifi connect sanshiro password ZnkLM2TS9aB4r)
ping -c 1 8.8.8.8 || (nmcli nm wifi off; sleep 5; nmcli nm wifi on)

fetch@fetch15:~$ cat restart-sixad.sh 
#!/bin/bash

LOG_FILE=/var/log/upstart/sixad.log
CONF_FILE=/etc/logrotate.d/upstart

set -x
ls -hal $LOG_FILE
find $(dirname $LOG_FILE) -name $(basename $LOG_FILE) -size +10M -exec logrotate --force $CONF_FILE  \; -exec service sixad restart \;
```