jsk-ros-pkg/jsk_robot #710: ２台のPR２での時間の同期

h-kamada, 2016-11-08:
2台のPR2を用いたデモを作る上で、2台のPR2の時間が同期されておくことが望ましいと考えています。

pr1012とpr1040は`/etc/chrony/chrony.conf`を見るにそれぞれのbase stationに同期を取りにいくよう設定されています。
同期を取る上でこれを変更し、２つのPR2が同一の時間に対して同期を取る必要があると考えていますが、どのようにするのが良いでしょうか。
ネットから時間を取得して時間を合わせる、といったようなことは可能ではないのでしょうか。

また、pr1012のchrony.confはdefaultのchrony.confから書き換えられていますが、これはどういった経緯があって書き換わったものなのでしょうか。

h-kamada, 2016-11-08:
* pr1040のchrony.conf
````
# This the default chrony.conf file for the Debian chrony package.  It is
# suitable for a system with an intermittent dial-up connection.  After
# editing this file use the command 'invoke-rc.d chrony restart' to make
# your changes take effect.  
# John Hasler <jhasler@debian.org> 3 Dec. 1998

#publish_uncooked_time

commandkey 1
cmdallow 10.68.0.0/24 
manual on
pidfile /var/run/chrony-internal.pid

# See www.pool.ntp.org for an explanation of these servers.  Please
# consider joining the project if possible.  If you can't or don't want to
# use these servers I suggest that you try your ISP's nameservers.  We mark
# the servers 'offline' so that chronyd won't try to connect when the link
# is down.  Scripts in /etc/ppp/ip-up.d and /etc/ppp/ip-down.d use chronyc
# commands to switch it on when the link comes up and off when it goes
# down.  If you have an always-on connection such as cable omit the
# 'offline' directive and chronyd will default to online.

# c2 gets its time from fw1, but step adjust its time to c2's time when 
# chrony is restarted to get in synch ASAP.
# These are the 3 IP's where we might find the basestation
# server basestation-novpn minpoll 3 maxpoll 15 maxdelay .010
server 10.68.255.1 minpoll 3 maxpoll 15 maxdelay .010
server 10.68.0.6   minpoll 3 maxpoll 15 maxdelay .010
#peer c2 minpoll 3 maxpoll 5 maxdelay .0005
initstepslew 0 basestation-novpn 10.68.255.1 10.68.0.6 0.debian.pool.ntp.org 1.debian.pool.ntp.org 2.debian.pool.ntp.org 3.debian.pool.ntp.org 

# Look here for the admin password needed for chronyc.  The initial
# password is generated by a random process at install time.  You may
# change it if you wish.

keyfile /etc/chrony/chrony.keys

# Set runtime command key.  Note that if you change the key (not the
# password) to anything other than 1 you will need to edit
# /etc/ppp/ip-up.d/chrony, /etc/ppp/ip-down.d/chrony, and
# /etc/cron.weekly/chrony as these scripts use it to get the password.

commandkey 1

# I moved the driftfile to /var/lib/chrony to comply with the Debian
# filesystem standard.

driftfile /var/lib/chrony/chrony.drift

# Comment this line out to turn off logging.

log tracking measurements statistics
logdir /var/log/chrony

# Stop bad estimates upsetting machine clock.

maxupdateskew 100.0

# Dump measurements when daemon exits.

dumponexit

# Specify directory for dumping measurements.

dumpdir /var/lib/chrony

# Let computer be a server when it is unsynchronised.

local stratum 10

# Allow computers on the unrouted nets 10 and 192.168 to use the server.

allow 10/8
allow 192.168/16
allow 133.11.216/24
allow 172.16/12

# This directive forces `chronyd' to send a message to syslog if it
# makes a system clock adjustment larger than a threshold value in seconds.

logchange 0.5

# This directive defines an email address to which mail should be sent
# if chronyd applies a correction exceeding a particular threshold to the
# system clock.

# mailonchange root@localhost 0.5

# This directive tells chrony to regulate the real-time clock and tells it
# Where to store related data.  It may not work on some newer motherboards
# that use the HPET real-time clock.  It requires enhanced real-time
# support in the kernel.

rtcfile /var/lib/chrony/chrony.rtc

# If the last line of this file reads 'rtconutc' chrony will assume that
# the CMOS clock is on UTC (GMT).  If it reads '# rtconutc' or is absent
# chrony will assume local time.  The line (if any) was written by the
# chrony postinst based on what it found in /etc/default/rcS.  You may
# change it if necessary. 
rtconutc
````

* pr1012のchrony.conf
````
cmdallow 10.68.0.0/24
makestep 100 -1

server 10.68.255.1      minpoll 0 maxpoll 5 maxdelay .001 maxdelaydevratio 2 minstratum 1 prefer
server 133.11.216.200 minpoll 0 maxpoll 5 maxdelay .001 maxdelaydevratio 2 minstratum 10

initstepslew 0 10.68.0.1

keyfile /etc/chrony/chrony.keys
driftfile /var/lib/chrony/chrony.drift
log tracking measurements statistics
logdir /var/log/chrony
maxupdateskew 10000.0
dumponexit
dumpdir /var/lib/chrony
local stratum 15
allow 10/8
logchange 0.5
rtcfile /var/lib/chrony/chrony.rtc
#rtconutc
````

jsk-members, 2016-11-08:
両方共外部のNTPみにいけば、必要なレベルでは時間はあうとおもうので、まずは、その状態で（今の状態で）、いいと思います．

h-kamada, 2016-11-09:
以前自身のPCをpr1040に時間を合わせるように設定していましたが、pr1012に対してtfを解決しようとした際、自身のPCの方がpr1012より時間が微妙に進んでおり、未来に対して解決しようとしてerrorを吐くことがありました。
再びそのような状況が起これば報告します。

h-kamada, 2016-11-09:
pr1012とpr1040で時間がかなりずれており、wait-for-transformがfailします。
ntpdate コマンドで修正しても良いでしょうか？
[image]

YoheiKakiuchi, 2016-11-09:
chronyのサーバー (pr2のchronyが見ている先)が同じ外部のNTPを見に行って、時計があっていれば
時間がずれないはずではないでしょうか？

現在のズレを直しても、見る先がちゃんとしていないと結局ずれてくるように思います。

h-kamada, 2016-11-10:
prbase1040のchronyの動作確認をしたところ、どこも見ていないようです。
[image]

YoheiKakiuchi, 2016-11-10:
baseでのchronyの見に行く先をprbase1012とprbase1014で同じにすればいいように思います。

h-kamada, 2016-11-10:
prbase1040において
`/etc/init.d/chrony restart`
した上で、全てに対して
```
$ chronyc
> password
> makestep
```
することによって時間の同期がとれました。
(pr1012のみchronyのversionがDEVELOPMENT versionになっており、makestepコマンド使用不能だったので、これもrestart)

[image]

h-kamada, 2016-11-10:
今回のずれはprbase1040のchronydがどことも同期されていなかったのが原因なのかと思っています。
prbase1012とprbase1040はともにデフォルトのchrony.confで、pool.ntp.orgに設定されるランダムなサーバを見ていますが、同一に指定するべきでしょうか？

k-okada, 2016-11-10:
2016年11月10日 21:26 h-kamada <notifications@github.com>:

> pool.ntp.orgに設定されるランダムなサーバを見ています

まずはこれで良いと思います．

--
◉ Kei Okada

closed 2016-11-11 by h-kamada

chiwunau, 2016-12-13:
pr1012の中でchronyをrestartしても

pr2admin@pr1012:~$ chronyc tracking
517 Protocol version mismatch
pr2admin@pr1012:~$ chronyc sources
506 Cannot talk to daemon

になっていて動かないですが、versionが
pr2admin@pr1012:~$ chronyc --version
chronyc (chrony) version DEVELOPMENT (+READLINE +IPV6 -DEBUG)
になっているせいですか
@h-kamada どうやって直しました？

h-kamada, 2016-12-14:
pr1012のchronyのversionは自分がいじる以前からdevelopmentとなってました。結構色々試して昔のversionに戻そうとしましたがその時は出来ませんでした。
ので、chronyをrestartして対応したような気がします。

k-okada, 2016-12-14:
でなおったのかな．
pr2adminでchrony trackinはできる？

chiwunau, 2016-12-15:
PR1012を再起動して
sudo service chrony restart
chronyc trackingをしてもやはり
517 Protocol version mismatchというメセージがでます。

/etc/chrony/chrony.confを見る限りpr1012とpr1040は一緒になっているので
やはりVer をdevelopmentから
applications@pr1040:~$ chronyc --version
chronyc (chrony) version 1.24
に戻さないといけなさそうです。

k-okada, 2016-12-15:
development には誰がどうやってあげたんだろう？

furushchev, 2016-12-15:
これは僕が数年前に上げました。当時`chronyc tracking`をすると`chronyd`がよく強制終了してしまっていたため、chronyの対策済みのバージョンへ上げて使っていました。
バージョンを上げるとchronyc trackingも正常に動いていました。(Protocol version mismatchは出ずに、時刻も同期されていた)
が、今は動いていないようなので、バージョンを戻しておきました。
(`/usr/local/bin`に置いていただけなので、単純に削除しました。)