jsk-ros-pkg/jsk_robot #713: [fetch]fetchでなにか作ってみる

itohdak, 2016-11-09:
- 荷物を受け取って（サインして）、「受け取りました」と返事をして（宛名を読んで）、隣の部屋に行って、渡して、帰ってくる
- task1 隣の部屋に行く（73A1；稲葉）（73A2；岡田）（73A3；山口）（73B1；浅野）、帰ってくる @YutoUchimi
- task2 手を伸ばして、掴む動作を作る。掴んだかどうか判定するプログラムを作る。 @Affonso-Gui 
- task3 （人を見つけて（顔認識を使う？））ものを渡す動作を作る。ものが引っ張られたら手を離す。（関節が動かされたことを感知する？、重力加速度計で向きの変化を感知する？、接触センサで距離の変化を感知？、（力センサで力を感知？）） @708yamaguchi
- task4 宛名を読む、サインをする（印鑑を押す） @itohdak

k-okada, 2016-11-10:
@itohdak @Affonso-Gui @YutoUchimi @708yamaguchi
昨日作ったコードをhttps://gist.github.com/ にドラッグ＆ドロップして、そこへのリンクをこのページに貼り付けてくれると助かります．

また、Kinematics Simulatorで angle-vector が使えない問題は直したので、いつものようにhttps://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_fetch_robot/jsk_fetch.rosinstall に従いupdateしておいてください．

708yamaguchi, 2016-11-10:
@k-okada 
昨日作ったコードです。
https://gist.github.com/708yamaguchi/b86f3625c259044093b2cf0e2b956500

YutoUchimi, 2016-11-10:
@k-okada 
山口君と同様、昨日のゼミで作ったコードです。
https://gist.github.com/YutoUchimi/57c2f71477ca8d01bdf1450500c63e9b

itohdak, 2016-11-10:
@k-okada 
先日のゼミでのコードです。
https://gist.github.com/itohdak/eb4fffc923393f71071d23a254249b24

Kinematics Simulatorの件、ありがとうございました。うまく動作しました。

k-okada, 2016-11-11:
大体出来ていますね．幾つかコメントいれたものを
https://gist.github.com/k-okada/99aa464583780aa731feb5f3d4178c1b
に置きました．参考にして下さい．

Affonso-Gui, 2016-11-11:
@k-okada 
遅くなりました。コードは以下にあります。
https://gist.github.com/Affonso-Gui/f201ed8becb78379316c7dd53cf52cba

掴めたかどうかを確認するため /gripper_controller/gripper_action/result を読み込もうと思っていましたが、うまくできませんでした。
尚、本タスクでは /go_to_room からメッセージを受け取り、 /received にメッセージを渡すつもりでしたが、それも恐らく間違っています。

708yamaguchi, 2016-11-12:
@k-okada 
岡田先生、task4についての質問なのですが、家で顔認識のプログラムを書こうとして、
`rosrun uvc_camera uvc_camera_node`
でパソコンのカメラ画像のトピック/image_rawを配信して、
`roslaunch opencv_apps face_detection.launch image:=image_raw`
で顔認識をしました。
さらにこれが/face_detection/facesというトピックを配信した時に、
`(ros::subscribe "/face_detection/faces" opencv_apps::FaceArrayStamped #'find-people)`
などとしてサブスクライバを立ち上げ、find-peopleというコールバック関数の中で顔の位置や大きさの情報を得ようと思っているのですが、どのようにプログラムを書けばそのような情報が得られるのでしょうか。

k-okada, 2016-11-13:
@Affonso-Gui 
jsk_robot をupdateして、（git pull origin master して）https://github.com/jsk-ros-pkg/jsk_robot/pull/723が含まれるようにして、
```
(send *ri* :speak "hello" :wait t)
```
するとよいです．

つかめたかどうかを調べるには、send *ri* :start-grasp の返り値をみてみましょう
https://github.com/k-okada/jsk_robot/blob/5a944dc2de4f109cc85273368da0cbac37a21325/jsk_fetch_robot/fetcheus/fetch-interface.l#L74
https://github.com/k-okada/jsk_robot/blob/5a944dc2de4f109cc85273368da0cbac37a21325/jsk_fetch_robot/fetcheus/fetch-interface.l#L88

@708yamaguchi 
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_tutorials/src/face-detector-mono.l
を参考にしてみてください．

Affonso-Gui, 2016-11-16:
Updated version of pick_up.l
https://gist.github.com/Affonso-Gui/f763a02ea33b7e38295b58f360b30825

YutoUchimi, 2016-11-16:
Updated go-to-room.l
https://gist.github.com/YutoUchimi/6dc42541fadbaccbc9892777133144b6

k-okada, 2016-11-17:
大体良く出来ているとおもいます．

>  (send *ri* :angle-vector (send *fetch* :angle-vector #f(9.14772 75.5041
86.975 -11.4777 90.0 0.038312 90.0 -0.038098 0.407858 0.540208)) 6000)

はどうやって出てきた数値化かな．後で直しやすいようにIkつかうか、send *fetch* :rarm :elbow-p :joint-angle
xxx としたほうがいいですね．

>   (send msg :data (format nil "ok"))

は (send msg :data "ok")でいいですね．

msg_arrived は帯域変数になっている letにいれたらいいですね．

A1,2,3かB1,2のどちらかは、
  (send *ri* :move-to (make-coords :pos #f(-1250 6300 0)
   :rpy (float-vector -pi 0 0))

> ;;wait-interpolationはmove-toに対しても有効か?? -> 要検証

:move-to すると、ゴールに到達するまで帰ってこない．かえってくるようにするには(:move-to ... :no-wait t) とする．
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L1198

特に何もしないなら、
(do-until-key
 (ros::spin-once)
)
は
(ros::spin)
のほうがいいかな．深い理由はないけど、eulispのコードが回っているのがちょっと怖い、というぐらい．