jsk-ros-pkg/jsk_robot #718: [jsk_pepper_startup] jsk_pepper_startup.launch のエラー

kochigami, 2016-11-10:
naoqi_dashboardでおそらく起こるようになった
```
;; extending gcstack 0x6960120[32738] --> 0x76e0150[65476] top=7fe1
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_console/message_proxy_model.py", line 118, in handle_highlight_filters_changed
    self.dataChanged.emit(self.index(0, 0), self.index(self.rowCount() - 1, self.columnCount() - 1), [])
TypeError: QAbstractItemModel.dataChanged[QModelIndex, QModelIndex] signal has 2 argument(s) but 3 provided
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_console/message_proxy_model.py", line 118, in handle_highlight_filters_changed
    self.dataChanged.emit(self.index(0, 0), self.index(self.rowCount() - 1, self.columnCount() - 1), [])
TypeError: QAbstractItemModel.dataChanged[QModelIndex, QModelIndex] signal has 2 argument(s) but 3 provided
```
dashboardの変化としては電池の画面が色つかなくなった
エラー文もそのように見える

これとは関係ないかもしれないが、
突然 jsk_pepper_startup.launchが落ちる問題が再発している。
しばらくの間はsshもできなくなるが、待っている or 再起動するとつながるようになる
エラー文にはmoduleがdestroyedと書かれている

（ルータを変えたからか・・？）

k-okada, 2016-11-11:
何が変わっただろう？
https://github.com/ros-naoqi/naoqi_dashboard/commit/537896979a2c52f51e8d9eb5aea96d277e179f6c
？

kochigami, 2016-11-11:
このエラーも出てきたので、上のエラーにも関連して何かに値が入らなくなったのだと思います。
私が最近```sudo apt-get update```した時起きたので、自分の問題かもしれませんが、historyを調べてみます。

```
[ERROR] [WallTime: 1478826019.000024] bad callback: <bound method ConsoleDashWidget._message_cb of <rqt_robot_dashboard.console_dash_widget.ConsoleDashWidget object at 0x7f10f808f218>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
    cb(msg)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_robot_dashboard/console_dash_widget.py", line 120, in _message_cb
    if not self._console._paused:
AttributeError: 'NoneType' object has no attribute '_paused'
```

> 突然 jsk_pepper_startup.launchが落ちる問題が再発している。

これは上とは違う原因だと思います。
同時に上げていた画像処理のプログラムを落としたら、安定しました。
ペッパー体内PCでデータを欲しがるプロセスが多くなって、module destroyed になったのかなと思いました。
ただ、その時destroyed moduleになっていたのはALMotionだった気がするので、確認が必要そうです。

kochigami, 2016-11-11:
最初の問題について
```
;; extending gcstack 0x6960120[32738] --> 0x76e0150[65476] top=7fe1
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_console/message_proxy_model.py", line 118, in handle_highlight_filters_changed
    self.dataChanged.emit(self.index(0, 0), self.index(self.rowCount() - 1, self.columnCount() - 1), [])
TypeError: QAbstractItemModel.dataChanged[QModelIndex, QModelIndex] signal has 2 argument(s) but 3 provided
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_console/message_proxy_model.py", line 118, in handle_highlight_filters_changed
    self.dataChanged.emit(self.index(0, 0), self.index(self.rowCount() - 1, self.columnCount() - 1), [])
TypeError: QAbstractItemModel.dataChanged[QModelIndex, QModelIndex] signal has 2 argument(s) but 3 provided
```
http://stackoverflow.com/questions/22560296/pyqt-list-view-not-responding-to-datachanged-signal
にあるように、PyQtのバージョンが私の環境で古いため。PyQt5がないといけない。

だから
> dashboardの変化としては電池の画面が色つかなくなった
エラー文もそのように見える

は違う

PyQtのバージョンは以下で確認

```
 from PyQt4 import QtCore
>>> print QtCore.PYQT_VERSION_STR
4.10.4
>>> print QtCore.QT_VERSION_STR
4.8.6
>>> from PyQt5 import QtCore
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named PyQt5
```
```
python -V
Python 2.7.6
```
そのようになったのは、
 rqt_robot_dashboard/package.xml　の中でpython_qt_bindingのバージョンが上がったため
https://github.com/ros-visualization/rqt_robot_plugins/commit/d9c421f2bedeed829f2ccd2549951ab7b394dc4c

python_qt_binding は、Qt5を使うようになっている
https://github.com/ros-visualization/python_qt_binding/commit/53759f1295809c0b58a0981e2e26774d89de1fc4

kochigami, 2016-11-11:
以下のように正しいインストールのされ方なので、自分の環境が悪そうです。
すみません・・・

```
ros-indigo-rqt-robot-dashboard:
  インストールされているバージョン: 0.5.5-0trusty-20161102-172811-0700
  候補:               0.5.5-0trusty-20161102-172811-0700
  バージョンテーブル:
 *** 0.5.5-0trusty-20161102-172811-0700 0
        500 http://packages.ros.org/ros-shadow-fixed/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status
```

```
ros-indigo-python-qt-binding:
  インストールされているバージョン: 0.2.19-0trusty-20160627-183533-0700
  候補:               0.2.19-0trusty-20160627-183533-0700
  バージョンテーブル:
 *** 0.2.19-0trusty-20160627-183533-0700 0
        500 http://packages.ros.org/ros-shadow-fixed/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status
```

kochigami, 2016-11-11:
python_qt_bindingのissue（３５番）を見ると、
indigoとjadeではQt4のみ対応
kineticsではQt5に対応
修論が終わったら環境をアップデートして、エラーが解決するか調べる

kochigami, 2016-11-12:
上記とは関係なく起こるが、３分くらい休ませると直る（自分のプログラムのせいだと思う）
```

[ERROR] [WallTime: 1478941399.110200] Exception in your execute callback: 	ALMotion::angleInterpolation
	Session closed
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/actionlib/simple_action_server.py", line 289, in executeLoop
    self.execute_callback(goal)
  File "/home/kochigami/catkin_ws/src/naoqi_bridge/naoqi_pose/nodes/pose_controller.py", line 261, in executeJointTrajectoryAction
    task_id = self.motionProxy.post.angleInterpolation(names, angles, times, (goal.relative==0))
  File "/home/kochigami/naoqi/pynaoqi-python2.7-2.4.3.28-linux64/naoqi.py", line 184, in __call__
    return self.__wrapped__.method_missing(self.__method__, *args, **kwargs)
  File "/home/kochigami/naoqi/pynaoqi-python2.7-2.4.3.28-linux64/naoqi.py", line 211, in method_missing
    raise e
RuntimeError: 	ALMotion::angleInterpolation
	Session closed
```

kochigami, 2016-11-13:
ネットワークの問題か、多発する
待っても駄目、再起動しないといけない（８分かかる）
どこでも起こりやすいか調べた方が良い・・・

kochigami, 2016-11-13:
> naoqi_dashboardのindigoで起きるエラー：indigoでもうごくものがある

naoqi_dashboardが動くものは単純にバージョンが古い
だから上の推測（Qt5を使うようになった変更）が正しく、将来はrosのバージョンを上げる

エラーが出ないパソコンは確かに落ちる頻度が減ったが落ちなかったわけではない
原因ではなさそう

```
sudo apt-cache policy ros-indigo-naoqi-dashboard 
ros-indigo-naoqi-dashboard:
  インストールされているバージョン: 0.1.4-0trusty-20160628-190740-0700
  候補:               0.1.4-0trusty-20160628-190740-0700
  バージョンテーブル:
 *** 0.1.4-0trusty-20160628-190740-0700 0
        500 http://packages.ros.org/ros/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status

sudo apt-cache policy ros-indigo-rqt-robot-dashboard 
ros-indigo-rqt-robot-dashboard:
  インストールされているバージョン: 0.4.3-0trusty-20160628-185341-0700
  候補:               0.4.3-0trusty-20160628-185341-0700
  バージョンテーブル:
 *** 0.4.3-0trusty-20160628-185341-0700 0
        500 http://packages.ros.org/ros/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status

sudo apt-cache policy ros-indigo-python-qt-binding
ros-indigo-python-qt-binding:
  インストールされているバージョン: 0.2.19-0trusty-20160627-183533-0700
  候補:               0.2.19-0trusty-20160627-183533-0700
  バージョンテーブル:
 *** 0.2.19-0trusty-20160627-183533-0700 0
        500 http://packages.ros.org/ros/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status
```

kochigami, 2016-11-18:
```
File "/opt/ros/indigo/lib/diagnostic_common_diagnostics/tf_monitor.py", line 185, in <module>
    diag_updater.update()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/diagnostic_updater/_diagnostic_updater.py", line 245, in update
    self._check_diagnostic_period()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/diagnostic_updater/_diagnostic_updater.py", line 320, in _check_diagnostic_period
    self.period = rospy.get_param("~diagnostic_period", 1)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 462, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 121, in __getitem__
    code, msg, value = self.target.getParam(rospy.names.get_caller_id(), resolved_key)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1301, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1448, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 975, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 835, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 797, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 778, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
socket.error: [Errno 111] Connection refused
```

kochigami, 2016-11-20:
デモの時発生（やめてくれ・・ :'( ） naoqi_dashboardエラー無し
エンコードの値から、自分で進んだ距離を計算しながら進んでいる時発生
たくさんデータを取ってくる時に発生？

```
[pepper_robot/pose/pose_manager-3] process has finished cleanly
log file: /home/kochigami/.ros/log/5327c4f0-aeca-11e6-97d3-d05099313e9f/pepper_robot-pose-pose_manager-3*.log
[W] 19042 qimessaging.transportsocket: connect: Operation canceled
[ERROR] [WallTime: 1479609487.374704] Could not create Proxy to "ALMotion". 
Exception message:
	ALBroker::createBroker
	Cannot connect to tcp://192.168.2.104:9559
[INFO] [WallTime: 1479609487.375547] Stopping pose_controller
[INFO] [WallTime: 1479609487.375660] pose_controller stopped
[pepper_robot/pose/pose_controller-2] process has died [pid 18725, exit code 1, cmd /home/kochigami/catkin_ws/src/naoqi_bridge/naoqi_pose/nodes/pose_controller.py --pip=192.168.2.104 --pport=9559 __name:=pose_controller __log:=/home/kochigami/.ros/log/5327c4f0-aeca-11e6-97d3-d05099313e9f/pepper_robot-pose-pose_controller-2.log].
log file: /home/kochigami/.ros/log/5327c4f0-aeca-11e6-97d3-d05099313e9f/pepper_robot-pose-pose_controller-2*.log
[WARN] [WallTime: 1479609490.871781] unable to communicate with ROS Master, registrations are now out of sync
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/diagnostic_common_diagnostics/tf_monitor.py", line 185, in <module>
    diag_updater.update()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/diagnostic_updater/_diagnostic_updater.py", line 245, in update
    self._check_diagnostic_period()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/diagnostic_updater/_diagnostic_updater.py", line 320, in _check_diagnostic_period
    self.period = rospy.get_param("~diagnostic_period", 1)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 462, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 121, in __getitem__
    code, msg, value = self.target.getParam(rospy.names.get_caller_id(), resolved_key)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1301, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1448, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 975, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 835, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 797, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 778, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
socket.error: [Errno 111] Connection refused
[tf_monitor-24] process has died [pid 18891, exit code 1, cmd /opt/ros/indigo/lib/diagnostic_common_diagnostics/tf_monitor.py __name:=tf_monitor __log:=/home/kochigami/.ros/log/5327c4f0-aeca-11e6-97d3-d05099313e9f/tf_monitor-24.log].
log file: /home/kochigami/.ros/log/5327c4f0-aeca-11e6-97d3-d05099313e9f/tf_monitor-24*.log
[ERROR] [1479609502.929702769]: unknown ros::param::get, key=/robot/type
[ERROR] [1479609504.929845469]: unknown ros::param::get, key=/robot/type
^C[speaking_program_is_started_or_terminated-27] killing on exit
[joy_client-26] killing on exit
[ERROR] [1479609533.089369462]: unknown ros::param::get, key=/robot/type
[naoqi_dashboard_aggregator-25] killing on exit
[naoqi_dashboard-23] killing on exit
[image_saver-22] killing on exit
[pepper_twitter-20] killing on exit
[teleop_twist_joy-19] killing on exit
[joy_node-18] killing on exit
[pepper_robot/camera/depth_registered_metric-17] killing on exit
[pepper_robot/camera/depth_registered_hw_metric_rect-16] killing on exit
PluginHandler.save_settings() plugin "naoqi_dashboard/NAOqiDashboard#0" raised an exception:
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 182, in save_settings
    self._save_settings(plugin_settings, instance_settings)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 109, in _save_settings
    self.emit_save_settings_completed()
[... 4 lines cut ...]
  File "/opt/ros/indigo/lib/python2.7/dist-packages/qt_gui/plugin_manager.py", line 433, in _close_application_shutdown_plugins
    self._shutdown_plugin(info['instance_id'], self._close_application_shutdown_callback)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/qt_gui/plugin_manager.py", line 321, in _shutdown_plugin
    handler.close_signal.disconnect(self.unload_plugin)
TypeError: disconnect() failed between 'close_signal' and 'unload_plugin'

^C[pepper_robot/camera/points_xyzrgb_hw_registered-15] killing on exit
[pepper_robot/camera/depth_registered_rectify_depth-14] killing on exit
[pepper_robot/camera/depth_registered_sw_metric_rect-13] killing on exit
[pepper_robot/camera/points_xyzrgb_sw_registered-12] killing on exit
[pepper_robot/camera/register_depth_front-11] killing on exit
[pepper_robot/camera/depth_metric-10] killing on exit
[pepper_robot/camera/depth_metric_rect-9] killing on exit
[... cut ...] g on exit
[ERROR] [1479609535.085731973]: unknown ros::param::get, key=/robot/type
[pepper_robot/camera/ir_rectify_ir-7] killing on exit
[pepper_robot/camera/bottom_rectify_color-6] killing on exit
[pepper_robot/camera/front_rectify_color-5] killing on exit
[pepper_robot/camera/camera_nodelet_manager-4] killing on exit
[pepper_robot/naoqi_driver-1] killing on exit
[naoqi_dashboard-23] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
```

kochigami, 2018-11-16:
・そもそもjsk_pepper_startup.launchが立ち上がらない・よく落ちる（ネットワーク問題？）：
ネットワークの問題のような気がしていて，無線ではなく有線接続をするようになってから，起きていないと思う．

・デモの途中でjsk_pepper_startup.launchが落ちる（多分自分のプログラムのせい・ALMotionが壊れる）：
これは，自分のプログラムが悪いと思う．たくさん角度列を送るとき（人の顔を見るなど），起きる．

・naoqi_dashboardで起きるエラー：
これは， #980 でも書いていて，https://github.com/ros-naoqi/naoqi_dashboard/pull/3　
で直した．

長い間放置していてすみませんでした．
また今振り返って読むと，使っているROSとペッパーのバージョンが分かりにくく，issueのタイトルも良くなかったなと反省しています．

closed 2018-11-16 by kochigami