jsk-ros-pkg/jsk_robot #722: [baxtereus] angle-vector-motion-plan move both arm

knorth55, 2016-11-11:
i execute code below and both arms move.
it seems that move-arm 

```
$ (send *ri* :angle-vector-motion-plan (send *baxter* :reset-pose) :move-arm :larm :total-time 3000 :start-offset-time 0.01 :use-torso nil :clear-velocities t)
```

knorth55, 2016-11-11:
Branches to reproduce:  (this is for baxter without electric gripper)
https://github.com/knorth55/baxter_simulator/tree/gripper-sim
https://github.com/knorth55/baxter_common/tree/if-gripper
https://github.com/knorth55/moveit_robots/tree/no-gripper
https://github.com/knorth55/jsk_robot/tree/gripper-moveit-init
```
roslaunch baxter_gazebo baxter_world.launch left_electric_gripper:=false right_electric_gripper:=false
roslaunch baxter_moveit_config demo_baxter.launch
rosrun jsk_2015_05_baxter_apc initialize_baxter.py
```

```
roscd baxtereus
roseus baxter-interface.l
```

knorth55, 2016-11-11:
this does not happen with https://github.com/start-jsk/jsk_apc/pull/1939

Branches to reproduce:
https://github.com/knorth55/jsk_robot/tree/gripper-moveit-init
https://github.com/knorth55/jsk_apc/tree/baxter-moveit-config

```
roslaunch jsk_2016_01_baxter_apc baxter_stow_sim.launch
roslaunch jsk_2016_01_baxter_apc baxter_moveit.launch
rosrun jsk_2015_05_baxter_apc initialize_baxter.py
```

```
roscd jsk_2016_01_baxter_apc/euslisp
roseus jsk_2016_01_baxter_apc/baxter-interface.l
```

k-okada, 2016-11-13:
please check `rostopic echo *_controller/follow_joint_trajectory*/goal` or something similar.

knorth55, 2016-11-14:
- command
```
6.irteusgl$ (send *ri* :angle-vector-motion-plan (send *baxter* :reset-pose) :move-arm :larm :total-time 3000 :start-offset-time 0.01 :use-torso nil :clear-velocities t)
[ INFO] [1479123938.858466417, 307.443000000]: ;; Planned Trajectory Total Time   1.628 [sec]
[ INFO] [1479123938.858710639, 307.443000000]: ;; Scaled Trajectory Total Time   3.000 [sec]
[ INFO] [1479123938.858733194, 307.443000000]: ;; generated 12 points for 3 sec using [left_arm] group
(nil nil nil)

```

- left arm joint trajectory 
```
$ rostopic echo /robot/limb/left/follow_joint_trajectory/goal
WARNING: no messages received and simulated time is active.
Is /clock being published?
header:
  seq: 1
  stamp:
    secs: 308
    nsecs: 447000000
  frame_id: ''
goal_id:
  stamp:
    secs: 0
    nsecs:         0
  id: 307447000000_/default_robot_interface_1479123764637664343_19748_/robot/limb/left/follow_joint_trajectory_4
goal:
  trajectory:
    header:
      seq: 1
      stamp:
        secs: 308
        nsecs: 447000000
      frame_id: ''
    joint_names: ['left_s0', 'left_s1', 'left_e0', 'left_e1', 'left_w0', 'left_w1', 'left_w2']
    points:
      -
        positions: [-0.08726389405858637, -0.9941809032851934, -1.1873836337933916, 1.9374545738682194, 0.6642067615254676, 1.030572065553116, -0.00015601789895125506]
        velocities: []
        accelerations: []
        effort: []
        time_from_start:
          secs: 0
          nsecs:   9999999
      -
        positions: [-0.04766478551739686, -0.9434513959912878, -1.1428392567706478, 1.8564685261082925, 0.5720400101373562, 1.0638598788641023, -0.00012675355988348773]
        velocities: []
        accelerations: []
        effort: []
        time_from_start:
          secs: 0
          nsecs: 577673344
      -
        positions: [-0.0080656769762074, -0.8927218886973818, -1.0982948797479049, 1.7754824783483665, 0.4798732587492447, 1.0971476921750876, -9.748922081572039e-05]
        velocities: []
        accelerations: []
        effort: []
        time_from_start:
          secs: 0
          nsecs: 813387904
      -
        positions: [0.03153343156498206, -0.8419923814034762, -1.053750502725161, 1.6944964305884405, 0.38770650736113343, 1.1304355054860737, -6.822488174795311e-05]
        velocities: []
        accelerations: []
        effort: []
        time_from_start:
          secs: 1
          nsecs:  12419919
      -
        positions: [0.07113254010617154, -0.7912628741095706, -1.009206125702418, 1.6135103828285144, 0.29553975597302196, 1.16372331879706, -3.89605426801858e-05]
[... 88 lines cut ...]
    header:
      seq: 0
      stamp:
        secs: 307
        nsecs: 446000000
      frame_id: ''
    joint_names: ['right_s0', 'right_s1', 'right_e0', 'right_e1', 'right_w0', 'right_w1', 'right_w2']
    points:
      -
        positions: [-0.34906585039886573, -0.4363323129985821, 0.6981317007977315,1.0471975511965974, 0.34906585039886573, 1.396263401595463, 0.0]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start:
          secs: 3
          nsecs:         0
  path_tolerance: []
  goal_tolerance: []
  goal_time_tolerance:
    secs: 0
    nsecs:         0
---
```

it seems that right arm moves without moveit

knorth55, 2016-11-19:
i found out that these lines cause this motion. 
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus_moveit/euslisp/robot-moveit.l#L492-L502

@k-okada 
move-armを指定しても、move-armではないほうの腕を送られてくるangle-vectorの通りに動かすようになっているのですがこの仕様に準拠したほうがいいですか？
個人的にはmove-armを指定するのならその腕だけ動いてmove-armではないほうの腕は動かないほうがわかりやすい気がするのですがどうでしょう？（その場合は該当行をコメントアウトするだけで対処できることを確認しました）

k-okada, 2016-11-20:
> _moveit/euslisp/robot-moveit.l#L492-L502

首みたいに（普通は）moveitのなんかでは動かさない関節があって、それを:angle-vectorから動かす方法がなくなるので、あったほうがいいのと、
:angle-vector で片手だけ動かすのは、`:angle-vector av tm :larm-controller`
という風に書くことになっているので、:angle-vector-motion-planでも片手だけ動かしたらいい場合はそうやってoptio
n/keywordで対応するのがいいとおもいます．
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/0.3.7/
pr2eus/robot-interface.l#L346

knorth55, 2016-11-21:
if you want to move one arm, you should specify controller
```
(send *ri* :angle-vector-motion-plan av tm :rarm-controller :move-arm :rarm)
```

closed 2016-11-21 by knorth55