jsk-ros-pkg/jsk_robot #739: [baxtereus] Infinite loop in angle-vector-sequence?

pazeshun, 2016-12-05:
When we use `:angle-vector-sequence` with `:use-send-angle-vector t` with move-it, `:angle-vector-motion-plan` called by `angle-vector-sequence` calls `:joint-trajectory-to-angle-vector-list`. And it calls `angle-vector-sequence`.
Is this infinite loop? I don't know what happens.

knorth55, 2016-12-05:
`:joint-trajectory-to-angle-vector-list` is in `robot-interface.l` and it calls `:angle-vector-sequence` in `robot-interface.l`?

pazeshun, 2016-12-05:
`:angle-vector-sequence` is overridden in baxter-interface.l and that is called, I think.
I did easy test like below.
```
(defclass a
  :super propertied-object
  :slots ()
  )

(defmethod a
  (:init () (print "a initialized"))
  (:parent () (print "Not overridden"))
  (:use-parent () (send self :parent))
  )

(defclass b
  :super a
  :slots ()
  )

(defmethod b
  (:init () (send-super :init) (print "b initialized"))
  (:parent () (print "overridden"))
  )
```
```
1.irteusgl$ setq *b* (instance b :init)
"a initialized"
"b initialized"
#<b #X639a9a8>
2.irteusgl$ send *b* :use-parent
"overridden"
"overridden"
3.irteusgl$ setq *a* (instance a :init)
"a initialized"
#<a #X63cd600>
4.irteusgl$ send *a* :use-parent
"Not overridden"
"Not overridden"
```

knorth55, 2016-12-05:
currently, `tms` is not a number, but a list.
therefore, `angle-vector-sequence` is executed as `angle-vector-sequence-raw`
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L999

however, `:angle-vector-sequence` with motion plan does not work properly now, we can remove the method in case.

pazeshun, 2016-12-05:
Oh, I understood that you means https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_baxter_robot/baxtereus/baxter-interface.l#L287-L291
I overlooked this. Thanks.

knorth55, 2016-12-05:
I move issue that `angle-vector-sequence` with `moveit` do not work properly to #740

closed 2016-12-05 by knorth55