jsk-ros-pkg/jsk_robot #745: [peppereus] move-end-posで失敗する

kochigami, 2016-12-12:
``` send *pepper* :reset-pose```
``` send *pepper* :larm :move-end-pos (float-vector 2 0 0 )``` 
(``` send *pepper* :larm :move-end-pos (float-vector 1 0 0) ```はうまくいくがそれより値を大きくすると失敗する
ただ、動いているようだが、IRT viewerで変化が分からない・・)
=>

```
send *pepper* :larm :move-end-pos (float-vector 2 0 0)
;; inverse-kinematics failed.
;; dif-pos : #f(1.75393 0.407376 -0.483591)/(1.86443/1)
;; dif-rot : #f(-1.934361e-06 -9.338498e-05 -7.193794e-05)/(0.000118/0.017453)
;;  coords : #<coordinates #X6fbff18  0.0 0.0 0.0 / 0.0 0.0 0.0>
;;  angles : (2.0 -2.0 -5.0 76.0112 7.47352 -62.6203 -10.6767 -45.8001 85.0 -10.0 70.0 20.0 40.0 0.0 0.0)
;;    args : ((#<coordinates #Xde4da60  91.672 195.749 -251.729 / 0.072 1.004 -0.227>) :move-target #<cascaded-coords #X6edda30 :larm-end-coords  91.226 195.428 -249.948 / 0.072 1.004 -0.227> :link-list (#<bodyset-link #X62a3c30 LShoulder  -57.0 149.74 86.82 / 0.0 1.323 0.0> #<bodyset-link #X62a3e28 LBicep  -57.0 149.74 86.82 / 0.487 1.292 0.47> #<bodyset-link #X61d0bc0 LElbow  -13.271 187.929 -85.482 / 0.314 1.146 -0.782> #<bodyset-link #X61c9b30 LForeArm  -13.271 187.929 -85.482 / 0.072 1.004 -0.995> #<bodyset-link #X619cf58 l_wrist  67.111 193.697 -211.994 / 0.072 1.004 -1.798>) :move-target #<cascaded-coords #X6edda30 :larm-end-coords  91.226 195.428 -249.948 / 0.072 1.004 -0.227> :collision-avoidance-link-pair nil :link-list (#<bodyset-link #X62a3c30 LShoulder  -57.0 149.74 86.82 / 0.0 1.323 0.0> #<bodyset-link #X62a3e28 LBicep  -57.0 149.74 86.82 / 0.487 1.292 0.47> #<bodyset-link #X61d0bc0 LElbow  -13.271 187.929 -85.482 / 0.314 1.146 -0.782> #<bodyset-link #X61c9b30 LForeArm  -13.271 187.929 -85.482 / 0.072 1.004 -0.995> #<bodyset-link #X619cf58 l_wrist  67.111 193.697 -211.994 / 0.072 1.004 -1.798>))
;; command : (let ((r (instance pepper-robot :init))) (progn (send r :newcoords (make-coords :4x4 #2f((1.0 0.0 0.0 0.0) (0.0 1.0 0.0 0.0) (0.0 0.0 1.0 0.0) (0.0 0.0 0.0 1.0)))) (mapc #'(lambda (j a) (send* j :joint-angle a nil)) (list (send r :kneepitch) (send r :hiproll) (send r :hippitch) (send r :lshoulderpitch) (send r :lshoulderroll) (send r :lelbowyaw) (send r :lelbowroll) (send r :lwristyaw) (send r :rshoulderpitch) (send r :rshoulderroll) (send r :relbowyaw) (send r :relbowroll) (send r :rwristyaw) (send r :headyaw) (send r :headpitch)) '(2.0 -2.0 -5.0 76.0112 7.47352 -62.6203 -10.6767 -45.8001 85.0 -10.0 70.0 20.0 40.0 0.0 0.0)) (objects (list r))) (send* r :inverse-kinematics (list (list (make-coords :pos #f(91.6718 195.749 -251.729) :rot #2f((0.535977 -0.258891 0.803557) (0.038417 0.958313 0.283126) (-0.843358 -0.120879 0.523579)))) :dump-command nil :debug-view t :move-target (let* ((p (send r :link "l_wrist")) (mt (make-cascoords :coords (send (send p :copy-worldcoords) :transform (make-coords :4x4 #2f((1.0 -1.079439e-16 4.906539e-17 45.0) (4.906539e-17 -3.330669e-16 -1.0 2.842171e-14) (1.079439e-16 1.0 -3.330669e-16 2.842171e-14) (0.0 0.0 0.0 1.0))))))) (send p :assoc mt) mt) :link-list (list (send r :link "LShoulder") (send r :link "LBicep") (send r :link "LElbow") (send r :link "LForeArm") (send r :link "l_wrist")) :move-target (let* ((p (send r :link "l_wrist")) (mt (make-cascoords :coords (send (send p :copy-worldcoords) :transform (make-coords :4x4 #2f((1.0 -1.079439e-16 4.906539e-17 45.0) (4.906539e-17 -3.330669e-16 -1.0 2.842171e-14) (1.079439e-16 1.0 -3.330669e-16 2.842171e-14) (0.0 0.0 0.0 1.0))))))) (send p :assoc mt) mt) :collision-avoidance-link-pair nil :link-list (list (send r :link "LShoulder") (send r :link "LBicep") (send r :link "LElbow") (send r :link "LForeArm") (send r :link "l_wrist")))))
;; dump debug command to /tmp/irtmodel-ik-5211/pepper-robot-2016-12-12-15-40-09-22-52-07-failure.l
;; (progn (load "/tmp/irtmodel-ik-5211/pepper-robot-2016-12-12-15-40-09-22-52-07-failure.l")(ik-setup)(ik-check))
nil
```

```
 (progn (load "/tmp/irtmodel-ik-5211/pepper-robot-2016-12-12-15-40-09-22-52-07-failure.l")(ik-setup)(ik-check))
=>
union-link-list: (LShoulder LBicep LElbow LForeArm l_wrist)
```

k-okada, 2016-12-12:
try
```
（send *pepper* :larm :move-end-pos #f(10 0 0)  :world :rotation-axis nil)
```

c.f. https://github.com/euslisp/jskeus/issues/405#issuecomment-266377527

kochigami, 2016-12-14:
Thank you very much for your help.

closed 2016-12-14 by kochigami

kochigami, 2016-12-14:
今ペッパーの腕の可動域をmove-end-posを使って調べているのですが、多分うまくいっていません。
例えば　```send *pepper* :larm :move-end-pos (float-vector 50 0 0 ) ``` をするときに
``` send *pepper* :larm :move-end-pos (float-vector 10 0 0 ) ``` を5回繰り返してやっているのですが、
ほとんどの範囲で失敗しています。

angle-vector を色々変えてend-coordsを取ってくるのを試します。

k-okada, 2016-12-15:
@haraduka ぜひ勉強して見てください