jsk-ros-pkg/jsk_robot #756: [Fetch] Moveit!を使っている時にrvizにcollision_objectが表示されない

YutoUchimi, 2017-03-09:
以下のプログラムを実行すると、rvizのMotion PlanningのScene Objectsにcollision_objectが追加されない問題が発生しました。
```
(setq *co* (instance collision-object-publisher :init))
(setq cube (make-cube 800 800 800))
(send *co* :add-object  cube :frame-id "base_link" :relative-pose (make-coords :pos #f(1000 0 400) :rpy #f(0 0 0)))
```
ROSでadvertiseとpublishを行うのが時間的に近すぎる際に発生する問題に起因すると判明しました。

pazeshun, 2017-03-09:
initとadd-objectの間に`(unix::sleep 3)`をいれると治ったので、上記のように判断しました

furushchev, 2017-03-09:
これはroseusでもrospyでも何でも共通ですが、advertiseからpublishまでは1秒くらいsleepを入れないと最初のpublishしたメッセージが受け取られず、その後のpublishも遅延すると思います。

-- ⌘ Yuki Furuta

k-okada, 2017-03-09:
sleepは最初はいいですが、だんだんsleepばかりになってロボットの動きがどんどんおそくなっていくので、気をつけたい所．

例えば、

```
#!/usr/bin/env roseus

(ros::load-ros-manifest "roseus")
(ros::roseus "talker")
(ros::advertise "chatter" std_msgs::string 1)
(setq msg (instance std_msgs::string :init))
(send msg :data (format nil "hello world ~a" (send (ros::time-now) :sec-nsec)))
(ros::ros-info "msg [~A]" (send msg :data))

(while (= (ros::get-num-subscribers "chatter") 0)
  (ros::ros-warn "waiting for /chatter subscriber")
  (unix:usleep 100000))
(ros::publish "chatter" msg)
```

でしょうか．しかしこれ、難しいですね．しらべると本当にみんな困っていますね．

http://ros-users.122217.n3.nabble.com/First-Message-Missing-td1462552.html
https://github.com/ros/ros_comm/issues/176
http://answers.ros.org/question/11167/how-do-i-publish-exactly-one-message/

最初のリンクの返事をしているKenはrospyつくった人だから信頼して良いことを考えると、latching というのがひとつの手でしょうか．

```
(ros::load-ros-manifest "roseus")
(ros::roseus "talker")
(ros::advertise "chatter" std_msgs::string 1 t) ;; queuesize = 1, latch = t            
(setq msg (instance std_msgs::string :init))
(send msg :data (format nil "hello world ~a" (send (ros::time-now) :sec-nsec)))
(ros::ros-info "msg [~A]" (send msg :data))
(ros::publish "chatter" msg)
```

これでどうでしょう． `:init` で `advertise`するときにlatchにしたらいいのかな．

furushchev, 2017-03-10:
なるほど、`latch`は思いつきませんでしたが、新しいnodeがsubscribeすると瞬間的にすごい数のメッセージが飛びそうという意味でこれかなりきわどいworkaroundだと思います。
よく考えて限定的に使わないといけないですね。

k-okada, 2017-03-10:
latchはlatest published message が出るだけなので、問題点があると、古いデータをうけつける、ということだとおもいます．
今回は、環境のstaticな状態を出しているんだと思うので、であれば問題ないと思います．

pazeshun, 2017-03-10:
なるほど、latchという手があるのですね。
今回に限って言えば、結局sleepは使わず、時間のかかるfetch-initの前に`*co*`のインスタンスを定義してしまうことで回避しています。

pazeshun, 2017-03-10:
今回はstaticですね

k-okada, 2017-03-10:
１年後に同じ問題が起こるので、

(setq *co* (instance collision-object-publisher :init))

で、launch でadvertiseするようにしておいてみて下さい．

knorth55, 2017-03-11:
これは確認なんですが、
この問題はMoveIt!固有の問題ではなく、`advertise`と`publish`が時間的に直近で行われると発生する問題という認識であってますか？
そして、これはroseusだけではなくroscpp、rospyでも起こりうるということですか？

少し思ったのですが、今はpublishするとtが返ってくるのですが、publisherのinitが完了されてない時にpublishしたらnilが返ってくるようにしてwhile loopを書くというのはどうなんでしょうか？

k-okada, 2017-03-11:
2017年3月11日 15:03 Shingo Kitagawa <notifications@github.com>:

> そして、これはroseusだけではなくroscpp、rospyでも起こりうるということですか？
>
>
はい．
http://ros-users.122217.n3.nabble.com/First-Message-Missing-td1462552.html
はrospyの事例ですね．
理由は多分だけど、
http://wiki.ros.org/ROS/Technical%20Overview#Establishing_a_topic_connection
にあるようにadvertiseすると、masterがそれをうけとって、そこから、subscriberに連絡がいって、
publisher にconnectしてlisten する（
http://www.jsk.t.u-tokyo.ac.jp/~ikuo/enshu/keisanki/chat-appli2.pdf )
するけど、それが終わる前にpublisherのpublish がかかるんだとおもいます．

なので、問題はpublisherではなくて、subscriberの準備ができていない、ということなのだとおもいます．

latch t にすると、listen した瞬間にpublisherはlast published
messageが送られるので、この問題が解決されている

ということなのでは、とおもいます．

コードも、デバッグ情報も見ずに、上の絵からの想像なので、確信度６０％ぐらいで、聞いて下さい．

--
◉ Kei Okada

knorth55, 2017-03-11:
なるほど、ありがとうございます。

latchも有効ですが、
publisherがinitするときにsubscriberがいる前提で書かれているプログラムなのであれば、subscriberのconnectが終わるまで待つというのも効果がありそうですね

knorth55, 2017-03-21:
Can we close this issue?

YutoUchimi, 2017-03-22:
Yes. Thanks for your kind answer.

closed 2017-03-22 by YutoUchimi