jsk-ros-pkg/jsk_robot #777: [pr2] Motors crash; Run-Stop unstable

Affonso-Gui, 2017-05-10:
While using the red pr2 today, the servos stopped working several times. The host remained reachable and was able to see/use topics normally, but no force on the motors.

Usually solved by pressing stop->run at Run-Stop, but sometimes needed to relaunch robot start or even restart the machine.

While trying to bring it up again at the end of the day, could not start Run-Stop properly. Pressing start button makes the LED goes up for a moment, but soon goes out. Sometimes motors move a bit, sometimes not.

Will get more exact error messages and rqt_pr2_dashboard states tomorrow.

@YutoUchimi
@furushchev

furushchev, 2017-05-10:
Can you see the terminal that you executed "robot start"?
Is there any error messages there or dashboard?

Affonso-Gui, 2017-05-10:
I remember there were a lot of error messages on both robot start terminal and pr2.launch one, but will confirm tomorrow.

As for the Run-Stop problem, the error on both robot start and dashboard was something like "cannot run calibration because motors are on halt. Please start Run-Stop.", but pressing start didn't solve it.

YutoUchimi, 2017-05-11:
今朝robot startしてtiltや首のモータをはじめすべてのモータが落ちました。やったことは
```
robot claim
robot start
```
で、Run-Stopをenableしろというメッセージが出たのでそれを行い、続いて
```
roslaunch jsk_pr2_startup pr2.launch
```
をしました。しばらく待っていると上記の通りモータが落ちました。

robot startを実行したterminalのメッセージを添付します。
[pr2_robot_start_test.txt](https://github.com/jsk-ros-pkg/jsk_robot/files/991676/pr2_robot_start_test.txt)

また、この時のpr2_dashboardの概要も動画で撮ったのでリンクを貼っておきます。
https://drive.google.com/open?id=0B1waEZT19wijZ1pCQWJYel9fTjA

Affonso-Gui, 2017-05-11:
Change the batteries of Run-Stop and seems to be working fine.

Will continue testing for a couple hours.

k-okada, 2017-05-11:
I think these batteries are re-chargeable one, so please remember re-charge
the batteries

Affonso-Gui, 2017-05-11:
Charged and placed on table under ROS flag (same place as the spare ones were).

Affonso-Gui, 2017-05-11:
Solved

closed 2017-05-11 by Affonso-Gui

k-okada, 2017-05-11:
Great!
Btw, did you noticed that some led on run stop device is blinking, if I
remember correctly it tell us that the battery is low.

Affonso-Gui, 2017-05-11:
The led near the stop button (with a battery icon) was indeed blinking. I didn't noticed yesterday, but Uchimi pointed it out this morning.

k-okada, 2017-05-12:
may be we should put tepra tape something like "when the icon is blinking,
change the batteries"

YutoUchimi, 2017-05-12:
> may be we should put tepra tape something like "when the icon is blinking, 
change the batteries"

Thanks for your advice. I put it on the Run-Stop just now.