jsk-ros-pkg/jsk_robot #785: PR1040でrosmasterのCPU使用率が高い問題

furushchev, 2017-05-31:
```diff
diff --git a/tools/rosmaster/src/rosmaster/master_api.py b/tools/rosmaster/src/rosmaster/master_api.py
index 8d0afd3..6e800a9 100644
--- a/tools/rosmaster/src/rosmaster/master_api.py
+++ b/tools/rosmaster/src/rosmaster/master_api.py
@@ -631,6 +631,7 @@ class ROSMasterHandler(object):
         finally:
             self.ps_lock.release()
         if service_url:
+            mlogwarn("Calling service: %s caller: %s url: %s" % (service, caller_id, service_url))^M
             return 1, "rosrpc URI: [%s]"%service_url, service_url
         else:
             return -1, "no provider", ''
```

をすると，service callをするときにroscoreを上げた画面にログが表示されますが，
実際にrobot startをしてみたところ，0.1Hzほどでservice callが行われているようです．
`roslaunch jsk_pr2_startup pr2.launch`をしてもservice callの頻度は変わらず，

- `robot start`

```bash
WARN: [1496189614.17] Called service: /projector_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
 WARN: [1496189614.17] Called service: /l_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
[WARN] [WallTime: 1496189614.662072] Caught exception <urlopen error [Errno 113] No route to host>
[INFO] [WallTime: 1496189614.917088] [CheckOpenNINode] sum of pixels is 1941370 at 1496189614
[ERROR] [WallTime: 1496189617.484173] Could not get message store services. Maybe the message store has not been started? Retrying..
[WARN] [WallTime: 1496189617.668096] Caught exception <urlopen error [Errno 113] No route to host>
[WARN] [WallTime: 1496189620.674085] Caught exception <urlopen error [Errno 113] No route to host>
[ERROR] [WallTime: 1496189622.632246] Could not get message store services. Maybe the message store has not been started? Retrying..
[WARN] [WallTime: 1496189623.680078] Caught exception <urlopen error [Errno 113] No route to host>
WARN: [1496189624.18] Called service: /projector_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189624.19] Called service: /head_camera_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189624.19] Called service: /l_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
 WARN: [1496189624.19] Called service: /r_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
[ WARN] [1496189625.416065599]: [/narrow_stereo/narrow_stereo_proc] [DebayerNodelet::imageCb] Color topic '/narrow_stereo/left/image_color' requested, but raw image data from topic '/narrow_stereo/left/image_raw' is grayscale
[WARN] [WallTime: 1496189626.686058] Caught exception <urlopen error [Errno 113] No route to host>
[ERROR] [WallTime: 1496189627.771933] Could not get message store services. Maybe the message store has not been started? Retrying..
[WARN] [WallTime: 1496189629.692138] Caught exception <urlopen error [Errno 113] No route to host>
[WARN] [WallTime: 1496189632.698068] Caught exception <urlopen error [Errno 113] No route to host>
[ERROR] [WallTime: 1496189632.921213] Could not get message store services. Maybe the message store has not been started? Retrying..
WARN: [1496189634.2] Called service: /l_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189634.2] Called service: /projector_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189634.2] Called service: /head_camera_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189634.2] Called service: /r_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
[WARN] [WallTime: 1496189635.704077] Caught exception <urlopen error [Errno 113] No route to host>
[ERROR] [WallTime: 1496189638.066638] Could not get message store services. Maybe the message store has not been started? Retrying..
[WARN] [WallTime: 1496189638.710068] Caught exception <urlopen error [Errno 113] No route to host>
[WARN] [WallTime: 1496189641.716168] Caught exception <urlopen error [Errno 113] No route to host>
[ERROR] [WallTime: 1496189643.209257] Could not get message store services. Maybe the message store has not been started? Retrying..
WARN: [1496189644.21] Called service: /projector_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189644.21] Called service: /r_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189644.22] Called service: /head_camera_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189644.22] Called service: /l_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
```
- detect cans

```bash
WARN: [1496189824.47] Called service: /projector_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058WARN: [1496189824.47] Called service: /r_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058

 WARN: [1496189824.47] Called service: /l_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189824.47] Called service: /head_camera_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
[ERROR] [WallTime: 1496189825.856065] Could not get message store services. Maybe the message store has not been started? Retrying..
[WARN] [WallTime: 1496189826.924120] Caught exception <urlopen error [Errno 113] No route to host>
[WARN] [WallTime: 1496189829.930050] Caught exception <urlopen error [Errno 113] No route to host>
[ERROR] [WallTime: 1496189830.995578] Could not get message store services. Maybe the message store has not been started? Retrying..
[WARN] [WallTime: 1496189832.936221] Caught exception <urlopen error [Errno 113] No route to host>
[ERROR] [1496189833.523627432]: [/runtime_logger] [master::execute] [getPublishedTopics] Failed to contact master at [c1:11311].  Retrying...
WARN: [1496189834.48] Called service: /projector_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189834.48] Called service: /l_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058 WARN: [1496189834.49] Called service: /r_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058

WARN: [1496189834.49] Called service: /head_camera_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
[WARN] [WallTime: 1496189835.942060] Caught exception <urlopen error [Errno 113] No route to host>
[ERROR] [WallTime: 1496189836.138571] Could not get message store services. Maybe the message store has not been started? Retrying..
[WARN] [WallTime: 1496189838.948208] Caught exception <urlopen error [Errno 113] No route to host>
[ERROR] [WallTime: 1496189841.278729] Could not get message store services. Maybe the message store has not been started? Retrying..
[WARN] [WallTime: 1496189841.954121] Caught exception <urlopen error [Errno 113] No route to host>
[ WARN] [1496189843.458414517]: [/narrow_stereo/narrow_stereo_proc] [DebayerNodelet::imageCb] Color topic '/narrow_stereo/left/image_color' requested, but raw image data from topic '/narrow_stereo/left/image_raw' is grayscale
WARN: [1496189844.5] Called service: /head_camera_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
 WARN: [1496189844.5] Called service: /projector_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189844.5] Called service: /r_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
WARN: [1496189844.5] Called service: /l_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:38058
[WARN] [WallTime: 1496189844.960081] Caught exception <urlopen error [Errno 113] No route to host>

k-okada, 2017-06-05:
1 robot start と pr2.launch で rosmaster の cpu load で差分は有りますか？
2 master_api.py の全ての関数の呼び出しを確認するとどうなりますか？

furushchev, 2017-06-05:
@k-okada 
以下のように変更すると、master apiのすべての呼び出しをrobot startを下画面に出力するようになりました。

```diff
$ diff /home/furushchev/ros_comm/tools/rosmaster/src/rosmaster/master_api.py /opt/ros/hydro/lib/python2.7/dist-packages/rosmaster/master_api.py
83c83
< LOG_API = False
---
> LOG_API = True
131c131
<                 #print "%s%s"%(f.func_name, str(args[1:]))
---
>                 print "[%s] DEBUG: %s%s"%(time.time(), f.func_name, str(args[1:]))
```

で、これでrobot startをすると、

```bash
[1496639782.47] DEBUG: getPublishedTopics('/runtime_logger', '')
[1496639782.54] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639782.84] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639783.15] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639783.45] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639783.47] DEBUG: getPublishedTopics('/runtime_logger', '')
[1496639783.75] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639784.05] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639784.35] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639784.47] DEBUG: getPublishedTopics('/runtime_logger', '')
[1496639784.57] DEBUG: getPublishedTopics('/narrow_stereo/narrow_stereo_proc', '')
[1496639784.66] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639784.96] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639785.26] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639785.47] DEBUG: getPublishedTopics('/runtime_logger', '')
[1496639785.56] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639785.87] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639786.17] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639786.47] DEBUG: getPublishedTopics('/runtime_logger', '')
[1496639786.47] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639786.78] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639787.08] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639787.16] DEBUG: lookupService('/camera_synchronizer_node', '/projector_trigger/set_waveform')
[1496639787.16] DEBUG: lookupService('/camera_synchronizer_node', '/head_camera_trigger/set_waveform')
WARN: [1496639787.16] Called service: /projector_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:41549
[1496639787.16] DEBUG: lookupService('/camera_synchronizer_node', '/r_forearm_cam_trigger/set_waveform')
WARN: [1496639787.16] Called service: /head_camera_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:41549
 [1496639787.16] DEBUG: lookupService('/camera_synchronizer_node', '/l_forearm_cam_trigger/set_waveform')
WARN: [1496639787.16] Called service: /r_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:41549
WARN: [1496639787.17] Called service: /l_forearm_cam_trigger/set_waveform caller_id: /camera_synchronizer_node url: rosrpc://pr1040:41549
[1496639787.38] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639787.47] DEBUG: getPublishedTopics('/runtime_logger', '')
[1496639787.69] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639787.99] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639788.29] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639788.47] DEBUG: getPublishedTopics('/runtime_logger', '')
[1496639788.59] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639788.89] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639789.2] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639789.47] DEBUG: getPublishedTopics('/runtime_logger', '')
[1496639789.5] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639789.8] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639790.1] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639790.41] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639790.47] DEBUG: getPublishedTopics('/runtime_logger', '')
[1496639790.71] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639791.01] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639791.31] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639791.47] DEBUG: getPublishedTopics('/runtime_logger', '')
[1496639791.61] DEBUG: lookupService('/replication_client', '/message_store/insert')
[1496639791.92] DEBUG: lookupService('/replication_client', '/message_store/insert')
```
[... 125 lines cut ...]
  - rosmasterのCPU使用率 1%
  - 全体のload 3.2
- pr2.launchしたとき
  - rosmasterのCPU使用率 25%
  - 全体のload 6.0
- detect cansをしたとき
  - rosmasterのCPU使用率 26%
  - 全体のload 9.0

になっていました。

さらに上のログを見ながら、

- `rosnode kill /active_user`

をすると、rosmasterのCPU使用率が16%くらい

- `rosnode kill /visualize_objectdetection /pr2_tweet_log`

をすると、10%くらいになりました。

ちょっとコードを見てみますが、mongodbへのinsertやquery(全部service)が原因の一員になっているのが考えられそうです。
insertについてはtopic publishに対応するPull Requestを送ってmergeされているので、リリースされたらこちらに切り替えていこうと思っています。

（しかし、以前見たときの90%くらいCPU使用率があったのはまた原因もこれなのでしょうか。。）

k-okada, 2017-06-05:
下、僕が犯人？
https://github.com/jsk-ros-pkg/jsk_robot/pull/787
みたいなのだと良くなるのかな．

furushchev, 2017-06-05:
@k-okada のちほど、誰も使っていない時に再度ベンチマークをしてみます。もしそれで下がれば、rosservice callだけでなくて、hasParamやgetParamも頻繁に呼ぶと負担がかかると言えそうです。