jsk-ros-pkg/jsk_robot #786: [jsk_pr2_desktop] PR2 icons are automatically added to Desktop

knorth55, 2017-06-05:
When I build `jsk_pr2_desktop`, `PR2` icons are automatically added by `catkin build jsk_pr2_desktop`.
This icon installation should be move to script program like `rosrun jsk_pr2_desktop install_icons.py`

[image]

furushchev, 2017-06-05:
IMO, `jsk_pr2_startup` is built for only PR2 users, and I think it's good idea to begin with this commands to start PR2.

knorth55, 2017-06-05:
These icons are for pr2 users only.
Those who use jsk_robot but not pr2user, i.e. baxter and fetch users do not want these.

furushchev, 2017-06-05:
@knorth55 So why they build `jsk_pr2_startup`?

pazeshun, 2017-06-05:
I think the following way is simplest:
make workspace -> wstool update -> rosdep install -> `catkin build`
I don't want to remember which package I should build.

knorth55, 2017-06-05:
Because it is included in jsk_robot package.
When i clone this repo, this package is also cloned.

knorth55, 2017-06-05:
> I think the following way is simplest:
make workspace -> wstool update -> rosdep install -> catkin build
I don't want to remember which package I should build.

What do you mean?

furushchev, 2017-06-05:
@knorth55 @pazeshun 
So basically you build all packages in workspace? Simply `catkin build` in workspace builds all packages in the workspace including unrelated packages and takes longer time on build. You can instead simply `catkin build jsk_apc` for example to build all packages that is necessary for this package.

furushchev, 2017-06-05:
Another possible idea is that changing desktop entry files from desktop to `~/.local/share/applications`, which is user space of menu icons.
This does not create desktop shortcuts but does menu in unity dashboard.

[image]

@k-okada How do you think of this?

knorth55, 2017-06-05:
I always use catkin build because my workspace is not large.
I know i can build with cakin build [package].
My point is adding icons on desktop is not build but installation, i think.

closed 2017-06-08 by knorth55