jsk-ros-pkg/jsk_robot #794: PR1012のEtherCATのNICでrx errorが多い

furushchev, 2017-06-19:
PR1012でサーボがたまに落ちてしまう問題があります。
現象としては

- ethercatのパケットをたまに（１時間に１回くらい）１パケット落とす
  - これだけでは別にサーボが落ちない
- 短時間に10パケット以上ロスする時がある。（よく起きるときは起きるが、起きない時は起きない）
  - 不定期
  - サーボが落ちる

となっている。

NICの統計値を見ると、rx_crc_errorとrx_frame_errorが大きそう。

```bash
root@pr1012:/opt/ros/indigo/share/ethercat_hardware# ethtool -S ecat0
NIC statistics:
     rx_packets: 174948809
     tx_packets: 174995994
     rx_bytes: 149951435090
     tx_bytes: 149990379825
     rx_broadcast: 174948809
     tx_broadcast: 174995940
     rx_multicast: 0
     tx_multicast: 54
     rx_errors: 26644
     tx_errors: 0
     tx_dropped: 0
     multicast: 0
     collisions: 0
     rx_length_errors: 0
     rx_over_errors: 0
     rx_crc_errors: 13328
     rx_frame_errors: 13316
     rx_no_buffer_count: 0
     rx_missed_errors: 0
     tx_aborted_errors: 0
     tx_carrier_errors: 0
     tx_fifo_errors: 0
     tx_heartbeat_errors: 0
     tx_window_errors: 0
     tx_abort_late_coll: 0
     tx_deferred_ok: 0
     tx_single_coll_ok: 0
     tx_multi_coll_ok: 0
     tx_timeout_count: 0
     tx_restart_queue: 0
     rx_long_length_errors: 0
     rx_short_length_errors: 0
     rx_align_errors: 13316
     tx_tcp_seg_good: 0
     tx_tcp_seg_failed: 0
     rx_flow_control_xon: 0
     rx_flow_control_xoff: 0
     tx_flow_control_xon: 0
     tx_flow_control_xoff: 0
     rx_csum_offload_good: 0
     rx_csum_offload_errors: 0
     rx_header_split: 0
     alloc_rx_buff_failed: 0
     tx_smbus: 0
     rx_smbus: 0
     dropped_smbus: 0
     rx_dma_failed: 0
     tx_dma_failed: 0
     rx_hwtstamp_cleared: 0
     uncorr_ecc_errors: 0
     corr_ecc_errors: 0
     tx_hwtstamp_timeouts: 0
```

参考： PR1040

```bash
pr2admin@pr1040:~$ sudo ethtool -S ecat0
NIC statistics:
     rx_packets: 4465636378
     tx_packets: 4465636479
     rx_bytes: 3851823554644
     tx_bytes: 3851828347313
     rx_broadcast: 4465636378
     tx_broadcast: 4465636442
     rx_multicast: 0
     tx_multicast: 37
     rx_errors: 34
     tx_errors: 0
     tx_dropped: 0
     multicast: 0
     collisions: 0
     rx_length_errors: 0
     rx_over_errors: 0
     rx_crc_errors: 17
     rx_frame_errors: 17
     rx_no_buffer_count: 0
     rx_missed_errors: 0
     tx_aborted_errors: 0
     tx_carrier_errors: 0
     tx_fifo_errors: 0
     tx_heartbeat_errors: 0
     tx_window_errors: 0
     tx_abort_late_coll: 0
     tx_deferred_ok: 0
     tx_single_coll_ok: 0
     tx_multi_coll_ok: 0
     tx_timeout_count: 0
     tx_restart_queue: 0
     rx_long_length_errors: 0
     rx_short_length_errors: 0
     rx_align_errors: 17
     tx_tcp_seg_good: 0
     tx_tcp_seg_failed: 0
     rx_flow_control_xon: 0
     rx_flow_control_xoff: 0
     tx_flow_control_xon: 0
     tx_flow_control_xoff: 0
     rx_long_byte_count: 3851823554644
     rx_csum_offload_good: 0
     rx_csum_offload_errors: 0
     rx_header_split: 0
     alloc_rx_buff_failed: 0
     tx_smbus: 0
     rx_smbus: 0
     dropped_smbus: 0
     rx_dma_failed: 0
     tx_dma_failed: 0
```

furushchev, 2017-06-19:
C1からecatのbase hubへのLANケーブルを新品と交換してみました。

```bash
ethtool -S ecat0
NIC statistics:
     rx_packets: 4539256
     tx_packets: 4539393
     rx_bytes: 3891239324
     tx_bytes: 3891314885
     rx_broadcast: 4539256
     tx_broadcast: 4539357
     rx_multicast: 0
     tx_multicast: 36
     rx_errors: 178
     tx_errors: 0
     tx_dropped: 0
     multicast: 0
     collisions: 0
     rx_length_errors: 0
     rx_over_errors: 0
     rx_crc_errors: 89
     rx_frame_errors: 89
     rx_no_buffer_count: 0
     rx_missed_errors: 0
     tx_aborted_errors: 0
     tx_carrier_errors: 0
     tx_fifo_errors: 0
     tx_heartbeat_errors: 0
     tx_window_errors: 0
     tx_abort_late_coll: 0
     tx_deferred_ok: 0
     tx_single_coll_ok: 0
     tx_multi_coll_ok: 0
     tx_timeout_count: 0
     tx_restart_queue: 0
     rx_long_length_errors: 0
     rx_short_length_errors: 0
     rx_align_errors: 89
     tx_tcp_seg_good: 0
     tx_tcp_seg_failed: 0
     rx_flow_control_xon: 0
     rx_flow_control_xoff: 0
     tx_flow_control_xon: 0
     tx_flow_control_xoff: 0
     rx_csum_offload_good: 0
     rx_csum_offload_errors: 0
     rx_header_split: 0
     alloc_rx_buff_failed: 0
     tx_smbus: 0
     rx_smbus: 0
     dropped_smbus: 0
     rx_dma_failed: 0
     tx_dma_failed: 0
     rx_hwtstamp_cleared: 0
     uncorr_ecc_errors: 0
     corr_ecc_errors: 0
     tx_hwtstamp_timeouts: 0
```

rx_errors / rx_packetsをみると、

- PR1012交換前: 0.015230%
- PR1012交換後: 0.003921%
- PR1040: 7.61369e-07%

PR1040には遠く及びませんが、エラー率は1/3になった？
Wiresharkなどでパケットを見たらどこが悪いかわかるのでしょうか。。。

furushchev, 2017-06-20:
ケーブルを変えた状態で半日サーボを入れたまま放置してみましたが、サーボが落ちることは今のところなくなった気がします。

`robot start`のログを見ると

```bash
readMailboxInternal : ERROR asking for read repeat after dropping 1 packets
```

が3回出ているだけで、サーボが落ちるときによく見るエラー

```bash
clearReadMailbox : WARN read mbx contained garbage data
waitForReadMailboxReady : ERROR error read mbx not full after 100 ms
readMailbox_ : ERRORwaiting for read mailbox
clearReadMailbox : WARN read mbx contained garbage data
waitForReadMailboxReady : ERROR error read mbx not full after 100 ms
readMailbox_ : ERRORwaiting for read mailbox
```

は出ていないようです。

furushchev, 2017-06-20:
ただやはりエラーになるパケットはそこそこあるようです。

```bash
7: ecat0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether 00:15:17:c3:b7:7a brd ff:ff:ff:ff:ff:ff
    RX: bytes  packets  errors  dropped overrun mcast
    88804041056 103544435 3466    0       0       0
    TX: bytes  packets  errors  dropped carrier collsns
    88804960303 103546421 0       0       0       0
```

iory, 2019-07-09:
この問題ですが、pr1012で再び起きていることを確認しました。
以下はethtoolで調べた結果です。

```
root@pr1012:~# ethtool -S ecat0
NIC statistics:
     rx_packets: 935657
     tx_packets: 940163
     rx_bytes: 773204684
     tx_bytes: 778778006
     rx_broadcast: 935657
     tx_broadcast: 940133
     rx_multicast: 0
     tx_multicast: 30
     rx_errors: 8448
     tx_errors: 0
     tx_dropped: 0
     multicast: 0
     collisions: 0
     rx_length_errors: 0
     rx_over_errors: 0
     rx_crc_errors: 4225
     rx_frame_errors: 4223
     rx_no_buffer_count: 0
     rx_missed_errors: 0
     tx_aborted_errors: 0
     tx_carrier_errors: 0
     tx_fifo_errors: 0
     tx_heartbeat_errors: 0
     tx_window_errors: 0
     tx_abort_late_coll: 0
     tx_deferred_ok: 0
     tx_single_coll_ok: 0
     tx_multi_coll_ok: 0
     tx_timeout_count: 0
     tx_restart_queue: 0
     rx_long_length_errors: 0
     rx_short_length_errors: 0
     rx_align_errors: 4223
     tx_tcp_seg_good: 0
     tx_tcp_seg_failed: 0
     rx_flow_control_xon: 0
     rx_flow_control_xoff: 0
     tx_flow_control_xon: 0
     tx_flow_control_xoff: 0
     rx_csum_offload_good: 0
     rx_csum_offload_errors: 0
     rx_header_split: 0
     alloc_rx_buff_failed: 0
     tx_smbus: 0
     rx_smbus: 0
     dropped_smbus: 0
     rx_dma_failed: 0
     tx_dma_failed: 0
     rx_hwtstamp_cleared: 0
     uncorr_ecc_errors: 0
     corr_ecc_errors: 0
 ```

knorth55, 2019-07-09:
FYI: 今のpr1040はこんな感じです
```
pr2admin@pr1040:~$ ethtool -S ecat0
NIC statistics:
     rx_packets: 526758618
     tx_packets: 526758713
     rx_bytes: 451834761504
     tx_bytes: 451835062890
     rx_broadcast: 526758618
     tx_broadcast: 526758621
     rx_multicast: 0
     tx_multicast: 92
     rx_errors: 4
     tx_errors: 0
     tx_dropped: 0
     multicast: 0
     collisions: 0
     rx_length_errors: 0
     rx_over_errors: 0
     rx_crc_errors: 2
     rx_frame_errors: 2
     rx_no_buffer_count: 0
     rx_missed_errors: 0
     tx_aborted_errors: 0
     tx_carrier_errors: 0
     tx_fifo_errors: 0
     tx_heartbeat_errors: 0
     tx_window_errors: 0
     tx_abort_late_coll: 0
     tx_deferred_ok: 0
     tx_single_coll_ok: 0
     tx_multi_coll_ok: 0
     tx_timeout_count: 0
     tx_restart_queue: 0
     rx_long_length_errors: 0
     rx_short_length_errors: 0
     rx_align_errors: 2
     tx_tcp_seg_good: 0
     tx_tcp_seg_failed: 0
     rx_flow_control_xon: 0
     rx_flow_control_xoff: 0
     tx_flow_control_xon: 0
     tx_flow_control_xoff: 0
     rx_csum_offload_good: 0
     rx_csum_offload_errors: 0
     rx_header_split: 0
     alloc_rx_buff_failed: 0
     tx_smbus: 0
     rx_smbus: 0
     dropped_smbus: 0
     rx_dma_failed: 0
     tx_dma_failed: 0
     rx_hwtstamp_cleared: 0
     uncorr_ecc_errors: 0
     corr_ecc_errors: 0
     tx_hwtstamp_timeouts: 0
```

ZdenekM, 2019-12-16:
@knorth55 Were you able to find out cause of the issue? At our lab, we have similar issue...

knorth55, 2020-08-01:
@ZdenekM sorry for late response. we are also looking for how to solve this issue.
it seems that some motor driver board are not working well, but we are not sure.

knorth55, 2020-12-14:
@ZdenekM in our lab, we found out that the slip rings of each caster are damaged, which causes a lot of packet drop.
we replaced the splip ring and the packet drop got reduced. it is worth checking, i think.