jsk-ros-pkg/jsk_robot #803: Cannot locate rosdep definition for [fetch_bringup]

knorth55, 2017-07-15:
```
rosdep install -q -y --rosdistro indigo -n -q --ignore-src --skip-keys=jsk_smart_gui --skip-keys=ros3djs --skip-keys=pr2_calibration_launch --skip-keys=jsk_android_gui_api9 --skip-keys=ros2djs --skip-keys=face_recognition --skip-keys=roslibjs --skip-keys=force_proximity_ros --skip-keys=safe_teleop_base --from-paths /opt/ros/indigo/share .

ERROR: the following packages/stacks could not have their rosdep keys resolved

to system dependencies:

jsk_fetch_startup: Cannot locate rosdep definition for [fetch_bringup]

+EXIT_STATUS=1
```

k-okada, 2017-07-15:
can you use install deb version of jsk_robot on jsk_apc _
https://github.com/start-jsk/jsk_apc/blob/master/.travis.rosinstall#L1-L4

knorth55, 2017-07-16:
I checked rosdistro and I think `fetch_bringup` or `fetch_robots` is not distributed in any rosdistro. 
Do we need to set `fetch_bringup`?

k-okada, 2017-07-16:
sorry, I just released latest version of `jsk_robot`, https://github.com/ros/rosdistro/pull/15522

if you want o install `fetch_bringup` on your local environment, try
```
wget http://packages.fetchrobotics.com/fetch/ubuntu/pool/main/r/ros-indigo-fetch-bringup/ros-indigo-fetch-bringup_0.7.3-0-20161129-1502--0800trusty_amd64.deb
wget http://packages.fetchrobotics.com/fetch/ubuntu/pool/main/r/ros-indigo-fetch-drivers/ros-indigo-fetch-drivers_0.7.15-0-20161129-1449--0800trusty_amd64.deb
sudo gdebi ros-indigo-fetch-drivers_0.7.15-0-20161129-1449--0800trusty_amd64.deb 
sudo gdebi ros-indigo-fetch-bringup_0.7.3-0-20161129-1502--0800trusty_amd64.deb 
```

knorth55, 2017-07-16:
>if you want o install fetch_bringup on your local environment, try

I mean if `packages` are not released and distributed in `rosdistro`, we have no need to add the package in `package.xml` because we need to install it manually. 
I think it is better to remove `fetch_bringup` from `package.xml` and write manual installation documents in README.

k-okada, 2017-07-17:
humm, I think it too strict to say "do not put put packages which is not
distributed in rosdistro", for example original baxter sim was only
available from private repo, so user who using baxter usually clone the
source tree to their workspace and compile them, in that case, we want some
of jsk_demo or jsk_robot package depends on baxter sim package, so if we
compile our baxter app package, it should compile baxter-sim too.
If fetch case, that is distributed as deb and we usually do not put them in
source tree, so it is ok to remove from package.xml...

knorth55, 2017-07-18:
OK. It doesn't matter in `jsk_apc` anymore, so I close this issue.

by the way, in my workspace, it says these packages cannot be found with deb. 

```
$ rosdep install --ignore-src --from-path . -y -r -i
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
jsk_baxter_web: Cannot locate rosdep definition for [ros3djs]
jsk_pr2_startup: Cannot locate rosdep definition for [jsk_smart_gui]
jsk_fetch_startup: Cannot locate rosdep definition for [fetch_bringup]
jsk_pr2_calibration: Cannot locate rosdep definition for [pr2_calibration_launch]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

closed 2017-07-18 by knorth55