jsk-ros-pkg/jsk_robot #819: [PR1012] PR1012 becomes servo-off suddenly

mmurooka, 2017-08-11:
PR2を使わせてもらっている中でのトラブルで周知の問題かどうかわからないので共有します．
cc: @furushchev , @asilx

During experiment, PR2 suddenly servo off. By pushing STOP and START button in run stop, sometimes PR2 becomes recover to servo on, but becomes servo off again right after that or after a few minutes.

I shut down c1 and set red toggle switch in back off and on, and tried again, but same problem occurred after starting experiments ten minutes. I feel that there is a small noise before servo is turned off suddenly.
@asilx said that he have not experienced this problem, so I'm wandering that I have some mistake in my PC settings. In my experiment, I send angle-vector and go-velocity from my euslisp node.
Maybe @furushchev said about this problem before. Do you know the trigger that this problem occurs, and how to recover when this problem occurs?

mmurooka, 2017-08-11:
Thank you for talking about this problem directly with @furushchev and @asilx .
I'll check following when I have the sample problem.

1．
```
rosrun rqt_robot_monitor rqt_robot_monitor
```
Error is displayed in
All devices -> Motors -> Full Name: /Motors/EtherCAT Device (br_caster_rotation_motor)

2．
In terminal that launch `robot start`,
Error message like followings are outputed:
```
clearReadMailbox : WARN read mbx contained garbage data
waitForReadMailboxReady : ERROR error read mbx not full after 100 ms
readMailbox_ : ERRORwaiting for read mailbox
```
or
Some message like `too many packets drops`

3．
@asilx move PR2 from node launched in `applications@pr1012` and had no problem,
and I move PR2 from node in my desktop or laptop PC and had problem.
That might effect the condition that problem occurs.

ref: https://github.com/furushchev/jsk_pr2/issues/44

furushchev, 2017-08-30:
@mmurooka 
I replaced base hub for ethercat and now it seems that it works great.
Please check and report if there is still any problem.
(See also detail if you'd like to: https://github.com/furushchev/jsk_pr2/issues/44 )

mmurooka, 2017-08-30:
Thank you very much! I reopen this issue if I got same problem again.

closed 2017-08-30 by mmurooka

ZdenekM, 2019-12-16:
@mmurooka @furushchev At our lab, we have similar problem now. Was it really solved by replacing the hub? Do you know part number / where to order it? Thank you very much...

knorth55, 2019-12-16:
@ZdenekM Hi, we solved the issue by replacing the hub, but the servo off still seldom happens.
we investigate with `ethtool -S ecat0` and our problem was there were so much error packets.
We replaced ethercat hub, cables and found out that replacing hub suppress the packet error.
This is just a quick reply, and now I'm digging up how we bought the hub, so please wait a bit.

knorth55, 2019-12-16:
@ZdenekM 
It seems we bought the hub from clearpath.
https://www.clearpathrobotics.com/assets/Price_List_PR2.pdf

ZdenekM, 2019-12-16:
Great, thanks for your response @knorth55 . We also have many error packets. I will try to ask Clearpath if the hub is available... It is really weird, because there are days where the robot works flawlessly and days where it is totally nonoperational - more off than on. Probably it is good idea to replace cables as well.

knorth55, 2019-12-17:
in our lab, also, the servo off sometimes suddenly happens and happens continuously after the first servo off.
but these days, it does not happen.
this is just a guess, but we doubt the high temperature of the hub, pc, battery and base because the aerodynamics is not well-considered in base design.