jsk-ros-pkg/jsk_robot #821: [naoqieus] nao_interaction_msgs is not released on ros kinetic

kochigami, 2017-08-23:
just a memo:

Based on http://wiki.ros.org/nao_interaction_msgs
nao_interaction_msgs release stops at jade.

It seems that I have to modify 
- package.xml (build, run depend)
- CMakeLists.txt (find_package)
- naoqi-interface.l (delete methods using nao_interaction_msgs)
- README.md (indigo => kinetic)

In addition, I had troubled in making roseus_msgs while executing ```catkin build```. 
However, I'm not sure what the actual problem is yet and I restarted installing ubuntu...  

```
catkin build (following https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/README.md)
Errors     << naoqieus:cmake /home/pepper/catkin_ws/logs/naoqieus/build.cmake.003.log
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "nao_interaction_msgs" with any of the following names:

    nao_interaction_msgsConfig.cmake
    nao_interaction_msgs-config.cmake

  Add the installation prefix of "nao_interaction_msgs" to CMAKE_PREFIX_PATH
  or set "nao_interaction_msgs_DIR" to a directory containing one of the
  above files.  If "nao_interaction_msgs" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "nao_interaction_msgs" with any of the following names:

    nao_interaction_msgsConfig.cmake
    nao_interaction_msgs-config.cmake

  Add the installation prefix of "nao_interaction_msgs" to CMAKE_PREFIX_PATH
  or set "nao_interaction_msgs_DIR" to a directory containing one of the
  above files.  If "nao_interaction_msgs" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

```

k-okada, 2017-08-23:
Asked for release at https://github.com/ros-naoqi/nao_interaction/issues/8,
for a meantime, download nao_interaction sourcetree in your workspace

kochigami, 2017-08-24:
Thank you very much for your reply.
However, I think it's ok to delete nao_interaction_launchers dependency.
Anyway I'll try deleting nao_interaction_msgs dependency even though it is not necessary.

- I think the way of written naoqi module is old in nao_interaction package.
- I don't think it's good that using nao-named package for pepper. 
- I wrote some functions available for this package (volume changer at naoqi_driver (not merged yet) and sound detection at naoqi_apps (not merged yet and I can merge it in the future)).
- The biggest problem is they are not merged yet and we have to use them from source by ```git merge``` by ourselves.

kochigami, 2018-11-22:
> However, I think it's ok to delete nao_interaction_msgs dependency.

`nao_interaction_msgs` is required for `naoqi-interface.l`
`AudioPlayback.srv` and `AudioMasterVolume.srv`

I'll explain this in README.

closed 2018-11-22 by kochigami