jsk-ros-pkg/jsk_robot #823: [naoqieus] how to create manifest.l automatically on kinetic?

kochigami, 2017-08-24:
I followed https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/README.md
and setup pepper's environment with kinetic.
However, it seems that manifest.l is not created automatically while ```catkin build``` on kinetic.

I'd like to know if it is an expected behavior and what I should do.

reference: #678 (it is working on indigo, but it doesn't on kinetic)

```
roscore
roseus pepper-interface.l

[ERROR] [1503541401.073768641]: Could not find roseus messages for naoqieus under (/home/pepper/catkin_ws/devel/share/roseus/ros /opt/ros/kinetic/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh naoqieus
[ERROR] [1503541401.073860996]: Could not find nil/manifest.l
try rosrun roseus generate-all-msg-srv.sh naoqieus
Call Stack (max depth: 20):
  0: at (apply #'ros::load-org-for-ros ros::fullname args)
  1: at (apply #'ros::load-org-for-ros ros::fullname args)
  2: at (let ((ros::fullname fname)) (when (substringp "package://" fname) (setq ros::fullname (ros::resolve-ros-path fname)) (when ros::*compile-message* (let* ((ros::urlname (url-pathname fname)) (package-name (send ros::urlname :host)) (ros::path-name (format nil "~A_~A" package-name (substitute 95 47 (concatenate string (subseq (send ros::urlname :directory-string) 1) (send ros::urlname :name))))) (ros::old-module (find ros::path-name *loaded-modules* :key #'lisp::load-module-file-name :test #'equal))) (unless ros::old-module (let* ((ros::ppath (unix:getenv "CMAKE_PREFIX_PATH")) (dir (format nil "~A/share/roseus/ros/~A" (subseq ros::ppath 0 (position 58 ros::ppath)) package-name))) (unless (probe-file dir) (unix:mkdir dir)) (compiler:compile-file-if-src-newer ros::fullname (format nil "~A/~A" dir ros::path-name)) (return-from load (load (format nil "~A/~A.so" dir ros::path-name) :entry (format nil "___~A" ros::path-name)))))))) (if (null ros::fullname) (error "file ~s not found" fname)) (apply #'ros::load-org-for-ros ros::fullname args))
  3: at (load "package://naoqieus/naoqi-interface.l")
/opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 no such package "NAO_INTERACTION_MSGS""NAO_INTERACTION_MSGS" in (apply #'ros::load-org-for-ros ros::fullname args)E:
```

```
pepper@pepper-desktop:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/naoqieus$ rosrun roseus generate-all-msg-srv.sh naoqieus
package:naoqieus -> /home/pepper/catkin_ws/src/jsk_robot/jsk_naoqi_robot/naoqieus
generating... 1 files with ALL=No, COMPILE=No option
writing output to.../home/pepper/catkin_ws/devel/share/roseus/ros
generating... /0
generating manifest... /home/pepper/catkin_ws/devel/share/roseus/ros/naoqieus/manifest.l
```

Then ```roseus pepper-interface.l``` & ```pepper-init t``` succeeded.

k-okada, 2017-08-24:
@kochigami This is very important report, thanks a lot!!

From jade, roseus is one of the "official" ros client, that means all catkin process generate roseus message, (see `dpkg -L ros-kinetic-naoqi-bridge-msgs | grep eus` on kinetic). But this will not make any roseus message for packages without msg/srv, so we need to skip reading manifest.l for these packages

This should be solved by, https://github.com/jsk-ros-pkg/jsk_roseus/pull/537, please check this.

kochigami, 2017-08-25:
Thank you very much for your reply. I'm very glad to hear roseus was promoted.

However, after I  tried jsk-ros-pkg/jsk_roseus#537, something is wrong.
I expect this is caused by my fault, and I'll look into it later.

What I did:
- ```git clone``` three pkgs
```
pepper@pepper-desktop:~/catkin_ws/src$ ls
jsk_robot (master)  jsk_roseus (#537)  nao_interaction (master)
```

- ```catkin clean -b``` and ```catkin clean -d```
- ```catkin build -c```
- ```roscore```, ```roscd peppereus & roseus pepper-interface.l``` 

```
roseus ;; loading roseus("1.6.2-5-g1524954") on euslisp((9.23 ip-172-31-22-159 Tue Jun 13 17:50:01 PST 2017  1.1.0))
eustf roseus_c_util Call Stack (max depth: 20):
  0: at (apply #'ros::load-org-for-ros ros::fullname args)
  1: at (apply #'ros::load-org-for-ros ros::fullname args)
  2: at (let ((ros::fullname fname)) (when (substringp "package://" fname) (setq ros::fullname (ros::resolve-ros-path fname)) (when ros::*compile-message* (let* ((ros::urlname (url-pathname fname)) (package-name (send ros::urlname :host)) (ros::path-name (format nil "~A_~A" package-name (substitute 95 47 (concatenate string (subseq (send ros::urlname :directory-string) 1) (send ros::urlname :name))))) (ros::old-module (find ros::path-name *loaded-modules* :key #'lisp::load-module-file-name :test #'equal))) (unless ros::old-module (let* ((ros::ppath (unix:getenv "CMAKE_PREFIX_PATH")) (dir (format nil "~A/share/roseus/ros/~A" (subseq ros::ppath 0 (position 58 ros::ppath)) package-name))) (unless (probe-file dir) (unix:mkdir dir)) (compiler:compile-file-if-src-newer ros::fullname (format nil "~A/~A" dir ros::path-name)) (return-from load (load (format nil "~A/~A.so" dir ros::path-name) :entry (format nil "___~A" ros::path-name)))))))) (if (null ros::fullname) (error "file ~s not found" fname)) (apply #'ros::load-org-for-ros ros::fullname args))
  3: at (load "package://naoqieus/naoqi-interface.l")
/opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 no such package "NAO_INTERACTION_MSGS""NAO_INTERACTION_MSGS" in (apply #'ros::load-org-for-ros ros::fullname args)E: 
```

memo:
- nao_interaction_msg has manifest.l
```
pepper@pepper-desktop:~/catkin_ws/devel/share/roseus/ros/nao_interaction_msgs$ ls
manifest.l  msg  srv
```
- source code is referenced
```
pepper@pepper-desktop:~/catkin_ws/devel/share/roseus/ros/nao_interaction_msgs$ roscd nao_interaction_msgs/
pepper@pepper-desktop:~/catkin_ws/src/nao_interaction/nao_interaction_msgs$
```

- roseus, peppereus, naoqieus sources are also referenced
```
pepper@pepper-desktop:~$ roscd roseus
pepper@pepper-desktop:~/catkin_ws/src/jsk_roseus/roseus$ cd 
pepper@pepper-desktop:~$ roscd peppereus/
pepper@pepper-desktop:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/peppereus$ roscd naoqieus/
pepper@pepper-desktop:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/naoqieus$
```

k-okada, 2017-08-25:
humm, basically https://github.com/jsk-ros-pkg/jsk_roseus/pull/537 changes our strategy to write roseus program. We will ask user to use `load-** <msg>` instead of  using `load-** package`, because we nolong build manifest file for package without msg/srv.

we still able to use `ros::load-ros-manifest`, but may be better to use `ros::load-ros-package` or `(ros::roseus-add-msgs)` or `(ros::roseus-add-srvs)`

```
--- a/jsk_naoqi_robot/naoqieus/naoqi-interface.l
+++ b/jsk_naoqi_robot/naoqieus/naoqi-interface.l
@@ -1,5 +1,5 @@
 (load "package://pr2eus/robot-interface.l")
-(ros::load-ros-manifest "naoqieus")
+(ros::load-ros-manifest "nao_interaction_msgs")
```
or
```
(ros::load-ros-package "nao_interaction_msgs")
```
or
```
(ros::roseus-add-msgs "nao_interaction_msgs")
(ros::roseus-add-srvs "nao_interaction_msgs")
```

k-okada, 2017-08-25:
i have updated https://github.com/jsk-ros-pkg/jsk_roseus/pull/537, this does not require you to change `naoqi-interface.l`

closed 2017-09-08 by k-okada

k-okada, 2018-01-07:
> i have updated jsk-ros-pkg/jsk_roseus#537, this does not require you to change naoqi-interface.l

I'm afraid we still need to change naoqi-interface.l on kinetic. Did you change locally? @kochigami

kochigami, 2018-01-11:
Actually, I've never succeeded in executing pepper-interface.l on kinetic.
After my last comment, I haven't tried https://github.com/jsk-ros-pkg/jsk_roseus/pull/537 ...
Sorry about that.

I tried these lines again.
```
pepper@pepper-desktop:~/catkin_ws/src$ ls
jsk_robot (master)  jsk_roseus (#537)  nao_interaction (master)

% git pull --rebase origin master
% #537: git fetch k-okada & git merge k-okada/fix_jade 
```

- ```catkin clean -b``` and ```catkin clean -d```
- ```catkin build -c```
- ```roscore```, ```roscd peppereus & roseus pepper-interface.l``` 

```
/opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 no such package "NAO_INTERACTION_MSGS""NAO_INTERACTION_MSGS" in (apply #'ros::load-org-for-ros ros::fullname args)E: 
```

pazeshun, 2018-01-11:
Oh, similar error to https://github.com/jsk-ros-pkg/jsk_roseus/issues/554...