jsk-ros-pkg/jsk_robot #825: pr2 leans after moving base

furushchev, 2017-08-24:
[image]

[frames.pdf](https://github.com/jsk-ros-pkg/jsk_robot/files/1248685/frames.pdf)

suspicious transformations:
/map -> /odom_combined -> /base_footprint
the former is published by /amcl and the latter is by /robot_pose_ekf

furushchev, 2017-08-24:
I found /robot_pose_ekf publishes strange tf viewing `rosrun tf tf_echo`...

furushchev, 2017-08-24:
Disabled to input imu topic to `robot_pose_ekf` and it looks it solved.

k-okada, 2017-08-24:
Did we use robot_pose_ekf on hydro?
I'm afraid you have solved THIS problrmb, but also we have lost good
localization result .

furushchev, 2017-08-24:
@k-okada You are right!
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_pr2_robot/jsk_pr2_startup/pr2_bringup.launch#L150
On hydro we also use imu data.
Maybe the problem is originated from that imu is not calibrated or `robot_pose_ekf` is changed on indigo.
I'll keep this issue open then.