jsk-ros-pkg/jsk_robot #830: Cannot connect joystick to robot

furushchev, 2017-09-05:
joystickのPSボタンを押してもつながらない時がある。
`rqt_runtime_monitor`にjoy stick not foundというエラーあり。

furushchev, 2017-09-05:
`sudo restart sixad`をしてから再度接続をするとつながるようになった。

closed 2017-09-05 by furushchev

k-okada, 2017-09-05:
言葉の定義の問題ですが、以下で直しているつもりなんだけど．．．

```
k-okada@fetch15:~$ sudo crontab -l
# m h  dom mon dow   command
* * * * * /home/fetch/restart-sixad.sh
*/10 * * * * /home/fetch/reconnect-wifi.sh
```

furushchev, 2017-09-05:
`restart-sixad.sh`の中身を少しだけ見ましたが、logのサイズが大きな時のみリスタートをしているようです
確かにlogを見ると同じ表示が繰り返し出ているようです。
僕がログファイルがまだ小さなときに(200KBくらいで)気づいてrestartするのが早いだけだと思います。

furushchev, 2017-09-06:
やっぱりlogの繰り返しの表示が出ていなくても接続ができていない場合がありそうです。

具体的には

```bash
D-Bus setup failed: Name already in use
sixad-bin[1424]: started
sixad-bin[1424]: sixad started, press the PS button now
```

という表示のままjoystickのボタンを押してもつながらないということになります。
その後`sudo restart sixad`をして10秒ほど待つと、

```bash
sixad-sixaxis[740]: started
sixad-sixaxis[740]: Connected 'PLAYSTATION(R)3 Controller (FC:62:B9:0C:88:D0)' [Battery 05]
```

というように正常に接続できるようになります。