jsk-ros-pkg/jsk_robot #837: Cannot ping PR1040

Affonso-Gui, 2017-09-12:
Cannot connect to red PR2.

 `ping pr1040` returns
```
From PC.jsk.imi.i.u-tokyo.ac.jp (133.11.216.118) icmp_seq=1 Destination Host Unreachable
```

furushchev, 2017-09-12:
I also confirmed ping does not reach to `pr1040`.
Pinging to `prbase1040` is also failed so I think there is something wrong with basestation of PR1040.
(BTW, for PR1012 there is no more base station.)
I resetted base station. Maybe because of power failure on the last weekend.

furushchev, 2017-09-12:
After rebooting basestation, I can ping to `pr1040`.

```bash
furushchev@musca:~$ ping pr1040
PING pr1040.jsk.imi.i.u-tokyo.ac.jp (133.11.216.211) 56(84) bytes of data.
64 bytes from pr1040.jsk.imi.i.u-tokyo.ac.jp (133.11.216.211): icmp_seq=1 ttl=63 time=0.791 ms
^C
--- pr1040.jsk.imi.i.u-tokyo.ac.jp ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 0.791/0.791/0.791/0.000 
```

Affonso-Gui, 2017-09-12:
Seems to be back to normal.

Thank you very much.

closed 2017-09-12 by Affonso-Gui

furushchev, 2017-09-12:
**MEMO** When turning on the base station, the case fan sounds strange.
I cleaned up the fan with air duster.