jsk-ros-pkg/jsk_robot #838: Cannot successfully init PR1040

Affonso-Gui, 2017-09-12:
Tilt laser doesn't come up and cannot communicate with robot through pr2eus after startup.  

`jsk_pr2_startup pr2.launch` returns
```
[ERROR] [1505208093.557711292]: Could not find roseus messages for jsk_interactive under (/home/affonso/ros/hydro/devel/share/roseus/ros /opt/ros/hydro/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh jsk_interactive

[ERROR] [1505208093.558065825]: Could not find nil/manifest.l
try rosrun roseus generate-all-msg-srv.sh jsk_interactive
```
However, trying `rosrun roseus generate-all-msg-srv.sh jsk_interactive` as suggested fails with
```
[rospack] Error: stack/package jsk_interactive not found
generating... 0 files with ALL=No, COMPILE=No option
writing output to.../home/affonso/ros/hydro/devel/share/roseus/ros
[rospack] Error: stack/package jsk_interactive not found
generating... /0
basename: missing operand
Try `basename --help' for more information.
[rospack] Error: no package/stack given
[rospack] Error: no package/stack given
[rospack] Error: no package/stack given
WARN in 
generating manifest... /home/affonso/ros/hydro/devel/share/roseus/ros//manifest.l
Usage: gen_eus.py file

gen_eus.py: error: please specify args
ERROR in 
```
Last week the problem got solved by running `rosrun roseus generate-all-msg-srv.sh` (without jsk_interactive designation), but that does not seem to be of any help today.

As far as I can tell `robot start` is working normally. 
Other errors / warnings given by `pr2.launch` are given bellow. 
```
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: [/home/affonso/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/launch/pointcloud_screenpoint.launch] unknown <arg> attribute 'doc'
WARNING: unrecognized tag sphinxdoc

[ERROR] [1505208061.397758324]: Skipping XML Document "/opt/ros/hydro/share/laser_filters_jsk_patch/laser_filters_plugins.xml" which had no Root Element.  This likely means the XML is malformed or missing.

[ERROR] [WallTime: 1505208069.968501] Could not get message store services. Maybe the message store has not been started? Retrying..

  File "/usr/lib/python2.7/dist-packages/pymongo/connection.py", line 653, in __find_node
    raise AutoReconnect(', '.join(errors))
AutoReconnect: could not connect to localhost:27017: [Errno 111] Connection refused

[ERROR] [WallTime: 1505208092.938773] pr2_tuck_arms: left_joint_client action server did not come up within timelimit

remote[c2-0]:  connect to MongoDB: couldn't connect to server localhost:27017

[ERROR] [1505208092.804886743]: Failed to connect to MongoDB: couldn't connect to server localhost:27017

[ERROR] [1505208093.557711292]: Could not find roseus messages for jsk_interactive under (/home/affonso/ros/hydro/devel/share/roseus/ros /opt/ros/hydro/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh jsk_interactive

[ERROR] [1505208093.558065825]: Could not find nil/manifest.l
try rosrun roseus generate-all-msg-srv.sh jsk_interactive
```

Also, not directively related, but maintainer email in [http://wiki.ros.org/jsk_interactive](http://wiki.ros.org/jsk_interactive) seems to be misspelled.

Affonso-Gui, 2017-09-12:
Apparently left caster calibration could not be completed because it was stuck in the carpet, creating problems in further initialization.

`jsk_interactive` msg didn't seem to be the problem.

closed 2017-09-12 by Affonso-Gui

furushchev, 2017-09-12:
You can see what is going on during `robot start` by adding following script to `~/.bashrc`:

```bash
function robot() {
    if [ "$1" == "start" ]; then
        /usr/bin/robot start --debug
    else
        /usr/bin/robot $*
    fi
}
```

and do `robot start` again with new terminal.

For more detail please see https://docs.google.com/presentation/d/1In6V-P77rRCYffv3ze1oqxkTYBQwDKrEqSNUxeRhFF4/edit#slide=id.g124f7eef9_062