jsk-ros-pkg/jsk_robot #840: [fetch] :state :potentio-vectorの挙動

708yamaguchi, 2017-09-28:
fetchの`/joint_states`には、グリッパに関するトピックとそれ以外に関するトピックの2種類が出ているので、
`(send *ri* :state :potentio-vector :wait-until-update t)`
としないと、全ての関節角情報が更新されない。

knorth55, 2017-09-28:
これは/joint_statesのHzはどれくらい？

708yamaguchi, 2017-09-28:
およそ200Hzでした。
```
$ rostopic hz /joint_states 
subscribed to [/joint_states]
average rate: 200.854
	min: 0.000s max: 0.012s std dev: 0.00437s window: 192
average rate: 199.933
	min: 0.000s max: 0.012s std dev: 0.00446s window: 391
```

mmurooka, 2017-09-28:
参考までにHRP2は，全身関節と指の/joint_statesを別のノードが出していて，
片方を`rosnode kill`して`rostopic hz`すると，全身関節角は50Hz，指は10Hzでした．
`(send *ri* :state :potentio-vector)`で`:wait-until-update t`にすることはめったにないと思いますがいつも全身が更新されている気分です．（実は6回に1回は正しくできていないのかもしれません．）

708yamaguchi, 2017-10-19:
#843 で解決されました。
ありがとうございました。

closed 2017-10-19 by 708yamaguchi