jsk-ros-pkg/jsk_robot #842: [pepper] pepper_full.launchが上がらない

kochigami, 2017-09-28:
Cc: @shoheiishikura 
書いてね

shoheiishikura, 2017-09-28:
以下がroslaunch pepper_bringup pepper_full.launchを実行したときのエラーです．
ーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーー
leus@leus-ThinkPad-T450:~$ roslaunch pepper_bringup pepper_full.launch nao_ip:=133.11.216.155 network_interface:=wlan0
... logging to /home/leus/.ros/log/7d8ca0f8-a406-11e7-ad00-1002b53754f6/roslaunch-leus-ThinkPad-T450-28800.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.180 :40632/

SUMMARY
========

PARAMETERS
 * /pepper_robot/camera/camera_nodelet_manager/num_worker_threads: 4
 * /pepper_robot/camera/depth_rectify_depth/interpolation: 0
 * /pepper_robot/camera/depth_registered_rectify_depth/interpolation: 0
 * /pepper_robot/pose/pose_manager/poses/init/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/init/positions: [0.0, 0.0, 1.39, ...
 * /pepper_robot/pose/pose_manager/poses/init/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/poses/zero/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/zero/positions: [0.0, 0.0, 0.0, 0...
 * /pepper_robot/pose/pose_manager/poses/zero/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/xap: /home/leus/ros/in...
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /pepper_robot/camera/
    bottom_rectify_color (nodelet/nodelet)
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    front_rectify_color (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_front (nodelet/nodelet)
  /pepper_robot/pose/
    pose_controller (naoqi_pose/pose_controller.py)
    pose_manager (naoqi_pose/pose_manager.py)
  /
    pepper_robot (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[pepper_robot-1]: started with pid [28818]
process[pepper_robot/pose/pose_controller-2]: started with pid [28819]
process[pepper_robot/pose/pose_manager-3]: started with pid [28820]
Receiving information about robot model
process[pepper_robot/camera/camera_nodelet_manager-4]: started with pid [28833]
process[pepper_robot/camera/front_rectify_color-5]: started with pid [28834]
process[pepper_robot/camera/bottom_rectify_color-6]: started with pid [28837]
process[pepper_robot/camera/ir_rectify_ir-7]: started with pid [28858]
[W] 28818 qimessaging.remoteobject: Return signature might be incorrect depending on the value, from m to s
process[pepper_robot/camera/depth_rectify_depth-8]: started with pid [28882]
process[pepper_robot/camera/depth_metric_rect-9]: started with pid [28896]
process[pepper_robot/camera/depth_metric-10]: started with pid [28902]
process[pepper_robot/camera/register_depth_front-11]: started with pid [28911]
process[pepper_robot/camera/points_xyzrgb_sw_registered-12]: started with pid [28945]
process[pepper_robot/camera/depth_registered_sw_metric_rect-13]: started with pid [28959]
process[pepper_robot/camera/depth_registered_rectify_depth-14]: started with pid [28979]
process[pepper_robot/camera/points_xyzrgb_hw_registered-15]: started with pid [28989]
process[pepper_robot/camera/depth_registered_hw_metric_rect-16]: started with pid [29002]
[... 84 lines cut ...]
[pepper_robot/camera/depth_registered_sw_metric_rect-13] killing on exit
[pepper_robot/camera/points_xyzrgb_sw_registered-12] killing on exit
[pepper_robot/camera/register_depth_front-11] killing on exit
[pepper_robot/camera/depth_metric-10] killing on exit
[pepper_robot/camera/depth_metric_rect-9] killing on exit
[pepper_robot/camera/depth_rectify_depth-8] killing on exit
[pepper_robot/camera/ir_rectify_ir-7] killing on exit
[pepper_robot/camera/bottom_rectify_color-6] killing on exit
[pepper_robot/camera/front_rectify_color-5] killing on exit
[pepper_robot/camera/camera_nodelet_manager-4] killing on exit
[pepper_robot/pose/pose_manager-3] killing on exit
[pepper_robot/pose/pose_controller-2] killing on exit
[pepper_robot-1] killing on exit
[INFO] [WallTime: 1506598170.882129] [0.000000] Stopping pose_controller
[INFO] [WallTime: 1506598170.882274] [0.000000] pose_controller stopped
[I] 29546 qi.Application: Sending the stop command...
[INFO] [WallTime: 1506598171.013674] [0.000000] nao pose_controller stopped.
^C^C^C^C^C^C^C^C^C^C^C^C^C^C^[[A[pepper_robot-1] escalating to SIGTERM
[I] 29546 qi.Application: Sending the stop command...
[pepper_robot-1] escalating to SIGKILL
Shutdown errors:
 * process[pepper_robot-1, pid 28818]: required SIGKILL. May still be running.
shutting down processing monitor...
... shutting down processing monitor complete
done

kochigami, 2017-09-28:
```using ip address: 127.0.0.1 @ wlan0 ```　が怪しいと思うので、明日ソースコードを見てみます。

メモ：
- １．pepper_robot, naoqi_driverをソースコードで入れる
- ２．pepper_robot のファイルの一部をこれの変更を見て（今のところは）書き換える
https://github.com/ros-naoqi/pepper_robot/pull/40

- ペッパーの調子が悪かったらROS-Driver のエラーが出る
- ペッパーのネットワークとパソコンのネットワークが違う
- 2.5.5のpynaoqiを使うとpythonpathのエラーが出るんだけど、私は問題なかった
- naoqiの2.4.4と2.5.5で構成がちょっと違うようだ 2.5.5 だと /pynaoqi/lib もパスにいれないといけない

％もし、pepper_full.launch （名前を忘れた・・・）を立ち上げた時に、
大量のログが出て問題になったら、
一様これを変えると消える
https://github.com/ros-naoqi/naoqi_driver/pull/88/files

kochigami, 2017-09-28:
引数 roscore_ip に手元のパソコンのipを入れてあげると値が入る
- jsk_pepper_startup.launchだと、$ROS_IPの値を取ってきて勝手に入れてくれる

shoheiishikura, 2017-09-28:
pepper_full.launchの6行目を書き換えました．
https://github.com/ros-naoqi/pepper_robot/blob/master/pepper_bringup/launch/pepper_full.launch
<arg name="roscore_ip"          default="$(optenv ROS_IP)" />

メッセージを見ると，
``using ip address: 133.11.216.180 @ wlan0``
と変わりましたが，
``Failed to contact master at [133.11.216.180:11311].  Retrying...``
と言われました．
ーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーー
leus@leus-ThinkPad-T450:~/ros/indigo-pepper/src/pepper_robot/pepper_bringup/laun
ch$ roslaunch pepper_bringup pepper_full.launch nao_ip:=133.11.216.155  network_interface:=wlan0 roscore_ip:=133.11.216.180
... logging to /home/leus/.ros/log/91d79ff2-a444-11e7-86cd-1002b53754f6/roslaunch-leus-ThinkPad-T450-7544.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.180 :37481/

SUMMARY
========

PARAMETERS
 * /pepper_robot/camera/camera_nodelet_manager/num_worker_threads: 4
 * /pepper_robot/camera/depth_rectify_depth/interpolation: 0
 * /pepper_robot/camera/depth_registered_rectify_depth/interpolation: 0
 * /pepper_robot/pose/pose_manager/poses/init/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/init/positions: [0.0, 0.0, 1.39, ...
 * /pepper_robot/pose/pose_manager/poses/init/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/poses/zero/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/zero/positions: [0.0, 0.0, 0.0, 0...
 * /pepper_robot/pose/pose_manager/poses/zero/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/xap: /home/leus/ros/in...
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /pepper_robot/camera/
    bottom_rectify_color (nodelet/nodelet)
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    front_rectify_color (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_front (nodelet/nodelet)
  /pepper_robot/pose/
    pose_controller (naoqi_pose/pose_controller.py)
    pose_manager (naoqi_pose/pose_manager.py)
  /
    pepper_robot (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[pepper_robot-1]: started with pid [7562]
process[pepper_robot/pose/pose_controller-2]: started with pid [7563]
process[pepper_robot/pose/pose_manager-3]: started with pid [7564]
process[pepper_robot/camera/camera_nodelet_manager-4]: started with pid [7577]
process[pepper_robot/camera/front_rectify_color-5]: started with pid [7578]
process[pepper_robot/camera/bottom_rectify_color-6]: started with pid [7584]
process[pepper_robot/camera/ir_rectify_ir-7]: started with pid [7606]
process[pepper_robot/camera/depth_rectify_depth-8]: started with pid [7610]
process[pepper_robot/camera/depth_metric_rect-9]: started with pid [7625]
[... 51 lines cut ...]
registered subscriber:	speech
nodehandle reset 
[ERROR] [1506600035.570258536]: [registerPublisher] Failed to contact master at [133.11.216.180:11311].  Retrying...
^C[pepper_robot/camera/depth_registered_metric-17] killing on exit
[pepper_robot/camera/depth_registered_hw_metric_rect-16] killing on exit
[pepper_robot/camera/points_xyzrgb_hw_registered-15] killing on exit
[pepper_robot/camera/depth_registered_rectify_depth-14] killing on exit
[pepper_robot/camera/depth_registered_sw_metric_rect-13] killing on exit
[pepper_robot/camera/points_xyzrgb_sw_registered-12] killing on exit
[pepper_robot/camera/register_depth_front-11] killing on exit
[pepper_robot/camera/depth_metric-10] killing on exit
[pepper_robot/camera/depth_metric_rect-9] killing on exit
[pepper_robot/camera/depth_rectify_depth-8] killing on exit
^C[pepper_robot/camera/ir_rectify_ir-7] killing on exit
[pepper_robot/camera/bottom_rectify_color-6] killing on exit
[pepper_robot/camera/front_rectify_color-5] killing on exit
[pepper_robot/camera/camera_nodelet_manager-4] killing on exit
[pepper_robot/pose/pose_controller-2] killing on exit
[INFO] [WallTime: 1506600040.058005] Stopping pose_controller
[INFO] [WallTime: 1506600040.058275] pose_controller stopped
[pepper_robot-1] killing on exit
[INFO] [WallTime: 1506600040.714418] nao pose_controller stopped.
shutting down processing monitor...
... shutting down processing monitor complete
done

kochigami, 2017-09-28:
ログありがとう！役に立つか分からないけれど・・・

0. ubuntu 14.04 ros indigoで
pynaoqi のバージョンは2.5.5でしょうか。もしそうだったら、これは何か別の問題を持っていそう？なので、一旦2.4.3にしてもらっていいでしょうか。

1. ```naoqi_driver.launch```と ```pepper_full.launch```の```roscore_ip```のファイル上の引数の変更を元に戻して、
の```roslaunch pepper_bringup pepper_full.launch network_interface:=wlan0 roscore_ip:=<ifconfig のwlan0のip>```　してみたらどうなるでしょうか。
また、同じエラーが出たら、実行している端末でifconfig => wlan0の値を確認したらどうなるでしょうか。```echo $ROS_IP``` はどうなるでしょうか。

2. 今までの変更は一旦リセットして、
> １．pepper_robot, naoqi_driverをソースコードで入れる
２．pepper_robot のファイルの一部をこれの変更を見て（今のところは）書き換える
ros-naoqi/pepper_robot#40

をもう一回やってもらって、```roslaunch jsk_pepper_startup jsk_pepper_startup.launch network_interface:=wlan0``` したらどうなるでしょうか。

３．2つのパッケージのビルドをし直すのはどうでしょうか。

kochigami, 2017-09-29:
メモ：
nao_ipは設定しないといけないが、そうしないといけないことが分かりやすくなっていないかも
http://ros-naoqi.github.io/naoqi_driver/start.html#before-starting-the-module
にあるけど

shoheiishikura, 2017-09-29:
@kochigami さん

まず1番を試した結果になります．
今までのものと同じエラーかと思います．

お願いします．
ーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーー
```
leus@leus-ThinkPad-T450:~/ros/indigo-pepper/src/pepper_robot/pepper_bringup/laun
ch$ roslaunch pepper_bringup pepper_full.launch network_interface:=wlan0 roscore_ip:=133.11.216.180
... logging to /home/leus/.ros/log/91d79ff2-a444-11e7-86cd-1002b53754f6/roslaunch-leus-ThinkPad-T450-10859.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.180 :43923/

SUMMARY
========

PARAMETERS
 * /pepper_robot/camera/camera_nodelet_manager/num_worker_threads: 4
 * /pepper_robot/camera/depth_rectify_depth/interpolation: 0
 * /pepper_robot/camera/depth_registered_rectify_depth/interpolation: 0
 * /pepper_robot/pose/pose_manager/poses/init/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/init/positions: [0.0, 0.0, 1.39, ...
 * /pepper_robot/pose/pose_manager/poses/init/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/poses/zero/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/zero/positions: [0.0, 0.0, 0.0, 0...
 * /pepper_robot/pose/pose_manager/poses/zero/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/xap: /home/leus/ros/in...
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /pepper_robot/camera/
    bottom_rectify_color (nodelet/nodelet)
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    front_rectify_color (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_front (nodelet/nodelet)
  /pepper_robot/pose/
    pose_controller (naoqi_pose/pose_controller.py)
    pose_manager (naoqi_pose/pose_manager.py)
  /
    pepper_robot (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[pepper_robot-1]: started with pid [10877]
process[pepper_robot/pose/pose_controller-2]: started with pid [10878]
process[pepper_robot/pose/pose_manager-3]: started with pid [10879]
Receiving information about robot model
process[pepper_robot/camera/camera_nodelet_manager-4]: started with pid [10894]
process[pepper_robot/camera/front_rectify_color-5]: started with pid [10895]
process[pepper_robot/camera/bottom_rectify_color-6]: started with pid [10896]
process[pepper_robot/camera/ir_rectify_ir-7]: started with pid [10905]
[W] 10877 qimessaging.remoteobject: Return signature might be incorrect depending on the value, from m to s
process[pepper_robot/camera/depth_rectify_depth-8]: started with pid [10928]
process[pepper_robot/camera/depth_metric_rect-9]: started with pid [10941]
[... 85 lines cut ...]

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:65513255 errors:0 dropped:0 overruns:0 frame:0
          TX packets:65513255 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:234059792530 (234.0 GB)  TX bytes:234059792530 (234.0 GB)

wlan0     Link encap:Ethernet  HWaddr 10:02:b5:37:54:f6  
          inet addr:133.11.216.180  Bcast:133.11.216.255  Mask:255.255.255.0
          inet6 addr: fe80::1202:b5ff:fe37:54f6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3638185 errors:0 dropped:4 overruns:0 frame:0
          TX packets:1419844 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4845226477 (4.8 GB)  TX bytes:160580579 (160.5 MB)
```

```
leus@leus-ThinkPad-T450:~/ros/indigo-pepper/src/pepper_robot/pepper_bringup/laun
ch$ echo $ROS_IP
133.11.216.180
```

shoheiishikura, 2017-09-29:
一度再起動し，すべてのshellを落としたら，
``
[ERROR]Failed to contact master at [133.11.216.180:11311].  Retrying...
``
のエラーはでなくなりました．

shoheiishikura, 2017-09-29:
``roslaunch pepper_bringup pepper_full.launch``を実行し``/pepper_robot/camera/front/image_raw``をrvizで確認しようとした所，imageが表示されませんでした．（以下がその時の状況です．）

このとき``rostopic echo /pepper_robot/camera/front/image_raw``をしましたが，値は返されませんでした．
また``/pepper_robot/camera/depth_registered/points``も同様の状況です．
一方で``rostopic echo /tf``すると，値が返されました．

ーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーー

```
leus@leus-ThinkPad-T450:~$ roslaunch pepper_bringup pepper_full.launch network_interface:=wlan0
... logging to /home/leus/.ros/log/c93332da-a4db-11e7-94b9-1002b53754f6/roslaunch-leus-ThinkPad-T450-32583.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.180 :44629/

SUMMARY
========

PARAMETERS
 * /pepper_robot/camera/camera_nodelet_manager/num_worker_threads: 4
 * /pepper_robot/camera/depth_rectify_depth/interpolation: 0
 * /pepper_robot/camera/depth_registered_rectify_depth/interpolation: 0
 * /pepper_robot/pose/pose_manager/poses/init/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/init/positions: [0.0, 0.0, 1.39, ...
 * /pepper_robot/pose/pose_manager/poses/init/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/poses/zero/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/zero/positions: [0.0, 0.0, 0.0, 0...
 * /pepper_robot/pose/pose_manager/poses/zero/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/xap: /home/leus/ros/in...
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /pepper_robot/camera/
    bottom_rectify_color (nodelet/nodelet)
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    front_rectify_color (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_front (nodelet/nodelet)
  /pepper_robot/pose/
    pose_controller (naoqi_pose/pose_controller.py)
    pose_manager (naoqi_pose/pose_manager.py)
  /
    pepper_robot (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[pepper_robot-1]: started with pid [32601]
process[pepper_robot/pose/pose_controller-2]: started with pid [32602]
process[pepper_robot/pose/pose_manager-3]: started with pid [32603]
Receiving information about robot model
[W] 32601 qimessaging.remoteobject: Return signature might be incorrect depending on the value, from m to s
process[pepper_robot/camera/camera_nodelet_manager-4]: started with pid [32618]
process[pepper_robot/camera/front_rectify_color-5]: started with pid [32619]
process[pepper_robot/camera/bottom_rectify_color-6]: started with pid [32620]
process[pepper_robot/camera/ir_rectify_ir-7]: started with pid [32633]
process[pepper_robot/camera/depth_rectify_depth-8]: started with pid [32645]
[W] 32601 qimessaging.remoteobject: Return signature might be incorrect depending on the value, from m to [[m]]
[... 154 lines cut ...]
[I] 1726 qi.Application: Sending the stop command...
Audio Extractor: Stop
[INFO] [WallTime: 1506665646.255840] nao pose_controller stopped.
ROS-Driver-RightBumperPressed : Stop
ROS-Driver-HandRightBackTouched : Stop
ROS-Driver-FrontTactilTouched : Stop
naoqi driver is shutting down..
shutting down processing monitor...
... shutting down processing monitor complete
done
```

ちなみにtopic情報は以下の様に受けとることができました．
```
leus@leus-ThinkPad-T450:~$ rostopic info /pepper_robot/camera/front/image_raw
Type: sensor_msgs/Image

Publishers: 
 * /pepper_robot (http://133.11.216.180:33382/)

Subscribers: None
```

どのようにすれば，RGB画像や点群の情報を得られますでしょうか．
よろしくお願いします．

k-okada, 2017-09-29:
pepperリブートかな．

shoheiishikura, 2017-09-30:
@k-okada 先生， @kochigami さん

pepperをrebootしたところ，カメラ画像と点群のtopicを取得できました．

ちなみに，以下のFiles changedを行った状態で``roslaunch pepper_bringup pepper_full.launch``を実行しました．
https://github.com/ros-naoqi/pepper_robot/pull/40/files

ありがとうございました．

closed 2017-09-30 by shoheiishikura

kochigami, 2017-10-11:
@shoheiishikura くん
ログ等貼ってくれてありがとうございました。
READMEに反映していこうと思います。 

メモ：
pynaoqi-python2.7-2.5.5.5-linux64のパスの通し方
```
export PYTHONPATH=$HOME/pynaoqi/pynaoqi-python2.7-2.5.5.5-linux64/lib/python2.7/site-packages:$PYTHONPATH
```
になると思う

理由：
naoqi_poseパッケージの中のプログラムで関係
naoqi_driver_py/src/naoqi_node.pyが
python SDKの中のnaoqi.pyを読み込みたくて、そこにパスを誘導してあげないといけないから

（%別件で、https://github.com/ros-naoqi/naoqi_driver/issues/68
には2.5.5はkineticで使おうという意見がある）