jsk-ros-pkg/jsk_robot #844: [fetch] testが通りません

708yamaguchi, 2017-09-29:


k-okada, 2017-09-29:
無限回転軸だからかな．

何を正解とすべきかはむづかしいんだけど、多分だけど、
```
5.irteusgl$ send *fetch* :angle-vector
#f(385.822 13.4851 29.5611 -158.632 113.043 -182.46 80.8974 -18.0974 0.0 0.0)
6.irteusgl$ send *ri* :state :potentio-vector
#f(380.0 14.4741 30.3673 -156.29 112.813 174.715 81.1236 -17.8094 0.0 0.0)
```
となっていて送るときは-182を希望するんだけど、`*ri*`のなかで
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L72
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L890
あたりで、近い方の178を目標に補間されている
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/pr2-interface.l#L368
のようにして、-182側に廻るようにすると、期待通りになることはなる．

knorth55, 2017-09-29:
これmasterのtestが通っているのはなんでなんですか？

708yamaguchi, 2017-09-29:
上の`test`関数で`:wait-interpolation`が抜けていたので
```
(defun test ()
  (let (diff-av)
    (send *fetch* :inverse-kinematics (make-coords :pos #f(800 0 1300)))
    (send *ri* :angle-vector (send *fetch* :angle-vector))
    (send *ri* :wait-interpolation)
    (setq diff-av (v- (send *ri* :state :potentio-vector) (send *fetch* :angle-vector)))
    (assert (eps= (norm diff-av) 0.0 5.0) (format nil "diff-av ~A" diff-av))
    ))
```
としましたが、同じ現象が起きました。

knorth55, 2017-10-12:
@708yamaguchi Can we close this issue?

708yamaguchi, 2017-10-12:
すみません。閉じます。

closed 2017-10-12 by 708yamaguchi