jsk-ros-pkg/jsk_robot #859: HDD空き容量がない

furushchev, 2017-11-01:
```bash
furushchev@fetch15:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7.8G  4.0K  7.8G   1% /dev
tmpfs           1.6G  1.4M  1.6G   1% /run
/dev/sda2        94G   90G     0 100% /
none            4.0K     0  4.0K   0% /sys/fs/cgroup
none            5.0M     0  5.0M   0% /run/lock
none            7.8G  2.4M  7.8G   1% /run/shm
none            100M   44K  100M   1% /run/user
/dev/sda1       511M  3.4M  508M   1% /boot/efi
```

```bash
furushchev@fetch15:/var$ sudo du -d1 -h
7.4G	./lib
4.0K	./mail
4.0K	./crash
4.0K	./opt
4.0K	./metrics
4.0K	./local
68K	./spool
1.5M	./www
19M	./backups
258M	./cache
18G	./log
4.0K	./tmp
26G	.
```

```bash
furushchev@fetch15:/home$ sudo du -d1 -h
13G	./fetch
1000K	./s-kitagawa
25G	./furushchev
2.7G	./affonso-gui
3.6G	./yamaguchi
11G	./wkentaro
2.7G	./k-okada
23M	./iory
160K	./murooka
57G	.
```

furushchev, 2017-11-01:
別のロボットで運用していたのと同じく，jskの認識関係のデータ（ニューラルネットの重みやrosbagなど）をユーザごとにダウンロードしていたのを共有するようにして，各ユーザディレクトリの重みを一度消しました．

```bash
$ cat /etc/profile.d/jsk.sh 
# added by furushchev (2017.11.2)
export ROSLAUNCH_SSH_UNKNOWN=1
export JSK_DATA_CACHE_DIR=/etc/ros/jsk_data
```

```bash
fetch@fetch15:/home$ sudo du -d1 -h
8.7G	./fetch
1000K	./s-kitagawa
8.6G	./furushchev
2.7G	./affonso-gui
3.6G	./yamaguchi
8.3G	./wkentaro
2.7G	./k-okada
23M	./iory
160K	./murooka
35G	.
```

furushchev, 2017-11-01:
別の原因として，`/var/log/ros`以下がlogrotateされていないようです．
`logrotate`を手動で実行してみると，

```bash
$ sudo /usr/sbin/logrotate /etc/logrotate.d/ros
error: skipping "/var/log/ros/fd645e8c-9a09-11e5-8547-d8cb8a40210c/head_camera-depth_registered_rectify_depth-12-stdout.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/ros/fd645e8c-9a09-11e5-8547-d8cb8a40210c/head_camera-driver-6-stdout.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/ros/fd645e8c-9a09-11e5-8547-d8cb8a40210c/head_camera-points_xyzrgb_hw_registered-13-stdout.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
```

というようにパーミッションの設定がおかしくなっていそうです．

```bash
furushchev@fetch15:/var/log/ros$ ls -lFahd
drwxrwsr-x 381 ros ros 36K Nov  2 01:45 ./
```

他の同等のことをしているロボットでは

```bash
furushchev@XXXX:/var/log/ros$ ls -lFahd
drwxr-xr-x 6 ros ros 36K 11月  1 15:25 ./
```

となっているので，以下を実行

```bash
sudo chmod g-w /var/log/ros
(cd /var/log/ros && find -type d | xargs sudo chmod g-w)
```

を実行しました．

furushchev, 2017-11-01:
logrotateを強制的に実行して，圧縮されたものに関してはもとのディレクトリを削除しました．

```bash
$ sudo logrotate -f /etc/logrotate.d/ros
$ (cd /var/log/ros && find -type d | grep ./ | xargs sudo rm -rf)  # 結構危ない
```

また`sudo restart robot`をしてみましたが，新しく出来たフォルダ`/var/log/ros/03d6ac60-bf5c-11e7-97cd-d8cb8a40210c`のパーミッションは

```bash
furushchev@fetch15:/var/log/ros/03d6ac60-bf5c-11e7-97cd-d8cb8a40210c$ ls -lFahd
drwxr-sr-x 2 ros ros 12K Nov  2 08:39 ./
```

となっていて，logrotateでエラーが起きない正常なものになっているようです．

furushchev, 2017-11-01:
```bash
furushchev@fetch15:/var$ sudo du -d1 -h
[sudo] password for furushchev: 
7.4G	./lib
4.0K	./mail
4.0K	./crash
4.0K	./opt
4.0K	./metrics
4.0K	./local
68K	./spool
1.5M	./www
19M	./backups
258M	./cache
5.7G	./log
4.0K	./tmp
14G	.
```

logディレクトリも12GBほど削減できました．

furushchev, 2017-11-01:
```bash
furushchev@fetch15:/var$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7.8G  4.0K  7.8G   1% /dev
tmpfs           1.6G  1.4M  1.6G   1% /run
/dev/sda2        94G   71G   19G  80% /
none            4.0K     0  4.0K   0% /sys/fs/cgroup
none            5.0M     0  5.0M   0% /run/lock
none            7.8G   80K  7.8G   1% /run/shm
none            100M   44K  100M   1% /run/user
/dev/sda1       511M  3.4M  508M   1% /boot/efi
```

20GBだと余裕はあまりないかもしれませんが，これで一応様子見ということにしたいと思います．
また足りなくなったら相談しましょう．
@708yamaguchi

closed 2017-11-01 by furushchev

k-okada, 2017-11-02:
ありがとう．
https://github.com/jsk-ros-pkg/jsk_robot/pull/861
に情報をまとめました．

708yamaguchi, 2017-11-19:
今日fetchを使っていたのですが、同じ状態になってしまいました。どうすればいいでしょう？
```
yamaguchi@fetch15 20:45:57 :~ $
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7.8G  4.0K  7.8G   1% /dev
tmpfs           1.6G  1.3M  1.6G   1% /run
/dev/sda2        94G   91G     0 100% /
none            4.0K     0  4.0K   0% /sys/fs/cgroup
none            5.0M     0  5.0M   0% /run/lock
none            7.8G     0  7.8G   0% /run/shm
none            100M  8.0K  100M   1% /run/user
/dev/sda1       511M  3.4M  508M   1% /boot/efi
overflow        1.0M  4.0K 1020K   1% /tmp
```

```
yamaguchi@fetch15 21:01:37 :/var $
$ sudo du -d1 -h
12G	./lib
4.0K	./mail
4.0K	./crash
4.0K	./opt
4.0K	./metrics
4.0K	./local
68K	./spool
1.5M	./www
20M	./backups
263M	./cache
18G	./log
4.0K	./tmp
29G	.
```

```
yamaguchi@fetch15 20:54:42 :/home $
$ sudo du -d1 -h
[sudo] password for yamaguchi: 
11G	./fetch
1000K	./s-kitagawa
8.6G	./furushchev
2.7G	./affonso-gui
3.6G	./yamaguchi
8.3G	./wkentaro
2.7G	./k-okada
24M	./iory
160K	./murooka
37G	.
```

furushchev, 2017-11-19:
var logにデータたまり過ぎなんじゃないかな。あとfetchユーザも心持ち減らせそう？

708yamaguchi, 2017-11-20:
とりあえず、
@708yamaguchi と @Affonso-Gui 
のユーザの中の余分なファイルを消しました。

```
yamaguchi@fetch15 11:47:26 :/home $
$ sudo du -d1 -h
11G	./fetch
1000K	./s-kitagawa
8.6G	./furushchev
321M	./affonso-gui
1.4G	./yamaguchi
8.3G	./wkentaro
2.7G	./k-okada
24M	./iory
160K	./murooka
32G	.
```

```
yamaguchi@fetch15 11:48:21 :/home $
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7.8G  4.0K  7.8G   1% /dev
tmpfs           1.6G  1.4M  1.6G   1% /run
/dev/sda2        94G   80G  9.3G  90% /
none            4.0K     0  4.0K   0% /sys/fs/cgroup
none            5.0M     0  5.0M   0% /run/lock
none            7.8G   68K  7.8G   1% /run/shm
none            100M   12K  100M   1% /run/user
/dev/sda1       511M  3.4M  508M   1% /boot/efi
overflow        1.0M  1.0M     0 100% /tmp
```

k-okada, 2017-11-20:
遅くなりました．

まずは，

https://askubuntu.com/questions/36111/whats-a-command-line-way-to-find-large-files-directories-to-remove-and-free-up

を参考に，

sudo du -aBM 2>/dev/null | sort -nr | head -n 10

としてみるんでしょうか．

furushchev, 2017-11-20:
@k-okada こちらこそ遅れましたが，
@708yamaguchi と話していて，mongodbのデータが大きいことに気がついたので，ひとまずmuscaの方へデータベースをreplicateしました．

```bash
# muscaで実行
mongodump -v -h 133.11.216.217 -d jsk_robot_lifelog -c fetch15  # データ吸い出し
mongorestore -v --db jsk_robot_lifelog mongodump/jsk_robot_lifelog  # muscaのmongodbへロード
```

```bash
# fetch15で実行
$ mongo localhost/jsk_robot_lifelog
$ db.dropDatabase()
$ exit
```

```bash
furushchev@fetch15:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7.8G  4.0K  7.8G   1% /dev
tmpfs           1.6G  1.4M  1.6G   1% /run
/dev/sda2        94G   69G   21G  77% /
none            4.0K     0  4.0K   0% /sys/fs/cgroup
none            5.0M     0  5.0M   0% /run/lock
none            7.8G   80K  7.8G   1% /run/shm
none            100M   40K  100M   1% /run/user
/dev/sda1       511M  3.4M  508M   1% /boot/efi
```

```bash
furushchev@fetch15:/var $ sudo du -d1 -h .
[sudo] password for furushchev: 
3.5G	./lib
4.0K	./mail
4.0K	./crash
4.0K	./opt
4.0K	./metrics
4.0K	./local
68K	./spool
1.5M	./www
20M	./backups
263M	./cache
16G	./log
4.0K	./tmp
19G	.
```