jsk-ros-pkg/jsk_robot #860: /head_camera/depth_registered/hw_registered/image_rect/compressedDepthをsubscribeするとerrorが出る

furushchev, 2017-11-01:
```
fetch@fetch15:~$ sudo start robot
fetch@fetch15:~$ rostopic hz /head_camera/depth_registered/hw_registered/image_rect    
subscribed to [/head_camera/depth_registered/hw_registered/image_rect]
# なにもでない
^Cfetch@fetch15:~/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetchtartup$ sudo tail -f /var/log/upstart/robot.log
ERROR] [1509579195.613769061]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509579197.615442869]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509579199.620944051]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509579201.652954617]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509579203.655143971]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509579205.657322027]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509579207.659404807]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509579209.661157313]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509579211.663491045]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509579213.665408696]: Expected data of type [16UC1], got [32FC1]
```

`rostopic hz`をkillしたあとも出続ける

furushchev, 2017-11-01:
関連ノードを`image_rect_raw`を使うように変更

k-okada, 2017-11-01:
どこかのノードがうまくデータタイプを扱えていないようですね．どこでしょう？ノード名がわかれば @wkentaro がこの辺はいろいろimage_pipeline, cv_bridge で直していたと思います．

knorth55, 2017-11-02:
`image_transport/Republish`をつかってみるのはどうでしょうか？

furushchev, 2017-11-02:
@k-okada ↓のあたりが怪しそうです．
https://github.com/ros-drivers/rgbd_launch/blob/2.1.2/launch/includes/depth_registered.launch.xml#L83-L89

参考として，fetchのopt以下の各packageのversion:

```bash
$ rosversion rgbd_launch
2.1.2
$ rosversion depth_image_proc
1.12.19
$ rosversion cv_bridge
1.11.13
```

（手元のlaptopに比べると1,2バージョン古そうです．）
もう少し調べて見ます．

@knorth55 image_rectの普通のやつからrepublishしてもう１つcompressedDepthを出すということかな？
確かに，もう少し頑張ってどうしてもダメそうならその手段も考えてみます．
ありがとう！

knorth55, 2017-11-02:
`jsk_topic_tools/Relay`をつかったことがないので知らないのですが同じことをやっているのなら`image_transport`のほうが信頼なるのかと思っただけです。

furushchev, 2017-11-02:
ログを詳細にしてみました．

```bash
export ROSCONSOLE_FORMAT='[${severity}] [${time}] [${node}:${logger}]: ${message}'
su ros -c ". /opt/ros/indigo/setup.bash && roslaunch /etc/ros/indigo/robot.launch --wait
[ERROR] [1509602288.509145184] [/head_camera/head_camera_nodelet_manager:ros.depth_image_proc./head_camera/depth_registered_hw_metric_rect]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509602290.510486702] [/head_camera/head_camera_nodelet_manager:ros.depth_image_proc./head_camera/depth_registered_hw_metric_rect]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509602292.512553336] [/head_camera/head_camera_nodelet_manager:ros.depth_image_proc./head_camera/depth_registered_hw_metric_rect]: Expected data of type [16UC1], got [32FC1]
[ERROR] [1509602294.514808475] [/head_camera/head_camera_nodelet_manager:ros.depth_image_proc./head_camera/depth_registered_hw_metric_rect]: Expected data of type [16UC1], got [32FC1]
```

furushchev, 2017-11-02:
@wkentaro くんの修正が見つかりました．
https://github.com/ros-perception/image_pipeline/blob/indigo/depth_image_proc/CHANGELOG.rst#11220-2017-04-30
たしかにこれで治ってそうですが，fetch内部は古いrepositoryになっているので，

```bash
$ rosversion depth_image_proc
1.12.19
```

`/etc/init/robot.conf`の方も治す必要がありそうですね．．．
@k-okada 次の手段が考えられそうですが，どうするのが良さそうでしょうか？

1. `depth_image_proc`の1.12.20をdebで手動インストールする
2. `/etc/init/robot.conf`にも`/home/fetch/ros/indigo/devel/setup.bash`をソースするようにする
3. `/etc/init/robot.conf`用に新しくdepth_image_procだけのcatkin workspaceを作って，それをソースするようにする
4. その他

knorth55, 2017-11-08:
@furushchev これは解決しましたか？

furushchev, 2017-11-08:
@knorth55 いいえ，直近で何か困っていそうですか？

@k-okada 直前のコメントについて，depth_image_procだけバージョンを手動で上げつつ，サポートの方にissueを投げてみようと思っていますが，良いでしょうか？

furushchev, 2017-11-09:
サポートに聞いてみました。

furushchev, 2017-11-24:
サポートからは手元でcatkin workspaceを作ってコンパイルして欲しい旨返信が来ました．
（つまり上のコメントでいう2もしくは3番）

@k-okada 3のほうが安全な気がしますので，この方法で対応したいと思います．

furushchev, 2017-11-24:
```
I created a catkin workspace just for `robot.conf` by the commands as followings:

    mkdir -p ~/ros/indigo_robot/src
    cd ~/ros/indigo_robot/src
    wstool init
    rosinstall_generator --tar depth_image_proc --rosdistro indigo | wstool merge -
    wstool up
    cd ..
    rosdep install --from-paths src -i -r -n -y
    source /opt/ros/indigo/setup.bash
    catkin config --init --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebugInfo
    catkin build

And then changed `/etc/init/robot.conf`:

    env ROS_LOG_DIR=/var/log/ros

    start on roscore_is_up
    stop on roscore_is_down

    respawn

    env ROSCONSOLE_FORMAT='[${severity}] [${time}] [${node}:${logger}]: ${message}'

    script
    #    exec su ros -c ". /opt/ros/indigo/setup.bash && roslaunch /etc/ros/indigo/robot.launch --wait"
        exec su ros -c ". /home/fetch/ros/indigo_robot/devel/setup.bash && roslaunch /etc/ros/indigo/robot.launch --wait"
    end script

Then restarted roscore: `sudo stop roscore && sudo start roscore`.
And now I confimed depth `image_rect` topic is published.
```

closed 2017-11-24 by furushchev