jsk-ros-pkg/jsk_robot #863: fetch15 doesn't move in unsafe mode

knorth55, 2017-11-02:
`fetch15` does not move in unsafe mode.
In safe mode, it moves correctly.

knorth55, 2017-11-02:
cc. @708yamaguchi

furushchev, 2017-11-02:
@knorth55 @708yamaguchi If you see the same error, please could you trace which node blocks joy command to velocity?
I believe @708yamaguchi made a diagram for the pipeline.

708yamaguchi, 2017-11-06:
@furushchev 
L1とジョイスティックを押した時(safe mode)は`/teleop/cmd_vel/unsafe`に指令が送られていましたが、
L1とR2とジョイスティックを押した時(unsafe mode)は、`/teleop/cmd_vel/unsafe`に指令が送られていませんでした。（ジョイスティックを動かしても、`linear`も`angular`もすべて0のまま変化しませんでした。）

以前僕が作ったダイアグラムは下にあります。
https://github.com/jsk-ros-pkg/jsk_robot/pull/801#issuecomment-314641920

furushchev, 2017-11-06:
> L1とR2とジョイスティックを押した時(unsafe mode)は、/teleop/cmd_vel/unsafeに指令が送られていませんでした。

「指令が送られていない」＝「全要素が0のメッセージがpublishされている」or 「何もpublishされていない」のどちらでしょう？
また，
R2を押した時と話した時で，`rostopic echo -n5 /teleop/cmd_vel/safe`した結果と，
`rosnode info /safe_teleop_base`の結果も教えてくれると嬉しいです．

708yamaguchi, 2017-11-07:
> 「指令が送られていない」＝「全要素が0のメッセージがpublishされている」or 「何もpublishされていない」のどちらでしょう？

全要素が0のメッセージがpublishされています。

R2を押さなかった時は、
```
$ rostopic echo -n5 /teleop/cmd_vel/safe
linear: 
  x: 0.357392773964
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.364600792949
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.364600792949
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.359637894631
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.366639126186
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
```
```
$ rosnode info /safe_teleop_base
--------------------------------------------------------------------------------
Node [/safe_teleop_base]
Publications: 
 * /safe_teleop_base/local_costmap/costmap [nav_msgs/OccupancyGrid]
 * /safe_teleop_base/local_costmap/inflater/parameter_updates [dynamic_reconfigure/Config]
 * /safe_teleop_base/local_costmap/inflater/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /safe_teleop_base/local_costmap/obstacles/parameter_updates [dynamic_reconfigure/Config]
 * /safe_teleop_base/local_costmap/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rosout [rosgraph_msgs/Log]
 * /safe_teleop_base/local_costmap/costmap_updates [map_msgs/OccupancyGridUpdate]
 * /safe_teleop_base/local_costmap/obstacles/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /safe_teleop_base/local_costmap/footprint [geometry_msgs/PolygonStamped]
 * /safe_teleop_base/user_plan [nav_msgs/Path]
 * /safe_teleop_base/local_costmap/obstacles/clearing_endpoints [sensor_msgs/PointCloud]
 * /safe_teleop_base/local_plan [nav_msgs/Path]
[... 251 lines cut ...]
    * to: /robot_state_publisher (http://fetch15:38068/)
    * direction: inbound
    * transport: TCPROS
 * topic: /base_scan
    * to: /sick_tim_filter (http://fetch15:38833/)
    * direction: inbound
    * transport: TCPROS
 * topic: /head_camera/depth_downsample/camera_info
    * to: /head_camera/head_camera_nodelet_manager (http://fetch15:33986/)
    * direction: inbound
    * transport: TCPROS
 * topic: /head_camera/depth_downsample/image_raw
    * to: /head_camera/head_camera_nodelet_manager (http://fetch15:33986/)
    * direction: inbound
    * transport: TCPROS
 * topic: /odom
    * to: /robot_driver (http://fetch15:49378/)
    * direction: inbound
    * transport: TCPROS
 * topic: /teleop/cmd_vel/unsafe
    * to: /teleop (http://fetch15:45482/)
    * direction: inbound
    * transport: TCPROS
```
のような結果が出ました。

furushchev, 2017-11-07:
ありがとう．
あれ，おかしいな，R2を押していない時には`/teleop/cmd_vel/safe`はpublishされないのが正解なきがするのだけど．．．

furushchev, 2017-11-07:
あ，勘違いしていました．これで合っていますね．
とすると，R2を押した時に`/teleop/cmd_vel/unsafe`と`/unsafe_vel_mux/selected`，`/teleop/cmd_vel`はどうなっているでしょうか？

708yamaguchi, 2017-11-08:
L1, R2を押した時、`/teleop/cmd_vel/unsafe`、`/unsafe_vel_mux/selected`、`/teleop/cmd_vel`は順に以下のようになっていました。
```
$ rp echo /teleop/cmd_vel/unsafe
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
```

```
$ rostopic echo /unsafe_vel_mux/selected
data: /teleop/cmd_vel/unsafe
---
```

```
$ rostopic echo /teleop/cmd_vel
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---
```

furushchev, 2017-11-08:
うーん，`/teleop/cmd_vel/unsafe`の時点で0になっているとなると，`fetch_teleop`ノードがおかしいのかな．もしかしたらボタンアサインが変わって，何かバッティングしているのかも．

furushchev, 2017-11-08:
fetch_teleopのバージョンがfetchユーザだけ0.7.13 (ほかは0.7.9)だったので，オリジナルに戻してみたところ，うまく動くようになった気がします．

(今はこれで良いですが，将来的にアサインが変わって動かなくなりそうなので，別途アサインを考えておいたほうが良さそうです．)

knorth55, 2017-11-08:
これは #864 を`fetch_moveit_config`のみにすることで解決します

knorth55, 2017-11-08:
#864で解決しました。

closed 2017-11-08 by knorth55