jsk-ros-pkg/jsk_robot #880: fetch : go-pos-unsafe が動かない。

koga-yu, 2017-12-20:
。トピック名から違っている気がします。
```
--- a/jsk_fetch_robot/fetcheus/fetch-interface.l
+++ b/jsk_fetch_robot/fetcheus/fetch-interface.l
@@ -169,7 +173,7 @@
 
 ;; interface for simple base actions
 (defmethod fetch-interface
-  (:send-cmd-vel-raw (x d &key (topic-name "/base_controller/command"))
+  (:send-cmd-vel-raw (x d &key (topic-name "/cmd_vel"))
     (when (send self :simulation-modep)
       (return-from :send-cmd-vel-raw t))
```

knorth55, 2017-12-20:
自分が正しいと思うものをPR送ってみましょう！
（先に試せているとなおよし）

k-okada, 2017-12-21:
なぜ変更になったかが問題です． @furushchev は，以前は動いていたはずだ，fetchの体内をfetch.com -> ros.org に変更した副作用かという話もあります．

knorth55, 2017-12-21:
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_fetch_robot/jsk_fetch_startup/launch/fetch_teleop.xml#L16-L18
http://docs.fetchrobotics.com/api_overview.html#base-interface

これらを見る限り`base_controller/command`と`cmd_vel`は同じ意味に思えます。
ただdocは`cmd_vel`を使うことが推奨されているようにも読めます。
なぜこの変更で動かない時があるのかがわかりません。

@k-okada 
@koga-yu に聞いてみたところ`(send *ri* :state :odom :pose)`が`nil`で帰ってくるのが問題ということですが、これはこの変更で解決するのでしょうか？
正直`move_base`がどう動いているのかがよくわかっていないです。

knorth55, 2017-12-21:
`topic-name`は@k-okada先生が正しいとおもいます。
`cmd_vel`におくることでやばい動きをした場合にもコントローラから介入することができます。

`(send *ri* :state :odom :pose)`が`nil`かえってくる問題はhttps://github.com/jsk-ros-pkg/jsk_pr2eus/pull/333 で直しました。
問題は #843 で常に`:wait-until-update`のキーが入るようになったのが原因です。
古田さんはこの変更の前にやっていて動いていたようです。

knorth55, 2018-01-10:
このissueは https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/333 で治ります。