jsk-ros-pkg/jsk_robot #881: [fetch] 無線の調子が悪い

708yamaguchi, 2017-12-21:
fetchの無線の調子が悪かったのですが、5秒おきにネットワークの接続を確認するプログラムを起動時に実行するようにしました。
fetch内部で
`sudo tail -f /var/log/upstart/network-monitor.log`
するとプログラムの様子が確認できます。
以下がプログラムへのリンクになります。
https://drive.google.com/open?id=1p8oa9b5HYVzQ6mBcc8dYM5-hZzM2FOGE

#880

708yamaguchi, 2017-12-26:
たった今`fetch`の無線の調子が悪かったので報告します。
症状としては、`ping`も`ssh`も出来ない状態でした。

この状態でfetchに直接モニタとキーボードをつないで
`sudo tail -f /var/log/upstart/network-monitor.log`
としたところ、以下のようなメッセージが出ていました。
```
fetch@fetch15:~$ sudo tail -f /var/log/upstart/network-monitor.log
[sudo] password for fetch: 
Error: Timeout 10 sec expired.
Trying to connect to 'sanshiro 190' (sanshiro)
Error: Unknown connection: sanshiro 190.
Trying to connect to 'sanshiro 1378' (sanshiro)
Error: Unknown connection: sanshiro 1378.
Trying to connect to 'sanshiro 1456' (sanshiro)
Error: Unknown connection: sanshiro 1456.
Trying to connect to 'sanshiro 1303' (sanshiro)
Error: Unknown connection: sanshiro 1303.
Trying to connect to 'sanshiro 586' (sanshiro)
Error: Unknown connection: sanshiro 586.
Trying to connect to 'sanshiro 141' (sanshiro)
Error: Unknown connection: sanshiro 141.
Trying to connect to 'sanshiro 50' (sanshiro)
Error: Unknown connection: sanshiro 50.
Trying to connect to 'sanshiro 1042' (sanshiro)
Error: Unknown connection: sanshiro 1042.
Trying to connect to 'sanshiro 1321' (sanshiro)
Error: Unknown connection: sanshiro 1321.
Trying to connect to 'sanshiro 118' (sanshiro)

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/1465: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/475: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/559: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/544: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/23: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/1578: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/1647: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/93: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/1605: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/1568: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/183: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/1219: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/297: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/681: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/1380: The maximum number of pending replies per connection has been reached

** (process:32725): WARNING **: Could not create object for /org/freedesktop/NetworkManager/Settings/148: The maximum number of pending replies per connection has been reached
```

`fetch`を再起動すると`ping`も`ssh`も通るようになりました。

@k-okada

k-okada, 2017-12-27:
良いバグレポートですね．
次はこの瞬間に自分のノートPCとか携帯からsanshiroに繋がるか調べてみましょう．

708yamaguchi, 2018-01-05:
@k-okada 

岡田先生にいただいたポケットルータをfetchにとりつけて無線LANの通信をしようという話だったのですが、 @mmurooka さん、 @iory さん、 @YutoUchimi といろいろ試してもうまくいきませんでした。もともと無線LAN子機としての機能がなかったように思われます。

その後、無線LAN用のUSBアダプタをfetchに取り付けてみたのですが、通信はできたものの、IPアドレスが変わってしまうことがありました。

新しい無線LAN子機を使うかUSBアダプタで対応するか、どちらがいいでしょうか。また、勉強のためにIPアドレスを固定する方法を教えていただきたいです。

furushchev, 2018-01-05:
これは昨日僕が渡した元PR2用のものかな？
どれを試したかな？
また試したのは具体的にどういうことをしたのかな？
僕も昨日の時点ではPR2の使い方（無線APでの使用）しか考えていませんでしたが、
渡したうちエレコムのは説明書をみるとfetchでの使い方に対応してるように見えます。

http://www2.elecom.co.jp/search/link/search.asp?kataban=WRH-733GBK&link_type=2
p44

また要らないものはPR2の予備として使いたいので、僕の机の上に戻しておいてくれると助かります。

708yamaguchi, 2018-01-06:
古田さんに手伝ってもらい、上のリンクにもある
`ELECOM WRH-733GBK`
というルータを無線LAN子機として使用することが出来ました。
ありがとうございました。

現在の通信速度としては、情報理工の有線から`fetch`の無線に`iperf`して`40Mbits/sec`くらい出ています。