jsk-ros-pkg/jsk_robot #883: [fetch] :go-grasp の :wait がきかないことがある

708yamaguchi, 2018-01-07:
`fetch`で`:go-grasp`を使うと、グリッパの初期位置によっては`:wait t`がきかず、すぐに関数から抜けてしまうことがあります。どうすれば常に`:go-grasp`の`:wait t`がきくようになるでしょうか？

## うまくいかない場合
```
(send *ri* :go-grasp :pos 0.1 :wait t) ;; グリッパを開く
```
をしてグリッパを開いた状態にしておいて、
```
(send *ri* :go-grasp :pos 0 :wait t) ;; グリッパを閉じる
```
とすると、`:wait t`がきかず、すぐに返り値が返ってきてしまいます。

`/gripper_controller/gripper_action/result`も、`(send *ri* :go-grasp :pos 0 :wait t)`をよんだ瞬間にトピックが来ます。
```
$ rostopic echo /gripper_controller/gripper_action/result
header: 
  seq: 26996
  stamp: 
    secs: 1515314191
    nsecs: 204861430
  frame_id: ''
status: 
  goal_id: 
    stamp: 
      secs: 1515314190
      nsecs: 704733467
    id: 1515314190697195276_/fetch_1515312215627196014_5117_/gripper_controller/gripper_action_125
  status: 3
  text: ''
result: 
  position: 0.092070043087
  effort: 40.09375
  stalled: True
  reached_goal: False
---
```

## うまくいく場合
```
(send *ri* :go-grasp :pos 0.09 :wait t) ;; グリッパをほぼ開く
```
のように、グリッパを完全には開かない状態にしておいて、
```
(send *ri* :go-grasp :pos 0 :wait t) ;; グリッパを閉じる
```
とすると、グリッパが閉じるまで関数から抜けず、`/gripper_controller/gripper_action/result`もグリッパが閉じてからトピックが来ます。（求めている動作）
```
$ rostopic echo /gripper_controller/gripper_action/result
header: 
  seq: 27004
  stamp: 
    secs: 1515314444
    nsecs:  18685669
  frame_id: ''
status: 
  goal_id: 
    stamp: 
      secs: 1515314441
      nsecs: 938582328
    id: 1515314441931000948_/fetch_1515312215627196014_5117_/gripper_controller/gripper_action_133
  status: 3
  text: ''
result: 
  position: 0.0129394382238
  effort: 40.0
  stalled: True
  reached_goal: False
---
```

k-okada, 2018-01-07:
すぐに関数から抜ける場合の実機の挙動はどうなっていますか．ロボットのグリッパは閉じるのかな？途中でとまるのかな．

```
result:
  position: 0.092070043087
  effort: 40.09375
  stalled: True
  reached_goal: False
```
みるとstalled true and reached_goal: false
だからちゃんと閉じたと思っていないようにみえるけどどうかな．effortを５０とか６０にするとどうなるかな．

708yamaguchi, 2018-01-07:
effortが40ですぐに関数から抜ける場合でも、実機のグリッパは最後まで閉じていました。

下のようにeffortを50にすると、すぐに関数から抜けるということはなくなりました。（60でも問題はありませんでした。）
```
(send *ri* :go-grasp :pos 0.1 :effort 50 :wait t)
(send *ri* :go-grasp :pos 0 :effort 50 :wait t)
```

これをなおすために、`:go-grasp`のデフォルトのeffortを40から50にするプルリクエストをつくろうと思うのですがどうでしょうか？

k-okada, 2018-01-07:
これグリッパにセンサつけての実験？
デフォルトの状態だと40で十分ということはないかな？

目安としてはデフォルトで

500mlのペットボトルを把持できる
でも、変形しない

とかどうでしょう

708yamaguchi, 2018-01-07:
今までの実験は僕の作ったセンサをのせた場合のものでした。
グリッパをデフォルトの状態に戻して実験しなおしましたが、やはりeffort 40では同じ症状が出ました。
この状態でeffortを50にしたところ、関数からすぐに抜けるということはなくなりました。
また、effort 50だと500mlのペットボトルを潰すことはなかったので、effortは50にしようと思います。

自分のメモのために残しておくのですが、`/gripper_controller/gripper_action/result`のtopicが出る条件はこのあたりです。
 https://github.com/fetchrobotics/robot_controllers/blob/74970680716f8af29c5ff3e1e8a76ea1089640ff/robot_controllers/src/parallel_gripper.cpp#L223-L275