jsk-ros-pkg/jsk_robot #892: 左腕の手先ジョイントのゼロ点がずれている

furushchev, 2018-01-19:
デモをやっていて気が付きましたが、
左腕の最終軸のゼロ点がいつもよりずれている気がします。

正しいジョイント位置（右腕）

[image]

ずれているジョイント位置（左腕）

[image]

`roslaunch pr2_bringup pr2_recalibrate.launch`を行っても同じ位のズレがあります。

[image]
[image]

TFではずれていないので、エンコーダ関連の問題かなと思っています。

furushchev, 2018-01-19:
@k-okada 過去にこういうような問題はありましたでしょうか？

furushchev, 2018-01-20:
今日キャリブレーションをしなおしてみようと思い、再起動したところ、治ってしまったように思います。
また同じ状況になったら再度openしようと思いまうs。

furushchev, 2018-02-26:
やっぱり実験をしていくとずれていく気がします。

`rqt_pr2_dashboard`を見ると、

[image]

のように腕のモータにワーニングが出ています。
https://github.com/PR2/pr2_ethercat_drivers/blob/e9e6a6bbec14cdbb3bd5675aa3eeef32939e597c/ethercat_hardware/src/motor_model.cpp#L382
の部分だと思いますが、
どうしたら治るのかまではちょっと調べても、わかりませんでした。
このときの正常なものとワーニングがでた軸のモータのdiagnosticsデータは
https://gist.github.com/furushchev/22535ff46fee39583123fd3fd7bbfac2
のような感じです。

何回かキャリブレーションしなおすとマシになる（完全には治っていない）ので、しばらく様子をみます。

iory, 2019-04-22:
pr1012でこの問題がまた起きました。
l_wrist_roll_jointが30度ほどずれています。

iory, 2019-04-22:
> 何回かキャリブレーションしなおすとマシになる（完全には治っていない）ので、しばらく様子をみます。

こちらで何度かrecalibrateし直すと元には戻りました。