jsk-ros-pkg/jsk_robot #894: [baxtereus] default :self-collision-check returns always t

knorth55, 2018-02-06:
I try to use `:self-collision-check` method, and it returns always some pairs.
one big reason is that torso link is too big.
[image]

```lisp
1.irteusgl$ (setq *baxter* (instance baxter-robot :init))
#<baxter-robot #X780f9f0 baxter  0.0 0.0 0.0 / 0.0 0.0 0.0>
2.irteusgl$ (objects (list *baxter*))
;; (make-irtviewer) executed
(#<baxter-robot #X780f9f0 baxter  0.0 0.0 0.0 / 0.0 0.0 0.0>)
3.irteusgl$ (send *baxter* :self-collision-check)
PQP Error! EndModel() called on model with no triangles
PQP Error! EndModel() called on model with no triangles
PQP Error! EndModel() called on model with no triangles
PQP Error! EndModel() called on model with no triangles
PQP Error! EndModel() called on model with no triangles
PQP Error! EndModel() called on model with no triangles
PQP Error! EndModel() called on model with no triangles
PQP Error! EndModel() called on model with no triangles
((#<bodyset-link #X7597790 torso  0.0 0.0 0.0 / 0.0 0.0 0.0> . #<bodyset-link #X6f29070 right_upper_shoulder  64.027 -259.027 129.626 / -1.134 0.0 0.0>) (#<bodyset-link #X7597790 torso  0.0 0.0 0.0 / 0.00.0 0.0> . #<bodyset-link #X680c978 right_lower_shoulder  93.188 -321.563 399.976 / -1.134 -0.436 -1.571>) (#<bodyset-link #X7597790 torso  0.0 0.0 0.0 / 0.0 0.0 0.0> . #<bodyset-link #X66be1d0 left_upper_shoulder  64.027 259.027 129.626 / 1.134 0.0 0.0>) (#<bodyset-link #X7597790 torso  0.0 0.0 0.0 / 0.0 0.0 0.0> . #<bodyset-link #X69a7758 left_lower_shoulder  93.188 321.563 399.976 / 1.134 -0.436 -1.571>) (#<bodyset-link #X7597790 torso  0.0 0.0 0.0 / 0.0 0.0 0.0> . #<bodyset-link #X77b39c8 screen  182.7 0.0 686.0 / 1.571 1.499e-15 1.751>) (#<bodyset-link #X77f9550 collision_head_link_2  110.0 0.0 750.0 / 0.0 0.0 0.0> . #<bodyset-link #X7806e88 collision_head_link_1  110.0 0.0 750.0 / 0.0 0.0 0.0>) (#<bodyset-link #X68b7d18 right_upper_elbow_visual  134.171 -409.452 445.196 / -1.134 1.134 6.693e-12> . #<bodyset-link #X690c0c0 right_upper_elbow  132.256 -405.345 443.083 / -0.03 0.767 0.943>) (#<bodyset-link #X68b7d18 right_upper_elbow_visual  134.171 -409.452 445.196 / -1.134 1.134 6.693e-12> . #<bodyset-link #X67cff88 right_lower_elbow  282.406 -622.397 506.082 / -0.485 0.401 -0.886>) (#<bodyset-link #X67a3be0 right_upper_forearm_visual  354.083 -660.196 471.766 / 1.531 0.794 2.161> . #<bodyset-link #X677eeb0 right_upper_forearm  366.781 -666.893 465.687 / 1.314 0.49 2.029>) (#<bodyset-link #X67a3be0 right_upper_forearm_visual  354.083 -660.196 471.766 / 1.531 0.794 2.161> . #<bodyset-link #X677f0a8 right_lower_forearm  589.515 -774.638 355.419 / 1.129 0.56 0.365>) (#<bodyset-link #X677f3d8 right_wrist  631.558 -685.847 293.786 / 2.499 -0.307 2.162> . #<bodyset-link #X675de08 right_hand_camera  655.517-561.757 246.338 / -2.636 0.913 2.624>) (#<bodyset-link #X69a5400 left_upper_elbow_visual  134.171 409.452 445.196 / 1.134 1.134 6.693e-12> . #<bodyset-link #X6e68ee0 left_upper_elbow  132.256 405.345 443.083 / 0.03 0.767 -0.943>) (#<bodyset-link #X69a5400 left_upper_elbow_visual  134.171 409.452 445.196 / 1.134 1.134 6.693e-12> . #<bodyset-link #X6759d90 left_lower_elbow  282.406 622.397 506.082 / 0.4850.401 -2.256>) (#<bodyset-link #X675a9d8 left_upper_forearm_visual  354.083 660.196 471.766 / -1.531 0.794 -2.161> . #<bodyset-link #X66ec2e8 left_upper_forearm  366.781 666.893 465.687 / -1.314 0.49 -2.029>) (#<bodyset-link #X675a9d8 left_upper_forearm_visual  354.083 660.196 471.766 / -1.531 0.794 -2.161> . #<bodyset-link #X66ec6d8 left_lower_forearm  589.515 774.638 355.419 / -1.129 0.56 2.777>) (#<bodyset-link #X78c1058 left_wrist  631.558 685.847 293.786 / -2.499 -0.307 -2.162> . #<bodyset-link #X7906b20 left_hand_camera  642.678 568.861 227.346 / -0.505 -0.913 2.624>))
4.irteusgl$  pprint (mapcar #'(lambda (x) (list (send (car x) :name) (send (cdr x) :name))) (send *baxter* :self-collision-check))
(("torso" "right_upper_shoulder")
 ("torso" "right_lower_shoulder")
 ("torso" "left_upper_shoulder")
 ("torso" "left_lower_shoulder")
 ("torso" "screen")
 ("collision_head_link_2" "collision_head_link_1")
 ("right_upper_elbow_visual" "right_upper_elbow")
 ("right_upper_elbow_visual" "right_lower_elbow")
 ("right_upper_elbow_visual" "left_upper_forearm_visual")
 ("right_upper_elbow_visual" "left_upper_forearm")
 ("right_upper_elbow" "left_upper_forearm_visual")
 ("right_upper_elbow" "left_upper_forearm")
 ("right_lower_elbow" "left_upper_forearm_visual")
 ("right_lower_elbow" "left_upper_forearm")
 ("right_lower_elbow" "left_lower_forearm")
 ("right_upper_forearm_visual" "right_upper_forearm")
 ("right_upper_forearm_visual" "right_lower_forearm")
 ("right_wrist" "right_hand_camera")
 ("left_upper_elbow_visual" "left_upper_elbow")
 ("left_upper_elbow_visual" "left_lower_elbow")
 ("left_upper_forearm_visual" "left_upper_forearm")
 ("left_upper_forearm_visual" "left_lower_forearm")
 ("left_wrist" "left_hand_camera"))
nil
5.irteusgl$
```

knorth55, 2018-02-06:
current solution taught from @mmurooka is as below.

Not using
```
(send *baxter* :self-collision-check)
```

Use
```
(send *baxter* :self-collision-check :pairs (send *baxter* :collision-check-pairs :links (send *baxter* :links)))
```

k-okada, 2018-03-12:
May be this is why we created `baxter-robot-safe` class, but not sure because we do not have any commit message for this..... https://github.com/jsk-ros-pkg/jsk_robot/pull/315

closed 2019-02-02 by knorth55