jsk-ros-pkg/jsk_robot #903: PepperのEuslispをつかった動かし方について

M-Onitsuka, 2018-03-28:
Pepperを動かそうと考えています。
今のところ
[http://wiki.ros.org/pepper/Tutorials](url)
を参考に
```roslaunch pepper_bringup pepper_full_py.launch nao_ip:=<yourRobotIP> roscore_ip:=<roscore_ip>```
として
```
rostopic pub /cmd_vel geometry_msgs/Twist "linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: -0.2"
```
などとして動かしています。
Euslispなどからangle-vectorを送る方法はありますでしょうか。
また、上記launchが
```
================================================================================REQUIRED process [naoqi_joint_states-2] has died!
process has finished cleanly
log file: /home/m-onitsuka/.ros/log/d618a2ba-326f-11e8-8659-b808cf4bc648/naoqi_joint_states-2*.log
Initiating shutdown!
================================================================================
```
となって止まるのですが、原因がよくわかりません。

k-okada, 2018-03-28:
eusからの動かし方は以下です．
https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot/peppereus

```
 [naoqi_joint_states-2] has died!
```
は
１．メッセージを表示するようにする `--screen` をつけて実行　http://wiki.ros.org/roslaunch/Commandline%20Tools#Launch_syntax
２．gdbをつかってエラーが出ている行を表示http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

してみてください．

M-Onitsuka, 2018-03-28:
```--screen```をつけたところ
```
[E] 21486 python: KeyError: ('timestamp',)
```
といった表示がずっとされるようになりました。

またpeppereusをcatkin buildをしたところ
```
/home/m-onitsuka/catkin_ws/src/naoqi_driver/include/../src/converters/../helpers/driver_helpers.hpp:26:41: fatal error: naoqi_bridge_msgs/SetString.h: そのようなファイルやディレクトリはありません
 #include <naoqi_bridge_msgs/SetString.h>
```
と出てきて、何かしらをビルドしていないのだと思うのですが、こちらも対処法がわかりません。

kochigami, 2018-03-28:
> 何かしらをビルドしていないのだと思うのですが、こちらも対処法がわかりません。

naoqi_bridge_msgsのバージョンが古いのではないかと思うのですが，ご都合の良い時に確認していただけますか．0.0.8になっていればあるはずです．
（もし古いバージョン説が合っていたら，どうして新しいバージョンが取ってこれないのかは明日考えます．）

KeyErrorについては，
私はnaoqi_driver_pyを使っていなくて，naoqi_driverを使っているのですが，
お力になれるかは分かりませんが，同じエラーが出るか明日確認してみます．

M-Onitsuka, 2018-03-28:
@kochigami さん
バージョンはpackage.xmlに記載されているものでよろしいでしょうか。
```/opt/ros/indigo/share/naoqi_bridge_msgs/package.xml```を確認したところ、たしかにバージョンは0.06でした。

kochigami, 2018-03-29:
まずpepper_bringup_pyについて：

私は動きました・・・
ubuntu 14.04, ros indigo, pepper_bringup: 0.1.10
nao_ipは`.bashrc`に書いています．（参照：https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/jsk_pepper_startup/README.md）

ちょっと思ったのは，ペッパーの体内でROSのプログラム実行しようとしているのかな，と思いました．
もし，そうだったら，手元のノートパソコンでやってみてください．もし違ったらごめんなさい！
naoqi_driver_py/naoqi_logger.py がエラーの元かな，と思いました．`timestamp`を取ってこようとしていますね．

ノートパソコンでの実行結果です．
roslaunch pepper_bringup pepper_full_py.launch 
```
... logging to /home/kochigami/.ros/log/be0a3e9a-330f-11e8-b56e-507b9d20c4fb/roslaunch-kochigami-ThinkPad-T450-10618.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.77:39241/

SUMMARY
========

PARAMETERS
 * /naoqi_joint_states/use_odometry: True
 * /pepper_robot/camera/camera_nodelet_manager/num_worker_threads: 4
 * /pepper_robot/camera/depth/camera/calibration_file_bottom: file:///home/koch...
 * /pepper_robot/camera/depth/camera/calibration_file_top: file:///home/koch...
 * /pepper_robot/camera/depth/camera/color_space: 17
 * /pepper_robot/camera/depth/camera/frame_rate: 15
 * /pepper_robot/camera/depth/camera/resolution: 1
 * /pepper_robot/camera/depth/camera/source: 2
 * /pepper_robot/camera/depth/camera/use_ros_time: True
 * /pepper_robot/camera/depth_rectify_depth/interpolation: 0
 * /pepper_robot/camera/depth_registered_rectify_depth/interpolation: 0
 * /pepper_robot/camera/front/camera/calibration_file_bottom: file:///home/koch...
 * /pepper_robot/camera/front/camera/calibration_file_top: file:///home/koch...
 * /pepper_robot/camera/front/camera/color_space: 13
 * /pepper_robot/camera/front/camera/frame_rate: 15
 * /pepper_robot/camera/front/camera/resolution: 1
 * /pepper_robot/camera/front/camera/source: 0
 * /pepper_robot/camera/front/camera/use_ros_time: True
 * /pepper_robot/pose/pose_manager/poses/init/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/init/positions: [0.0, 0.0, 1.39, ...
 * /pepper_robot/pose/pose_manager/poses/init/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/poses/zero/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/zero/positions: [0.0, 0.0, 0.0, 0...
 * /pepper_robot/pose/pose_manager/poses/zero/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/xap: /home/kochigami/c...
 * /pepper_robot/sonar/back/naoqi_sonar/frame_id: SonarBack_frame
 * /pepper_robot/sonar/back/naoqi_sonar/memory_key: Device/SubDeviceL...
 * /pepper_robot/sonar/back/naoqi_sonar/sonar_rate: 10
 * /pepper_robot/sonar/front/naoqi_sonar/frame_id: SonarFront_frame
 * /pepper_robot/sonar/front/naoqi_sonar/memory_key: Device/SubDeviceL...
 * /pepper_robot/sonar/front/naoqi_sonar/sonar_rate: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /pepper_robot/camera/depth/
    camera (pepper_sensors_py/camera.py)
  /pepper_robot/camera/front/
    camera (pepper_sensors_py/camera.py)
  /pepper_robot/sonar/back/
    naoqi_sonar (naoqi_sensors_py/sonar.py)
  /pepper_robot/pose/
    pose_controller (naoqi_pose/pose_controller.py)
    pose_manager (naoqi_pose/pose_manager.py)
  /
[... 287 lines cut ...]
/pepper_robot/pose/get_life_state
/pepper_robot/pose/joint_angles
/pepper_robot/pose/joint_angles_action/cancel
/pepper_robot/pose/joint_angles_action/feedback
/pepper_robot/pose/joint_angles_action/goal
/pepper_robot/pose/joint_angles_action/result
/pepper_robot/pose/joint_angles_action/status
/pepper_robot/pose/joint_stiffness
/pepper_robot/pose/joint_stiffness_trajectory/cancel
/pepper_robot/pose/joint_stiffness_trajectory/feedback
/pepper_robot/pose/joint_stiffness_trajectory/goal
/pepper_robot/pose/joint_stiffness_trajectory/result
/pepper_robot/pose/joint_stiffness_trajectory/status
/pepper_robot/pose/joint_trajectory/cancel
/pepper_robot/pose/joint_trajectory/feedback
/pepper_robot/pose/joint_trajectory/goal
/pepper_robot/pose/joint_trajectory/result
/pepper_robot/pose/joint_trajectory/status
/pepper_robot/sonar/back/sonar
/pepper_robot/sonar/front/sonar
/rosout
/rosout_agg
/tf
/tf_static
```

kochigami, 2018-03-29:
naoqi_bridge_msgsについて：

https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/jsk_pepper_startup/README.md
を参照して，最初から環境構築してみていただけますか．
（READMEどこを読めばよいか，分かりにくくてすみません．環境構築部分は別に分けて更新しようと思います．）

http://packages.ros.org/ros/ubuntu/pool/main/r/ros-indigo-naoqi-bridge-msgs/
を見ると，0.0.8が現れているので，インストールされる気がします．
これで駄目だったら，また教えていただけますか．

M-Onitsuka, 2018-04-09:
ros-indigo-naoqi-bridge-msgsをpurgeして再度installしたところ、無事peppereusのビルドが成功しました
また
```
wstool set jsk-ros-pkg/jsk_pr2eus https://github.com/jsk-ros-pkg/jsk_pr2eus.git
catkin build
```
としてpr2eusをビルドしたところ、roseus上でpeppe-interface.lがload出来ました。