jsk-ros-pkg/jsk_robot #909: PR1040no

furushchev, 2018-04-13:


furushchev, 2018-04-13:
交換しました。劣化したバッテリが交換されてFullCapacityが回復したことを確認しました。

```
                   Battery Name|     CycleCount|   FullCapacity|     HardwareID|ManufactureDate|RemainingCapacity
/Power System/Smart Battery 0.0|             30|           6433|           9679|       3/6/2015|              0
/Power System/Smart Battery 0.1|             31|           6215|           9662|       3/6/2015|           6131
/Power System/Smart Battery 0.2|             29|           6629|           9681|       3/6/2015|           6552
/Power System/Smart Battery 0.3|             30|           6337|           9683|       3/6/2015|           6261
/Power System/Smart Battery 1.0|              1|           6766|          24829|     11/27/2017|           1171
/Power System/Smart Battery 1.1|              1|           6744|          24830|     11/27/2017|           1179
/Power System/Smart Battery 1.2|              1|           6731|          24826|     11/27/2017|           1202
/Power System/Smart Battery 1.3|              1|           6795|          24850|     11/27/2017|           1195
/Power System/Smart Battery 2.0|             28|           6562|           9680|       3/6/2015|           6521
/Power System/Smart Battery 2.1|             28|           6295|           9670|       3/6/2015|           6235
/Power System/Smart Battery 2.2|             28|           6535|           8771|      1/21/2015|           6533
/Power System/Smart Battery 2.3|             27|           6391|           9674|       3/6/2015|           6391
/Power System/Smart Battery 3.0|             32|           6201|           9652|       3/6/2015|           6144
/Power System/Smart Battery 3.1|             31|           6112|           9663|       3/6/2015|           6029
/Power System/Smart Battery 3.2|             33|           6171|           8775|      1/21/2015|           6101
/Power System/Smart Battery 3.3|             30|           6335|           9668|       3/6/2015|           5911
```

closed 2018-04-13 by furushchev

takayuki5168, 2018-06-13:
@furushchev what command did you use when displaying battery infomation

furushchev, 2018-06-13:
```bash
rossetmaster pr1040
rosrun jsk_tools battery_capacity_summary.py
```