jsk-ros-pkg/jsk_robot #911: Navigation unsafe mode faster than safe mode?

Affonso-Gui, 2018-04-13:
When testing fetch, I got the impression that unsafe navigation is faster than safe one.

For safety reasons, the opposite would be desired.

k-okada, 2018-04-15:
I agree unsafe mode should be slower than normal one, but we have found
that when people turn on the unsafe mode, they are very confident with
their decision (I do not say the decision  is correct), and can not wait
the robot cleaning up the cost map, so they will complain for the slow
motion, so if you change the behavior, you'd better to expect reactions.

if we use
https://github.com/SharedAutonomyToolkit/shared_autonomy_manipulation/blob/hydro-devel/safe_teleop_base/src/safe_trajectory_planner_ros.cpp#L67-L70
parameter, and
https://github.com/fetchrobotics/fetch_ros/blob/indigo-devel/fetch_teleop/src/joystick_teleop.cpp#L105-L109,
sohuld we set slower value for fetchc_telop node?

By the way, does fetch roboth have speedup mode (B-dush mode) like pr2?
https://github.com/PR2/pr2_apps/blob/kinetic-devel/pr2_teleop_general/src/pr2_teleop_general_joystick.cpp#L426

k-okada, 2018-04-16:
> I don't think fetch have speedup mode.
>
> I believe that reducing the max speed in unsafe mode and adding speedup
> mode should be a good solution, also for being the same as PR2
> implementation.

ok, please update parameters and send PR,

Affonso-Gui, 2018-04-17:
cc @708yamaguchi 
We tried to set the `max_vel_x` to 0.1, but the velocity did not seemed to change, staying about 1.0 m/s for both  safe and unsafe mode.

Also doing some more tests, shifting from safe to unsafe mode did not seem to alter  the velocity at all. Maybe it was only impression that unsafe mode is faster, and they actually have the same limits (?)

708yamaguchi, 2018-04-17:
(Add supplementation)
First, we set `max_vel_x` at 0.1 in `/opt/ros/indigo/share/fetch_navigation/config/fetch/move_base.yaml`.
Next, restart fetch by `sudo reboot`
Then, move fetch with joystick with `rostopic echo /teleop/cmd_vel`, but the x direction velocity is `1.0`

708yamaguchi, 2018-04-20:
このファイルの設定(`/opt/ros/indigo/share/fetch_navigation/config/fetch/move_base.yaml`)は`teleop`ではなく、`navigation`でfetchを動かすときのみ適用される速度制限でした。

closed 2020-01-31 by Affonso-Gui