jsk-ros-pkg/jsk_robot #924: device:=“hw:2” をROSノードに与えた際にros::InvalidNameExceptionが出る問題について

k-okada, 2018-05-03:
これ、
https://github.com/ros-drivers/audio_common/blob/master/audio_capture/src/audio_capture.cpp#L38
にデバイス番号を与えたらいいんだけど、その時は
```
audio_capture device:=“hw:2”
```
ではなくて
```
audio_capture _device:=“hw:2”
```

とすればいい、という話だった、という理解でよかったかな？
http://wiki.ros.org/Nodes#Remapping_Arguments.Node_parameter_assignment

Cc: @708yamaguchi @iory

708yamaguchi, 2018-05-03:
はい。現状だとそれでエラーが出なくなります。

また、僕のコミット
https://github.com/708yamaguchi/ros_comm/commit/67740883b46b1160b2689686cd5f10a6cb92f756
を使うと
```
rosrun audio_capture audio_capture device:="hw:2,0"
```
としてもエラーが出なくなることを確認しました。

k-okada, 2018-05-03:
> 708yamaguchi/ros_comm@6774088

いや、これはROSノード的にはremappingを指定したことになっているのでは？という指摘です。

http://wiki.ros.org/Nodes#Remapping_Arguments.Node_parameter_assignment
を見てみてください

708yamaguchi, 2018-05-03:
```
rosrun audio_capture audio_capture device:="hw:2,0"
```
はremappingを指定しているが、
```
rosrun audio_capture audio_capture _device:="hw:2,0"
```
はrosnode内のパラメータを指定している、ということでしょうか。

申し訳ありませんが、この2つの違いについて教えていただけないでしょうか。

k-okada, 2018-05-03:
remap はrosノードの中の名前を変更する仕組みです．名前はノード，トピック，サービス，パラーメータの名前が対応します．
これは image というトピックをパブリッシュするプログラムをつくた時に，このプログラムをステレオカメラに使いたくてimage_r, image_l
に変更したいと
思った際に，プログラムのソースを変更して再編集するこんとなく実現したものです．これは今は当たり前だけど非常に画期的なものになります．
例えば，eusでデータをとって /tmp/jiken.txt
というファイル結果を書き出すプログラムを作った時に，ファイル名を変えたいと思った時にどうなるか，
ということを考えたらいいとおもいます．

パラメータはlaunchのrosparam で与えられるもので，プログラムの中ではros::param:...
で実行時にプログラムにパラメータをセットする仕組みです．
これの面白いところはパラメータはroscoreに蓄えられるので，プログラムが途中でsegfaultしてrespawnした時に，最終状態をparameterに書き出していれば，
その状態からプログラムを続けることができます．これがlaunchのrespawn="true" に関連しています．これも画期的な仕組みです．

launchファイルの中でこれらを制御するには<remap> <param>/<rosparam>
をつかいますが，プログラムを実行するときにコマンドからこれを制御する方法が
http://wiki.ros.org/Nodes#Command-line_Remapping_Arguments
に書いてあって，それが hoge:=fuga と _hoge:=fuga です．あとは  __name:=fuga というルールも上に書いてあります．

普通のCのプログラムでこういう引数をとるようなことはなくて，
rosrun audio_capture audio_capture --device=hw:2
みたいに引数をしていするんですが，こういうふうに書くとrosの指令だという風に思わずに，プログラム的には名前チェックがおこりまｓねん．

今回はgstreamer が hoge:=fuga スタイルで引数を与えるので，ROS側のルールとコンフリクトしているということになります．
ただ，audio_captureは引数はparam経由で与えるようにできているように見えるので，使い方の問題だったという理解なんだけど，
正しいかな．

http://wiki.ros.org/roslaunch/XML/remap
http://wiki.ros.org/Names#Valid_Names-1
http://wiki.ros.org/Remapping%20Arguments
とかを見てみましょう．

708yamaguchi, 2018-05-04:
岡田先生
丁寧なご説明ありがとうございます。

最初の話に戻りますと、
```
rosrun audio_capture audio_capture device:="hw:2,0"
```
とかくと、`/audio_capture/device`というパラメータを`/audio_capture/hw:2,0`という名前でremappingしようとするので、エラーが起きました。今回は、`/audio_capture/device`の値をhw:2,0にセットするというのが目的だったので、
```
rosrun audio_capture audio_capture _device:="hw:2,0"
```
とするのが正解ということですね。僕がコマンドライン引数の使い方を間違えていました。

パラメータやremap周りの良い勉強になりました。ありがとうございました。