jsk-ros-pkg/jsk_robot #939: Strange sounds are coming from PR2 1040 base

takayuki5168, 2018-06-12:
Today, after a long time, when I commanded PR2 1040 to move base, strange sounds were comming from base.

Help me anyone.

takayuki5168, 2018-06-12:
I heared sounds carefully, and the sounds were comming from surface of left forward caster unit, but not inside of it. So I think it is easy to find case of this sounds.

I opend the base cover of PR2, of course I shutdown servo of PR2, then easily I found nuts of left forward caster unit(red circles in below image) were loose. So I fixed it strongly.

<br>

Now, no strange sounds are comming!

Maybe this loose nuts was caused when this issue https://github.com/jsk-ros-pkg/jsk_robot/issues/912 were fixed.

[image]

takayuki5168, 2018-06-12:
@furushchev @k-okada 

This issue was solved, so would you close after you read. And maybe I have no right to set Labels, so please label this issue PR1040.

furushchev, 2018-06-13:
Talked @takayuki5168 and confirmed that this problem is now solved.
@takayuki5168 Thx!

closed 2018-06-13 by furushchev

k-okada, 2018-06-13:
@@takayuki5168 very good, did you use PR2 special lock tight in PR2 tool
box? If not, please use that.

takayuki5168, 2018-06-13:
@k-okada 

I didnt use it. Its PR2 special lock tight can be removed when we want to? Thats bacause, when we fix PR2 caster unit, the nuts I locked today must be took off.

k-okada, 2018-06-13:
yes if you do not put them "appropriate" amount