jsk-ros-pkg/jsk_robot #940: PR2 1012 suddenly servo off

takayuki5168, 2018-06-13:
PR2 1012 suddenly servo off many times from yesterday. Once suddenly servo off, servo on and off are repeated automatically speedly.

Below images are which I captured pr2 dashboard.

[image]
[image]

takayuki5168, 2018-06-19:
Today, few days layer, I used PR1012 but servo didn't be off at all. So I close this issue.

If servo off again, I will reopen this issue.

closed 2018-06-19 by takayuki5168

furushchev, 2018-06-19:
@takayuki5168 Thanks for report! Please watch carefully dashboard and also the output on the screen you are running `robot start` where logs related to `ethercat` or communication problem will be displayed.