jsk-ros-pkg/jsk_robot #941: low battery of PR1040

takayuki5168, 2018-06-14:
Speed of decreasing battery of PR1040 is very fast.

There was same issue(https://github.com/jsk-ros-pkg/jsk_robot/issues/909#issuecomment-396788057) in the past. Below is the result of commanding `rosrun jsk_tools battery_capacity_summary.py`

```
[http://pr1040:11311][133.11.216.112] takayuki@takayuki-ThinkPad-T460s:~$ rosrun jsk_tools battery_capacity_summary.py 
aggregating battery info...
                   Battery Name|     CycleCount|   FullCapacity|     HardwareID|ManufactureDate|RemainingCapacity
/Power System/Smart Battery 0.0|             33|           1997|           9679|       3/6/2015|           1937
/Power System/Smart Battery 0.1|             35|           3915|           9662|       3/6/2015|           3830
/Power System/Smart Battery 0.2|             34|           4719|           9681|       3/6/2015|           4660
/Power System/Smart Battery 0.3|             35|           4865|           9683|       3/6/2015|           5911
/Power System/Smart Battery 1.0|              5|           6957|          24829|     11/27/2017|           6643
/Power System/Smart Battery 1.1|              5|           6943|          24830|     11/27/2017|           6636
/Power System/Smart Battery 1.2|              5|           6956|          24826|     11/27/2017|           6651
/Power System/Smart Battery 1.3|              5|           6928|          24850|     11/27/2017|           6632
/Power System/Smart Battery 2.0|             32|           6084|           9680|       3/6/2015|           6044
/Power System/Smart Battery 2.1|             32|           4930|           9670|       3/6/2015|           4890
/Power System/Smart Battery 2.2|             33|           5967|           8771|      1/21/2015|           5927
/Power System/Smart Battery 2.3|             31|           5900|           9674|       3/6/2015|           5606
/Power System/Smart Battery 3.0|             36|           1713|           9652|       3/6/2015|           1660
/Power System/Smart Battery 3.1|             36|           1840|           9663|       3/6/2015|           1789
/Power System/Smart Battery 3.2|             37|           2071|           8775|      1/21/2015|           2017
/Power System/Smart Battery 3.3|             35|           4234|           9668|       3/6/2015|           4181
```

furushchev, 2018-06-15:
Seeing `FullCapacity` for each batteries, it looks batteries 3.X has no more performance than usual.
However, `CycleCount` of the batteries is only ~40 times which is far from their lifetime.

c.f. The expected lifetime can be found here at page 23:
http://www.inspired-energy.com/pdfs/datasheets/NL2024A22-spec.pdf

> 3.8. Reliability
3.8.1. Life Expectancy
Given normal storage & usage, user can expect the battery to deliver 80% or more of it's initial capacity after 300
charge/discharge cycles where the charge phase is CC/CV 4000mA, 16.80.05V and the discharge is 1320mA down
to 2.9V/Cell at 25°C.

furushchev, 2018-06-15:
IMO, you can replace dead batteries for each blocks (e.g. 3.0 ~ 3.3) with new ones at this time since we cannot use the robot with enough mobility, but at the same time we have to inspect where this issue comes from.
(Though I could not finally find any cause before.)

furushchev, 2018-06-15:
FYI: http://www.inspired-energy.com/pdfs/Smart-Battery-User-Notes.pdf

takayuki5168, 2018-10-01:
PR2は再起動するとバッテリーの表示が変わる説を検証します。

再起動前の `rosrun jsk_tools battery_capacity_summary.py` お願いします。 @Affonso-Gui

Affonso-Gui, 2018-10-01:
再起動前:
```
                   Battery Name|     CycleCount|   FullCapacity|     HardwareID|ManufactureDate|RemainingCapacity
/Power System/Smart Battery 0.0|             38|           3789|           9679|       3/6/2015|           3712
/Power System/Smart Battery 0.1|             40|           1847|           9662|       3/6/2015|           1775
/Power System/Smart Battery 0.2|             39|           4582|           9681|       3/6/2015|           4511
/Power System/Smart Battery 0.3|             41|           6389|           9683|       3/6/2015|           6292
/Power System/Smart Battery 1.0|             10|           6923|          24829|     11/27/2017|           6862
/Power System/Smart Battery 1.1|             10|           6926|          24830|     11/27/2017|           6865
/Power System/Smart Battery 1.2|             10|           6929|          24826|     11/27/2017|           6864
/Power System/Smart Battery 1.3|             10|           6922|          24850|     11/27/2017|           6859
/Power System/Smart Battery 2.0|             38|           6264|           9680|       3/6/2015|           6179
/Power System/Smart Battery 2.1|             38|           6031|           9670|       3/6/2015|           5943
/Power System/Smart Battery 2.2|             38|           6296|           8771|      1/21/2015|           6204
/Power System/Smart Battery 2.3|             36|           6250|           9674|       3/6/2015|           6167
/Power System/Smart Battery 3.0|             41|           4098|           9652|       3/6/2015|           4021
/Power System/Smart Battery 3.1|             41|           4632|           9663|       3/6/2015|           4553
/Power System/Smart Battery 3.2|             42|           5227|           8775|      1/21/2015|           5135
/Power System/Smart Battery 3.3|             40|           6700|           9668|       3/6/2015|           6595
```
再起動後:
```
/Power System/Smart Battery 0.0|             38|           3095|           9679|       3/6/2015|           3006
/Power System/Smart Battery 0.1|             40|           1849|           9662|       3/6/2015|           5911
/Power System/Smart Battery 0.2|             39|           4581|           9681|       3/6/2015|              0
/Power System/Smart Battery 0.3|             41|           6390|           9683|       3/6/2015|           6277
/Power System/Smart Battery 1.0|             10|           6923|          24829|     11/27/2017|           6847
/Power System/Smart Battery 1.1|             10|           6925|          24830|     11/27/2017|           6852
/Power System/Smart Battery 1.2|             10|           6929|          24826|     11/27/2017|           6852
/Power System/Smart Battery 1.3|             10|           6922|          24850|     11/27/2017|           6845
/Power System/Smart Battery 2.0|             38|           6262|           9680|       3/6/2015|           6165
/Power System/Smart Battery 2.1|             38|           6032|           9670|       3/6/2015|           5930
/Power System/Smart Battery 2.2|             38|           6296|           8771|      1/21/2015|           6191
/Power System/Smart Battery 2.3|             36|           6251|           9674|       3/6/2015|           6155
/Power System/Smart Battery 3.0|             41|           4116|           9652|       3/6/2015|           4023
/Power System/Smart Battery 3.1|             41|           4620|           9663|       3/6/2015|           4527
/Power System/Smart Battery 3.2|             42|           5219|           8775|      1/21/2015|           5115
/Power System/Smart Battery 3.3|             40|           6701|           9668|       3/6/2015|           6585
```

Affonso-Gui, 2018-10-01:
Full Capacity does not seem to change a lot, but the `0` on Battery 0.2 is bothering

Affonso-Gui, 2018-10-02:
After changing the batteries `0.0`~`0.3` 
```
                   Battery Name|     CycleCount|   FullCapacity|     HardwareID|ManufactureDate|RemainingCapacity
/Power System/Smart Battery 0.0|              1|           6749|          24839|     11/27/2017|           6671
/Power System/Smart Battery 0.1|              1|           6764|          24852|     11/27/2017|           6709
/Power System/Smart Battery 0.2|              1|           6741|          24847|     11/27/2017|           6741
/Power System/Smart Battery 0.3|              1|           6758|          24817|     11/27/2017|           6662
/Power System/Smart Battery 1.0|            N/A|            N/A|          24850|            N/A|           6883
/Power System/Smart Battery 1.1|            N/A|            N/A|          24850|            N/A|           6888
/Power System/Smart Battery 1.2|             11|           6907|          24826|     11/27/2017|           6890
/Power System/Smart Battery 1.3|             10|           6896|          24850|     11/27/2017|           6875
/Power System/Smart Battery 2.0|             38|           6252|           9680|       3/6/2015|           6214
/Power System/Smart Battery 2.1|             38|           5850|           9670|       3/6/2015|           5829
/Power System/Smart Battery 2.2|             39|           6272|           8771|      1/21/2015|           6224
/Power System/Smart Battery 2.3|             36|           6217|           9674|       3/6/2015|           6183
/Power System/Smart Battery 3.0|             41|           3814|           9652|       3/6/2015|           3766
/Power System/Smart Battery 3.1|             41|           4374|           9663|       3/6/2015|           4327
/Power System/Smart Battery 3.2|            N/A|            N/A|           9663|            N/A|           5175
/Power System/Smart Battery 3.3|            N/A|            N/A|           9663|            N/A|           5911
```

Full capacity is back to around 6,000 again, but getting some weird `N/A` errors. 
Maybe have some of the cables loose? Will have a check

Affonso-Gui, 2018-10-02:
Seems to be caused by callibration beeing on hold (not waiting the starting sequence to completely finish).

Now we have:
```
                   Battery Name|     CycleCount|   FullCapacity|     HardwareID|ManufactureDate|RemainingCapacity
/Power System/Smart Battery 0.0|              1|           6749|          24839|     11/27/2017|           6671
/Power System/Smart Battery 0.1|              1|           6764|          24852|     11/27/2017|           5911
/Power System/Smart Battery 0.2|              1|           6741|          24847|     11/27/2017|           6741
/Power System/Smart Battery 0.3|              1|           6758|          24817|     11/27/2017|           6662
/Power System/Smart Battery 1.0|             10|           6898|          24829|     11/27/2017|           6883
/Power System/Smart Battery 1.1|             10|           6905|          24830|     11/27/2017|           6888
/Power System/Smart Battery 1.2|             11|           6907|          24826|     11/27/2017|           6890
/Power System/Smart Battery 1.3|             10|           6896|          24850|     11/27/2017|           6875
/Power System/Smart Battery 2.0|             38|           5911|           9680|       3/6/2015|           6214
/Power System/Smart Battery 2.1|             38|           5850|           9670|       3/6/2015|           5911
/Power System/Smart Battery 2.2|             39|           6272|           8771|      1/21/2015|           6224
/Power System/Smart Battery 2.3|             36|           6217|           9674|       3/6/2015|           6183
/Power System/Smart Battery 3.0|             41|           3814|           9652|       3/6/2015|           3766
/Power System/Smart Battery 3.1|             41|           4374|           9663|       3/6/2015|           4327
/Power System/Smart Battery 3.2|             42|           5236|           8775|      1/21/2015|           5175
/Power System/Smart Battery 3.3|             40|           6678|           9668|       3/6/2015|           6619
```

takayuki5168, 2018-10-09:
上記、交換された古い0.xのバッテリーは2015年5月に交換されたものでした。

iory, 2018-11-13:
FullCapacityの値は下記となります。
3.0 1151mAh
3.1 1214mAh
3.2 1273mAh
3.3 2624mAh

furushchev, 2018-11-13:
ありがとう．その値を信じると明らかにバッテリにダメージを与える出来事がここ1ヶ月で起こったということだよね．（自然劣化ではなく）
原因を詳細に調べたいところです．．．

iory, 2018-11-13:
昨日DC-DCが怪しいかもしれないとのことでしたが、
DC-DCを変えてみるということは試してみる価値があるかと思います。

knorth55, 2022-09-11:
2022/09/11

PR1040のバッテリ3.0-3.3系列のバッテリが悪いようです．
当時はバッテリ充電ボードがおかしかったのもあって，かなりヘタリが早かったようです．
（このIssueの例だとサイクルカウント40くらいでヘタっていたようです)

今回はサイクルカウント180くらいなので，そろそろ交換なのかな，とおもいます．

```
                   Battery Name| Serial| ManufactureDate| FullCapacity(mAh)| RemainingCapacity(mAh)| Voltage(mV)| CycleCount| Status
/Power System/Smart Battery 0.0|  24842|      11/27/2017|              6669|                   6402|       16809|         36|    128
/Power System/Smart Battery 0.1|  24840|      11/27/2017|              6750|                   6513|       16799|         34|    128
/Power System/Smart Battery 0.2|  24843|      11/27/2017|              6516|                   6280|       16796|         37|    128
/Power System/Smart Battery 0.3|  24844|      11/27/2017|              6701|                   6455|       16799|         37|    128
/Power System/Smart Battery 1.0|  24833|      11/27/2017|              6671|                   6314|       16804|         63|    128
/Power System/Smart Battery 1.1|  24820|      11/27/2017|              7033|                   6633|       16736|         54|    128
/Power System/Smart Battery 1.2|  24825|      11/27/2017|              6567|                   6199|       16801|         63|    128
/Power System/Smart Battery 1.3|  24836|      11/27/2017|              6506|                   6147|       16803|         65|    128
/Power System/Smart Battery 2.0|  24828|      11/27/2017|              5805|                   5436|       16788|        154|    128
/Power System/Smart Battery 2.1|  24838|      11/27/2017|              5985|                   5515|       16799|        152|    128
/Power System/Smart Battery 2.2|  24837|      11/27/2017|              6000|                   5527|       16780|        154|    128
/Power System/Smart Battery 2.3|  24821|      11/27/2017|              5925|                   5530|       16781|        155|    128
/Power System/Smart Battery 3.0|  24831|      11/27/2017|              1472|                    492|       16449|        173|    128
/Power System/Smart Battery 3.1|  24827|      11/27/2017|              3451|                   1951|       15360|        172|    192
/Power System/Smart Battery 3.2|  24824|      11/27/2017|              1623|                    595|       15670|        175|    704
/Power System/Smart Battery 3.3|  24818|      11/27/2017|              1046|                    446|       16236|        176|    128

```

closed 2023-04-07 by takayuki5168