jsk-ros-pkg/jsk_robot #951: PR2のROSノードを自動起動する

furushchev, 2018-07-16:
他のロボットに合わせて自動で起動するようにしてみました。
現在PR1012, PR1040ともに電源（背中の赤スイッチ）を入れると、勝手に起動が始まるようになっています。

起動シーケンスは以下のようになりました。

- 【ユーザ】赤いスイッチを押す
- 【ロボット】ピアノの音が聞こえる(5分くらい)
- 【ロボット】Launchingとしゃべる
- 【ロボット】Now I am readyとしゃべる
- 【ユーザ】遠隔停止スイッチを解除する
- （初回のみ）【ロボット】キャリブレーションが始まる
- 【ロボット】チルトレーザが回り始める

upstartという仕組みを使って自動起動をしていますが、
今まで手動でやっていたことをやりたい場合は各ユーザにログインして、

`robot start` -> `sudo initctl start robot`
`robot stop` -> `sudo initctl stop robot`

としてください。

`sudo initctl start robot`をすると、従来の`robot start`をしたあとに自動で`roslaunch jsk_pr2_startup pr2.launch`が起動します。
`sudo initctl stop robot`をすると、先に`pr2.launch`が終了するのを待ってから`robot stop && robot release`を実行します。

furushchev, 2018-07-16:
設定ログです

```bash
# /etc/init/robot.conf 
author "Yuki Furuta <furushchev@jsk.imi.i.u-tokyo.ac.jp>"
description "robot start"

start on c2-comp-up
stop on shutdown

respawn
console log

env ROS_LOG_DIR=/var/log/ros
env ROS_ENV_LOADER=/home/applications/ros/indigo/devel/env.sh
env ROSCONSOLE_FORMAT='[${severity}] [${time}] [${node}:${logger}]: ${message}'
env PID_FILE=/var/tmp/ros.pid
env LAUNCH_FILE=/etc/ros/robot.launch

pre-start script
  chown -R applications:applications /var/log/ros
  exec >"/tmp/ros_run_id"
  echo "export ROS_IP=$(ip -o route get 8.8.8.8 | awk '{print $7;}')"
end script

script
  . "/tmp/ros_run_id"
  exec su applications -c ". /tmp/ros_run_id; robot claim -f; robot stop -f; $ROS_ENV_LOADER roslaunch $LAUNCH_FILE --pid $PID_FILE"
end script

post-start script
  echo "waiting for roscore to come up"
  . /opt/ros/indigo/setup.sh
  ret=`rosnode list`
  while [ "$ret" = '' ]
  do
    ret=`rosnode list`
    sleep 1;
  done
  rostopic echo /calibrated -n1
  initctl emit robot-is-up
end script

pre-stop script
  initctl emit robot-is-down
  exec su applications -c "robot stop -f"
end script

post-stop script
  exec su applications -c "robot release -f"
end script
```

```
# /etc/init/jsk-pr2-startup.conf 
author "Yuki Furuta <furushchev@jsk.imi.i.u-tokyo.ac.jp>"
description "robot start"

start on robot-is-up
stop on robot-is-down

respawn
console log

env ROS_ENV_LOADER=/home/applications/ros/indigo/devel/env.sh
env ROSCONSOLE_FORMAT='[${severity}] [${time}] [${node}:${logger}]: ${message}'

pre-start script
  exec >"/tmp/ros_run_id"
  echo "ROS_IP=$(ip -o route get 8.8.8.8 | awk '{print $7;}')"
end script

script
  exec su applications -c ". /tmp/ros_run_id; $ROS_ENV_LOADER roslaunch jsk_pr2_startup pr2.launch --screen --wait"
end script
```

`robot.conf`の`c2-comp-up`イベントはC2が立ち上がって、ROSのノードを実行可能な状態になった時に発火されます。

また各ユーザに`initctl`を`sudo`で許可する設定を/etc/sudoersに加えました。

furushchev, 2018-08-01:
最終的に以下の設定でここのところ運用出来ていると思います．

```conf
# /etc/init/robot.conf
author "Yuki Furuta <furushchev@jsk.imi.i.u-tokyo.ac.jp>"
description "robot start"

start on c2-comp-up
stop on shutdown

respawn
console log

env ROS_LOG_DIR=/var/log/ros
env ROS_ENV_LOADER=/home/applications/ros/indigo/devel/env.sh
env ROSCONSOLE_FORMAT='[${severity}] [${time}] [${node}:${logger}]: ${message}'
env PID_FILE=/var/tmp/ros.pid
env LAUNCH_FILE=/etc/ros/robot.launch

pre-start script
  chown -R applications:applications /var/log/ros
  exec >"/tmp/ros_run_id"
  echo "export ROS_IP=$(ip -o route get 8.8.8.8 | awk '{print $7;}')"
end script

script
  . "/tmp/ros_run_id"
  exec su applications -c ". /tmp/ros_run_id; robot claim -f; robot stop -f; sleep 10; $ROS_ENV_LOADER roslaunch $LAUNCH_FILE --pid $PID_FILE"
end script

post-start script
  echo "waiting for roscore to come up"
  . /opt/ros/indigo/setup.sh
  ret=`rosnode list`
  for i in `seq 0 20`
  do
    if [ "$ret" = '' ]
    then
      break
    fi
    ret=`rosnode list`
    sleep 1;
  done
  if [ "$ret" = '' ]
    echo "Roscore is down"
  then
    rostopic echo /calibrated -n1
    initctl emit robot-is-up
  fi
end script

pre-stop script
  initctl emit robot-is-down
  exec su applications -c "robot stop -f"
end script

post-stop script
  exec su applications -c "robot release -f"
end script
```

```conf
# /etc/init/jsk-pr2-startup.conf
author "Yuki Furuta <furushchev@jsk.imi.i.u-tokyo.ac.jp>"
description "robot start"

start on c2-comp-up
stop on (robot-is-down or shutdown)

respawn
console log

env ROS_ENV_LOADER=/home/applications/ros/indigo/devel/env.sh
env ROSCONSOLE_FORMAT='[${severity}] [${time}] [${node}:${logger}]: ${message}'

pre-start script
  exec >"/tmp/ros_run_id"
  echo "ROS_IP=$(ip -o route get 8.8.8.8 | awk '{print $7;}')"
end script

script
  exec su applications -c ". /tmp/ros_run_id; $ROS_ENV_LOADER roslaunch jsk_pr2_startup pr2.launch --screen --wait"
end script
```

```diff
# /etc/sudoers
- ALL ALL=(ALL) NOPASSWD: /usr/bin/ckill, /usr/bin/ipmitool
+ ALL ALL=(ALL) NOPASSWD: /usr/bin/ckill, /usr/bin/ipmitool, /usr/bin/tail, /usr/bin/less, /sbin/initctl
```

問題があったらreopenしてください．

shmpwk, 2021-09-06:
最終的な設定についての質問です。

1. post-start scriptにおいて、roscore が立ち上がるまで、つまりrosnode list が空である間は待ち続けるようにしていると思ったのですが、
    if [ "$ret" = '' ]
    then
      break
    fi
とあるので、空のときに待ち状態から抜けていると思います。これで正しいのでしょうか。

2. 
post-start script で、20 seq待った後でもroscoreが立ち上がらず、rosnode listが空の時に echo "Roscore is down" して、次に/calibrated トピックが出ているか確認していくと解釈すると、then はelseではないでしょうか。

  if [ "$ret" = '' ]
    echo "Roscore is down"
  then
    rostopic echo /calibrated -n1
    initctl emit robot-is-up
  fi

knorth55, 2021-09-06:
1.

```
    if [ "$ret" != '' ]
    then
      break
    fi
```

https://gitlab.jsk.imi.i.u-tokyo.ac.jp/pr2/pr1040_etc/blob/master/init/robot.conf#L33-36

`!=`じゃないかな？

2.
はたしかにそうだね，ｓこれはタイポなのかな．

shmpwk, 2021-09-06:
1,2についてその修正をしてgitlabにPRを出したいと思います。

knorth55, 2021-09-06:
いまPR2でためしてみました．
1.は元からまちがっていないと思います．
https://gitlab.jsk.imi.i.u-tokyo.ac.jp/pr2/pr1040_etc/blob/master/init/robot.conf#L33-36
2.は修正してみました．

knorth55, 2021-09-06:
https://gitlab.jsk.imi.i.u-tokyo.ac.jp/pr2/pr1040_etc/commit/d49f606498829a513534efe7f0bdc57a401f0e60

shmpwk, 2021-09-06:
ご修正ありがとうございます．

closed 2021-09-06 by shmpwk