jsk-ros-pkg/jsk_robot #967: PR1040が起動しない

furushchev, 2018-09-10:
`robot start`すると，realtime_loopノードが死んでいるようです．
ethercatが動かない状態ですね．

```bash
waitForReadMailboxReady : ERROR error read mbx not full after 100 ms
readMailbox_ : ERRORwaiting for read mailbox
[FATAL] [1536556275.673315230] [/realtime_loop:ros.ethercat_hardware]: Unable to load configuration in
formation
[FATAL] [1536556275.673338736] [/realtime_loop:ros.ethercat_hardware]: Unable to initialize slave #18,
 product code: 6805005, revision: 100860182, serial: 3170
```

furushchev, 2018-09-10:
背中の赤スイッチのOFF/ONで再起動した所，正常に起動し（てしまい）ました．
その前に`pr2-shutdown --reboot`のみの再起動もしてみましたが，それだけではダメでした．

slave#18の調べ方はよくわかりませんが，試しにmotorconfを実行してみました．

```bash
applications@pr1040:~$ sudo initctl stop robot                                               [239/239]
robot stop/waiting
applications@pr1040:~$ su - root
Password: 
root@pr1040:~# source /opt/ros/indigo/setup.bash
root@pr1040:~# roscd ethercat_hardware/
root@pr1040:/opt/ros/indigo/share/ethercat_hardware# rosrun ethercat_hardware motorconf -i ecat0 > /tm
p/motorconf
```

```bash
applications@pr1040:~$ cat /tmp/motorconf  | grep 3170 -C3
[DEBUG] [1536557516.806313504] [${node}:ros.ethercat_hardware]:             Name: l_shoulder_pan_motor
[ WARN] [1536557516.810309408] [${node}:ros.ethercat_hardware]: l_shoulder_pan_motor EEPROM does not contain motor heating model parameters
[DEBUG] [1536557516.811307870] [${node}:ros.ethercat_hardware]: Read calibration from device l_shoulder_pan_motor: -52.626913
[DEBUG] [1536557516.811327901] [${node}:ros.ethercat_hardware]: Device #18: WG05 (0x67d60d) Firmware Revision 1.22, PCB Revision F.03, Serial #: 3170
[DEBUG] [1536557516.812302989] [${node}:ros.ethercat_hardware]:             Serial #: 03170
[DEBUG] [1536557516.816306812] [${node}:ros.ethercat_hardware]:             Name: l_shoulder_lift_motor
[ WARN] [1536557516.820302471] [${node}:ros.ethercat_hardware]: l_shoulder_lift_motor EEPROM does not contain motor heating model parameters
[DEBUG] [1536557516.821301414] [${node}:ros.ethercat_hardware]: Read calibration from device l_shoulder_lift_motor: -80.519020
```

ということで，`l_shoulder_lift_motor`のMCBが見えなかったということでしょうか．

furushchev, 2018-12-11:
一度閉じます．問題発生したらreopenします．

knorth55, 2019-09-24:
この問題が今日いまおきました
@Affonso-Gui @YutoUchimi

Affonso-Gui, 2019-09-24:
今回も背中の赤スイッチをOFF/ONした所で直りました。

closed 2019-09-24 by Affonso-Gui

knorth55, 2019-09-24:
報告です。
今日もpr1040で同様に
```
[FATAL] [1536556275.673315230] [/realtime_loop:ros.ethercat_hardware]: Unable to load configuration in
formation
[FATAL] [1536556275.673338736] [/realtime_loop:ros.ethercat_hardware]: Unable to initialize slave #18,
 product code: 6805005, revision: 100860182, serial: 3170
```
が表示されており、`pr2-shutdown`とRun Stop（remote controller and its backside)を押しても解決しませんでした。
なのでベースのブレーカから再起動しました。
すると問題になっていたMotor controller boardにも接続でき、キャリブレーションができました。