jsk-ros-pkg/jsk_robot #974: [pr1040] Calibration stops at shoulder

Affonso-Gui, 2018-10-16:
PR1040 calibration always stops on shoulder, with the following message:

```
[/calibrate_pr2:rosout]: Joint ['r_shoulder_lift', 'l_shoulder_lift'] is taking a long time to calibrate. It might be stuck and need some human help
```

Manually pushing the arms up solves the problem for now.

@takayuki5168

takayuki5168, 2018-10-22:
As I said to  @Affonso-Gui in tha past, this problem always happens inevitaly in PR1040 through my experience. 

Maybe maximum torque of shoulder motor when calibratoin is not enough because time has passed much since PR1040 was created. 

In my opinion there is no solution other than pushing arms up manually, and by making this issue our juniours or others can see this problem and solution. 

So I think this issue may be closed. How about @Affonso-Gui

Affonso-Gui, 2018-10-23:
I'll close this for now then. Please feel free to reopen if we have some new information

closed 2018-10-23 by Affonso-Gui