jsk-ros-pkg/jsk_robot #980: [jsk_naoqi_robot] kinetic environment setup

kochigami, 2018-10-25:
I use ubuntu 16.04, ros kinetic

I followed this (#969 ) and executed 
`roslaunch jsk_pepper_startup jsk_pepper_startup.launch network_interface:=enp0s25`

The log told me to install `naoqi_dashboard`, but I'm wondering whether it is released for kinetic or not.
(I have no time, so just wrote this issue without any trials.)
Maybe I have to try it with all the deb files. (Now I installed some packages from source.)

In addition, nao_apps (source package) can't be compiled without https://github.com/ros-naoqi/nao_robot/pull/37

```
roslaunch jsk_pepper_startup jsk_pepper_startup.launch network_interface:=enp0s25
... logging to /home/kanae/.ros/log/61930700-d832-11e8-af84-28d244f02e4c/roslaunch-kanae-ThinkPad-T440p-8900.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 749, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 721, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 589, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: naoqi_dashboard
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/kanae/catkin_ws/src/jsk_robot/jsk_aero_robot/aeroeus
ROS path [2]=/home/kanae/catkin_ws/src/jsk_robot/jsk_baxter_robot/baxtereus
ROS path [3]=/home/kanae/catkin_ws/src/jsk_robot/jsk_fetch_robot/fetcheus
ROS path [4]=/home/kanae/catkin_ws/src/jsk_robot/jsk_aero_robot/jsk_aero_startup
ROS path [5]=/home/kanae/catkin_ws/src/jsk_robot/jsk_baxter_robot/jsk_baxter_startup
ROS path [6]=/home/kanae/catkin_ws/src/jsk_robot/jsk_baxter_robot/jsk_baxter_desktop
ROS path [7]=/home/kanae/catkin_ws/src/jsk_robot/jsk_baxter_robot/jsk_baxter_web
ROS path [8]=/home/kanae/catkin_ws/src/jsk_robot/jsk_fetch_robot/jsk_fetch_startup
ROS path [9]=/home/kanae/catkin_ws/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup
ROS path [10]=/home/kanae/catkin_ws/src/jsk_robot/jsk_pr2_robot/jsk_pr2_calibration
ROS path [11]=/home/kanae/catkin_ws/src/jsk_robot/jsk_pr2_robot/jsk_pr2_startup
ROS path [12]=/home/kanae/catkin_ws/src/jsk_robot/jsk_robot
ROS path [13]=/home/kanae/catkin_ws/src/jsk_robot/jsk_robot_common/jsk_robot_startup
ROS path [14]=/home/kanae/catkin_ws/src/jsk_robot/jsk_robot_common/jsk_robot_utils
ROS path [15]=/home/kanae/catkin_ws/src/nao_robot/nao_apps
ROS path [16]=/home/kanae/catkin_ws/src/nao_robot/nao_bringup
ROS path [17]=/home/kanae/catkin_ws/src/nao_robot/nao_description
ROS path [18]=/home/kanae/catkin_ws/src/nao_robot/nao_robot
ROS path [19]=/home/kanae/catkin_ws/src/naoqi_bridge/naoqi_apps
ROS path [20]=/home/kanae/catkin_ws/src/naoqi_bridge/naoqi_bridge
ROS path [21]=/home/kanae/catkin_ws/src/naoqi_bridge_msgs
ROS path [22]=/home/kanae/catkin_ws/src/naoqi_driver
ROS path [23]=/home/kanae/catkin_ws/src/jsk_robot/jsk_naoqi_robot/jsk_nao_startup
ROS path [24]=/home/kanae/catkin_ws/src/naoqi_bridge/naoqi_driver_py
ROS path [25]=/home/kanae/catkin_ws/src/naoqi_bridge/naoqi_pose
ROS path [26]=/home/kanae/catkin_ws/src/naoqi_bridge/naoqi_sensors_py
ROS path [27]=/home/kanae/catkin_ws/src/naoqi_bridge/naoqi_navigation/python
ROS path [28]=/home/kanae/catkin_ws/src/naoqi_bridge/naoqi_tools
ROS path [29]=/home/kanae/catkin_ws/src/jsk_robot/jsk_naoqi_robot/naoqieus
ROS path [30]=/home/kanae/catkin_ws/src/jsk_robot/jsk_naoqi_robot/naoeus
ROS path [31]=/home/kanae/catkin_ws/src/pepper_robot/pepper_bringup
ROS path [32]=/home/kanae/catkin_ws/src/pepper_robot/pepper_description
ROS path [33]=/home/kanae/catkin_ws/src/pepper_robot/pepper_robot
ROS path [34]=/home/kanae/catkin_ws/src/pepper_robot/pepper_sensors_py
ROS path [35]=/home/kanae/catkin_ws/src/jsk_robot/jsk_naoqi_robot/peppereus
ROS path [36]=/home/kanae/catkin_ws/src/jsk_robot/jsk_naoqi_robot/jsk_201504_miraikan
ROS path [37]=/home/kanae/catkin_ws/src/jsk_robot/jsk_pr2_robot/pr2_base_trajectory_action
ROS path [38]=/home/kanae/catkin_ws/src/jsk_robot/jsk_pr2_robot/jsk_pr2_desktop
ROS path [39]=/home/kanae/catkin_ws/src/jsk_robot/jsk_robot_common/roseus_remote
ROS path [40]=/opt/ros/kinetic/share
```

kochigami, 2018-10-31:
Same log was reproduced when I set up a totally new PC in a same ubuntu and ROS version.
I found `naoqi_dashboard` is not released for kinetic.
(I searched for `ros-kinetic-naoqi-dashboard`in http://packages.ros.org/ros/ubuntu/pool/main/r/)
In addition, when I git cloned it,  the error occurred (below).

If I commented out 
`  <include file="$(find naoqi_dashboard)/launch/naoqi_dashboard.launch"/>`
of `jsk_pepper_startup.launch` in `jsk_pepper_startup`, `jsk_pepper_startup.launch` was executed successfully.

```
RosPluginProvider.load(naoqi_dashboard/NAOqiDashboard) exception raised in __builtin__.__import__(naoqi_dashboard.frame, [NAOqiDashboard]):
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/home/miraikan2/catkin_ws/devel/lib/python2.7/dist-packages/naoqi_dashboard/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/home/miraikan2/catkin_ws/src/naoqi_dashboard/src/naoqi_dashboard/frame.py", line 50, in <module>
    from .motors import Motors
  File "/home/miraikan2/catkin_ws/src/naoqi_dashboard/src/naoqi_dashboard/motors.py", line 37, in <module>
    from python_qt_binding.QtGui import QMessageBox
ImportError: cannot import name QMessageBox

PluginManager._load_plugin() could not load plugin "naoqi_dashboard/NAOqiDashboard":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 86, in load
    raise e
ImportError: cannot import name QMessageBox

================================================================================REQUIRED process [naoqi_dashboard-21] has died!
process has died [pid 29200, exit code 1, cmd /home/miraikan2/catkin_ws/src/naoqi_dashboard/scripts/naoqi_dashboard __name:=naoqi_dashboard __log:=/home/miraikan2/.ros/log/d5080e3e-dcd2-11e8-a67d-d05099313e9e/naoqi_dashboard-21.log].
log file: /home/miraikan2/.ros/log/d5080e3e-dcd2-11e8-a67d-d05099313e9e/naoqi_dashboard-21*.log
Initiating shutdown!
================================================================================
```

YutoUchimi, 2018-10-31:
```
ImportError: cannot import name QMessageBox
```
Maybe this error occurs in kinetic+, not in indigo, right?
Recently I found the same error in another repository (https://github.com/jsk-ros-pkg/jsk_visualization/pull/708)

kochigami, 2018-10-31:
Yes, it has never happened in indigo.
Thank you for sharing the information! I'll check it :)