jsk-ros-pkg/jsk_robot #1021: 【pr1012】Could not find joint fr_caster_rotation_joint

iory, 2018-12-11:
pr1012の台車が動かず原因を探っていると、

```
[ INFO] [1544515316.573938679] [/realtime_loop:ros.pr2_calibration_controllers]: Joint br_caster_r_wheel_joint is already calibrated at offset 15.885987
[ INFO] [1544515689.500947205] [/realtime_loop:ros.pr2_calibration_controllers]: Joint torso_lift_joint is already calibrated at offset 4608.145700
[ INFO] [1544515689.543803478] [/realtime_loop:ros.pr2_calibration_controllers]: Joint head_pan_joint is already calibrated at offset -3.279194
[ INFO] [1544515689.593066843] [/realtime_loop:ros.pr2_calibration_controllers]: Joint head_tilt_joint is already calibrated at offset 2.800416
[ INFO] [1544515689.640470962] [/realtime_loop:ros.pr2_calibration_controllers]: Joint laser_tilt_mount_joint is already calibrated at offset 0.084823
[ INFO] [1544515689.701197626] [/realtime_loop:ros.pr2_calibration_controllers]: Joint fl_caster_r_wheel_joint is already calibrated at offset -6.702064
[ERROR] [1544515689.834044313] [/realtime_loop:ros.pr2_calibration_controllers]: Could not find joint fr_caster_rotation_joint (namespace: /calibration_controllers/calibrate/cal_caster_fr)
[ERROR] [1544515689.834241797] [/realtime_loop:ros.pr2_controller_manager]: Initializing controller 'calibration_controllers/calibrate/cal_caster_fr' failed
[ INFO] [1544515689.843383563] [/realtime_loop:ros.pr2_calibration_controllers]: Joint bl_caster_r_wheel_joint is already calibrated at offset 32.609732
[ INFO] [1544515689.979623701] [/realtime_loop:ros.pr2_calibration_controllers]: Joint br_caster_r_wheel_joint is already calibrated at offset 15.885987
```

iory, 2018-12-11:
```rosrun rqt_robot_monitor rqt_robot_monitor```でdiagnosticsを確認するとcaster_frがmissingとなっています。

[image]

furushchev, 2018-12-11:
[image]

[image]

FRキャスタの３つあるドライバのうち１つだけが見えていて，その１つがキャリブに失敗しているように見えるけど，
FLキャスタと比較すると，Packet countに比べてDropsの値がかなり大きいので，どこかで断線している可能性が高そうです．

ほかに気づいたことは，
EtherCATでデイジーチェーンしているので，各ドライバあたり２つLANポートがあって，３つのドライバのうち２つは両ポート使用，のこり１つは片方ポートのみ使用になっているはずで，
FLキャスタを見るとそれはfl_caster_rotation_motorに見える．（Status Port 1: No Link, Closed, No Comm）
FRキャスタも同様と仮定すると本来はfr_caster_r_wheel_jointは両方のポートが使用されているはずなのに，こちらも片方のみリンクしている状態になっている気がする．
本当は違うかもしれないので，確認が必要（どのみち分解するときに確認可能）だけど，
HUB -> fr_caster_r_wheel_motor -> fr_caster_l_wheel_motor -> fr_caster_rotation_wheel_motor
だったとして，順番に「→」のコネクションを確認していくしか無いと思います．

すべての作業を始める前に，まずはc1のバックアップをとりましょう．

furushchev, 2018-12-11:
@iory さっきの作業ログです．

問題：
c1のSystem HDDのバックアップをとった後にHDDを戻してもC1が起動しなかった

原因：
c1に対してSystemとRemovableのHDDがあって，Systemを抜いた状態で起動してしまうとデフォルトのブートの順番がRemovableになってしまう．→Removableの領域にOSがないので起動に失敗

解決：
BMCにつないでBIOS画面を遠隔操作し，ブートの順番を変える
(JREが動くPCが必要．今回はmacbook proで動作確認)
1. PCを有線で体内LANに接続
2. 固定IPを適当にとる(例：10.68.0.50/24)
3. Webブラウザから`http://10.68.0.200` (C1のBMCのアドレス)にアクセス
4. ログインして画面左下のLaunchボタンを押す．
5. (ポップアップがブロックされるときはブロックを外す．jviewer.jnlpがダウンロードされるはず)
6. (インストールしていないとき)jreをインストール
7. MD5WithRSA認証を有効にする

    ```diff
    # /Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/lib/security/java.security
    - jdk.jar.disabledAlgorithms=MD2, MD5, RSA keySize < 1024 
    + # jdk.jar.disabledAlgorithms=MD2, MD5, RSA keySize < 1024
    ```

8. `jviewer.jnlp`を起動
9. 未認証のアプリとなり起動に失敗するので，環境設定を開く
10. 「セキュリティとプライバシー」ボタンを押し，「一般」パネルの一番下に「このまま開く」ボタンが現れるので，これを押す
11. セキュリティの警告が出るので，OKを数回押す

以上でつながりました．
画面上のタブから"Ctrl+Alt+Del"ボタンを押してC1を再起動して，
起動中にF2を押してBIOS画面に入り，Boot orderの一番上を"ubuntu"に変えて"Save changes and exit"をすると，正常に起動しました．

iory, 2018-12-13:
@furushchev 
ありがとうございます。

台車の修理は終了し、無事動くことを確認しました。
状況としては2016年12月26日に交換したcaster_frがあったため、
そこに刺さっていた基盤が正常に動くことを確認し、
もともと使っていたfr_caster_rotation_jointとfr_caster_l_wheel_jointの基盤に代替しました。

現在の残りのモータドライバ基盤は
2016年12月26日に交換したcaster_brについている基盤３つ（動くことを確認）、
以前購入した未使用の基盤２つ、
今回交換した接触不良の基盤１つ（こちらははんだ付けで治る可能性大）、
今回交換した全く通信ができなかった基盤１つ、
8階で見つかったおそらく壊れているであろう基盤２つ
となっています。

closed 2018-12-14 by iory

knorth55, 2020-11-30:
これはつまりどういう結果になってのでしょうか？
fr_caster_r_wheel_jointのMCBは問題がなく，fr_caster_rotation_jointとfr_caster_l_wheel_jointが問題だったため，交換したということでしょうか？
できれば修理ログに情報をまとめてほしいです．

iory, 2020-11-30:
https://docs.google.com/presentation/d/10-JKJcPoI-vs8MGn7LObBkm1z0R9ypl-EwKuiXTWgqE/edit#slide=id.p
こちらにまとめました。

> caster_frに3つの基板がありスリップリングから伸びているケーブルがつながっている順に1(fr_caster_r_wheel_joint), 2(fr_caster_l_wheel_joint), 3(fr_caster_rotation_joint)と名付けたときに電源を挿した場合の挙動を調べた。スリップリング側のポートをaとしてケーブルを以下の組み合わせで調べている。“-”はボード上でつながっていることを表す。“->”はケーブルでつながっていることを表す。
	スリップリング -> 1a - 1b -> 2a - 2b -> 3a - 3b (ここは何もささない）
		fr_caster_r_wheel_jointが見えるようになる
	スリップリング -> 3b - 3a -> 2b - 2a -> 1b - 1a (ここは何もささない)
		fr_caster_rotation_jointが見えるようになる
	スリップリング -> 1b - 1a -> 3b - 3a -> 2b - 2a (ここは何もささない)
		fr_caster_rotation_joint, fr_caster_r_wheel_jointが見える
以上のことからモータドライバ基板2(fr_caster_l_wheel_joint)が怪しいと判断しfr_caster_l_wheel_jointを別のものに変更した。
またパケットの損失(Drops)から、接触不良か何かしらの問題があると判断しfr_caster_rotation_jointも交換した。