jsk-ros-pkg/jsk_robot #1046: [jsk_pepper_startup/nodes/joint-client.l] naoqi_bridge_msgsの読み込み

kochigami, 2019-04-03:
@k-okada 

https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/jsk_pepper_startup/nodes/joy-client.l#L8-L9 の，
```
(if (not (find-package "NAOQI_BRIDGE_MSGS"))
    (make-package "NAOQI_BRIDGE_MSGS" :nicknames "NAOQI_MSGS"))
```
の所で質問があります．
この行の役割と，
これを消して，`(ros::load-ros-package "naoqi_bridge_msgs")`に変えても良いかを伺いたいです．

#### この行の役割についての予想

ROS hydroくらいの時に，`naoqi_bridge_msgs`ではなく`naoqi_msgs`という名前だったので，
`(ros::load-ros-package "naoqi_bridge_msgs")`の代用として書いてあると思ったのですが，合っていますでしょうか．

ここでは，NAOQI_BRIDGE_MSGS パッケージが見つからなかったら．NAOQI_MSGSというニックネーム付きで，NAOQI_BRIDGE_MSGS パッケージを作っていると思います．

ところが，naoqi_bridge_msgsが`/opt/ros/kinetic/`以下，`~/catkin_ws/src`以下にある環境でも，以下のように，NAOQI_BRIDGE_MSGSは見つかりません．
```
;; ~/catkin_ws/src以下に移動
roseus
emacs -f shell

1.irteusgl$ find-package "NAOQI_BRIDGE_MSGS"
nil
2.irteusgl$ find-package "naoqi_bridge_msgs"
nil
5.irteusgl$ make-package "NAOQI_BRIDGE_MSGS" :nicknames "NAOQI_MSGS"
#<package #X560a5d0 NAOQI_BRIDGE_MSGS>
6.irteusgl$ find-package "NAOQI_BRIDGE_MSGS"
#<package #X560a5d0 NAOQI_BRIDGE_MSGS>
```

NAOQI_BRIDGE_MSGSパッケージを`make-package`で作った後も，`naoqi_bridge_msgs::JointAngleWithSpeed`は見つかりません．
`roslaunch jsk_pepper_startup jsk_pepper_startup.launch network_interface:=<>`

```
/opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl roseus-error: unbound variable naoqi_bridge_msgs::jointangleswithspeed in (ros::advertise "/pepper_robot/pose/joint_angles" naoqi_bridge_msgs::jointangleswithspeed 1), exitting...
```

`(ros::load-ros-package "naoqi_bridge_msgs")`を書けばエラーが無くなります．
hydroはもうEOLなので，消しても大丈夫でしょうか．

pazeshun, 2019-04-03:
こちらですが、このコードが動いていた時代では、
問題の行より前にある、
```
(ros::load-ros-manifest "peppereus")
```
の内部で`(ros::load-ros-package "naoqi_bridge_msgs")`もしくは`(ros::load-ros-package "naoqi_bridge_msgs")`がされていたんだと思います。

kochigami, 2019-04-03:
ありがとうございます．
#997, #998  で教わったことを忘れていました．
(ros::load-ros-manifest "peppereus")の内部で
`naoqi_msgs`しか読み込めなかった場合に，
```
(if (not (find-package "NAOQI_BRIDGE_MSGS"))
    (make-package "NAOQI_BRIDGE_MSGS" :nicknames "NAOQI_MSGS"))
```
して，`naoqi_msgs`を`naoqi_bridge_msgs`として扱っているということなのでしょうか．

もしよかったら教えてほしいのですが，
`naoqi_bridge_msgs`があって，`naoqi_msgs`がなくて，`naoqi_bridge_msgs`を`naoqi_msgs`として扱うバージョンで試してみた所，
` (ros::advertise "/pepper_robot/pose/joint_angles" naoqi_msgs::JointAnglesWithSpeed 1)`でエラーが出てしまうのですが，これはなぜでしょうか．

```
eustf roseus_c_util 1.irteusgl$ (ros::roseus "joy_client")
t
2.irteusgl$ (ros::load-ros-manifest "peppereus")
[ WARN] [1554296997.697314823]: Calling (load-ros-manifest peppereus) for the package without msg/srv will be deprecated
[ WARN] [1554296997.697608782]: ACTION REQUIRED
[ WARN] [1554296997.697697842]:  Use (load-ros-{package,msg,srv} pkg) for the dependent packages with msg/srv
[ WARN] [1554296997.697801181]:  See https://github.com/jsk-ros-pkg/jsk_robot/issues/823

;; extending gcstack 0x6b4c4b0[32738] --> 0x7a76950[65476] top=7c6e
nil

3.irteusgl$ find-package "NAOQI_MSGS"
nil

4.irteusgl$ find-package "NAOQI_BRIDGE_MSGS"
#<package #X7316a40 NAOQI_BRIDGE_MSGS>

5.irteusgl$ (if (not (find-package "NAOQI_MSGS"))
(make-package "NAOQI_MSGS" :nicknames "NAOQI_BRIDGE_MSGS"))
#<package #X7868370 NAOQI_MSGS>

6.irteusgl$ find-package "NAOQI_MSGS"
#<package #X7868370 NAOQI_MSGS>

7.irteusgl$ find-package "NAOQI_BRIDGE_MSGS"
#<package #X7316a40 NAOQI_BRIDGE_MSGS>

8.irteusgl$ (ros::advertise "/pepper_robot/pose/joint_angles" naoqi_msgs::JointAnglesWithSpeed 1)
Call Stack (max depth: 20):
  0: at (ros::advertise "/pepper_robot/pose/joint_angles" naoqi_msgs::jointangleswithspeed 1)
  1: at #<compiled-code #X66ff708>
/opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: unbound variable naoqi_msgs::jointangleswithspeed in (ros::advertise "/pepper_robot/pose/joint_angles" naoqi_msgs::jointangleswithspeed 1)

10.E1-irteusgl$ (ros::advertise "/pepper_robot/pose/joint_angles" naoqi_bridge_msgs::JointAnglesWithSpeed 1)
t
```

kochigami, 2019-04-05:
### 意図したように動いていない？

```
(if (not (find-package "NAOQI_BRIDGE_MSGS"))
    (make-package "NAOQI_BRIDGE_MSGS" :nicknames "NAOQI_MSGS"))
```

今は，

from: `(ros::load-ros-manifest "peppereus")`を介して実行される，
`(ros::load-ros-package "naoqi_bridge_msgs")`, `(ros::load-ros-package "naoqi_msgs")`の中の`naoqi_msgs`

to: `(ros::advertise "/pepper_robot/pose/joint_angles" naoqi_bridge_msgs::JointAnglesWithSpeed 1)` の中の`naoqi_bridge_msgs`

があって，from側をいじっていると思っていたが，実はto側をいじっていた？

#### 意図した動作

`naoqi_msgs`しか読み込めなくて，`naoqi_bridge_msgs`が読み込めない場合，
`naoqi_msgs`パッケージを`naoqi_bridge_msgs`パッケージとしても使えるようにする

つまり，`naoqi_msgs::xxx`の継承関係を`naoqi_bridge_msgs::xxx`のように扱えるようにする．

#### 実際の挙動の予想

継承関係などは一切持たない新しい`naoqi_bridge_msgs`パッケージを作って，
それが`naoqi_bridge_msgs::xxx`, `naoqi_msgs::xxx`としても扱えるようにする．
だから，`naoqi_bridge_msgs::JointAnglesWithSpeed`などがない．

### 消してよいか

勝手に消さない．
元は，hydroで`naoqi_msgs`しかなかった頃の処置なので，
その意図した動作はちゃんと作るべき（実際にテストが通っていたかは，今となっては怪しい・・・）

- 案１．実際は，`(ros::load-ros-manifest "peppereus")`の内部で，
`make-package`している部分があるはず．そこで，:nickname `naoqi_msgs`するとよい？

- 案２．using namespace std; のような機能がある(`use-package`)それを使えば，そもそも`naoqi_bridge_msgs`や`naoqi_msgs`の違いを気にしなくてよくなるはず

- [ ] この部分のソースコードにコメントを書く
- [ ] 直し方を調べる・聞く

@pazeshun ありがとうございます！

kochigami, 2019-04-15:
k-okada先生に伺って，確かに意図したように動いていないようだということを確認した．
現在はこの部分が動かなくても困らないので（hydro用なので），プログラムにコメントを書くことにする．
パッケージ名変更で，同じように困る場合もあるかもしれないため．

### まとめ

std_msgsのmsgを読み込める，std_msgs2パッケージを作りたい
存在しない`std_msgs2`のメッセージを読み込もうと試みる

|プログラム名  |説明  |読み込めるか  |
|---|---|---|
|test0.l|何もしない  |x |
|test1.l|`make-package`で`std_msgs2`を作ってみる |x  |
|test2.l|`make-package`で`std_msgs2`を作ってみる，`ros::advertise`を関数にしてみる | x|
|test3.l|std_msgs2::Boolの定義を直書きする, `(make-package "STD_MSGS2" :nicknames "STD_MSGS")` なし | x|
|test4.l|std_msgs2::Boolの定義を直書きする`，(make-package "STD_MSGS2" :nicknames "STD_MSGS")` あり| o |

### test0.l  （何もしない）
```
(ros::load-ros-manifest "roseus")                                                                                                                     
(ros::roseus "hoge")
(defun hoge nil
  (ros::advertise "hoge" std_msgs::bool 1)
  (ros::advertise "fuga" std_msgs2::bool 1)
  )
```
=> `roseus test0.l`で`/opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 no such package "STD_MSGS2""STD_MSGS2" in #<compiled-code #X59ac038>E:`

### test1.l　(`make-package`で`std_msgs2`を作ってみる)
```
(ros::load-ros-manifest "roseus")
(make-package "STD_MSGS2" :nicknames "STD_MSGS") ; std_msgs2という名前でもstd_msgsのmsgを読み込みたい
(ros::roseus "hoge")
(ros::advertise "hoge" std_msgs::bool 1)
(ros::advertise "fuga" std_msgs2::bool 1)
```
=> std_msgs2という名前でstd_msgsのmsgを読み込めない
```
roseus test1.l

/opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 unbound variable std_msgs2::boolstd_msgs2::bool in (ros::advertise "fuga" std_msgs2::bool 1)E:
```
### test2.l (`make-package`で`std_msgs2`を作ってみる，`ros::advertise`を関数にしてみる)
```
(ros::load-ros-manifest "roseus")
(make-package "STD_MSGS2" :nicknames "STD_MSGS")　; std_msgs2という名前でもstd_msgsのmsgを読み込みたい
(ros::roseus "hoge")
(defun hoge nil
  (ros::advertise "hoge" std_msgs::bool 1)
  (ros::advertise "fuga" std_msgs2::bool 1)
  )
```
=> `roseus test2.l`は`/opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 unbound variable 　　 in #<compiled-code #X5dd9038>E: `

### test3.l  （std_msgs2::Boolの定義を直書きする, `(make-package "STD_MSGS2" :nicknames "STD_MSGS")` なし）
```
(ros::load-ros-manifest "roseus")
(ros::roseus "hoge")

---
ここに`/opt/ros/kinetic/share/roseus/ros/std_msgs/msg/Bool.l`の中身を`std_msgs`=>`std_msgs2`に変えてそのまま貼り付ける
---  

(defun hoge nil
  (ros::advertise "hoge" std_msgs::bool 1)
  (ros::advertise "fuga" std_msgs2::bool 1)
  )

(hoge)
```
=> `roseus test3.l`　で `/opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 no such package "STD_MSGS2""STD_MSGS2" in #<compiled-code #X6014038>E:`

### test4.l　（std_msgs2::Boolの定義を直書きする）
```
(ros::load-ros-manifest "roseus")
(make-package "STD_MSGS2" :nicknames "STD_MSGS")
(ros::roseus "hoge")

---
ここに`/opt/ros/kinetic/share/roseus/ros/std_msgs/msg/Bool.l`の中身を`std_msgs`=>`std_msgs2`に変えてそのまま貼り付ける
---  

(defun hoge nil
  (ros::advertise "hoge" std_msgs::bool 1)
  (ros::advertise "fuga" std_msgs2::bool 1)
  )

(hoge)
```
=> `roseus test4.l`は読み込める

closed 2019-04-15 by kochigami