jsk-ros-pkg/jsk_robot #1060: [pr1040] respeaker node does not start

knorth55, 2019-04-18:
USB connector is not good.
we need to redesign the attachment

k-okada, 2019-04-19:
where is the cad data?
we should put them in
https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_pr2_robot/jsk_pr2_accessories
--
◉ Kei Okada

knorth55, 2019-04-19:
I add more information about this issue.
this issue happens because USB connecter in ReSpeaker mic got loose.
Current attachment CAD is in Google team drive.
Should I manage it in github?

k-okada, 2019-04-19:
> Should I manage it in github?

yes
--
◉ Kei Okada

knorth55, 2019-04-19:
OK i will upload all CAD data.

knorth55, 2019-04-19:
I upload all cad data https://github.com/jsk-ros-pkg/jsk_robot/pull/1064

knorth55, 2019-04-24:
if this issue happens, please open this issues again.

closed 2019-04-24 by knorth55