jsk-ros-pkg/jsk_robot #1079: [jsk_pepper_startup] script/roslaunch-checkでreleaseされていないパッケージが落とされる

kochigami, 2019-05-11:
### 環境

Ubuntu 16.04, ROS kinetic

### 困ったこと

`naoqi_dashboard`と`naoqi_apps`(`naoqi_bridge`)で，kochigami-developという名前でコミットしていたものを`jsk_pepper_startup.launch`にも使いたい．
（そもそも独自のブランチを切っている理由は，ros-naoqiのパッケージの開発がゆっくり目なのと，自分の変更をmasterにPRする前に，手元で試した方が良いと思うため．）
しかし，テストの時はこれらのパッケージとして/opt/ros/kinetic/...以下のlaunchを使うので，
例えば以下のエラーが出る．
エラーが出てもテストを合格にするか，テストを免除して欲しい．

- `basic_awareness.launch`はkochigami-developにしかないからエラーになる
```
roslaunch.depends.RoslaunchDepsException: Cannot load roslaunch include '/opt/ros/indigo/share/naoqi_apps/launch/basic_awareness.launch' in '/home/travis/ros/ws_jsk_robot/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/launch/jsk_pepper_startup.launch'
```
- `launch_gui`引数はkochigami-developmentにしかないからエラーになる
```
<?xml version="1.0" encoding="UTF-8"?>

<testsuite tests="1" failures="1" time="1" errors="0" name="roslaunch-check_jsk_pepper_startup__ROS_IP_10.68.0.2 NAO_IP_10.68.0.3.launch.xml">

  <testcase name="test_ran" status="run" time="1" classname="Results">

  <failure message="roslaunch check [/home/travis/ros/ws_jsk_robot/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/launch/jsk_pepper_startup.launch] failed" type=""/>

  </testcase>

  <system-out><![CDATA[[

[/home/travis/ros/ws_jsk_robot/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/launch/jsk_pepper_startup.launch]:

	unused args [launch_gui] for include of [/opt/ros/indigo/share/naoqi_dashboard/launch/naoqi_dashboard.launch]

]]></system-out>

</testsuite>
```

### 調べたこと

`script/roslaunch-check`の挙動(print文で変数の中身を表示していった)
https://docs.ros.org/api/roslaunch/html/roslaunch.depends-module.html#roslaunch_deps
（ちなみに，docの中では，`roslaunch_deps`は`packages`変数を持つとあるが，実際はないと思う．）

### 分かったこと

`jsk_pepper_startup/script/roslaunch-check`は，デフォルトの引数がfalseになっていて実行しないことになっているファイルのテストを行わない．
例えば，
```
<arg name="launch_play_audio_stream" default="false" />
<node pkg="jsk_pepper_startup" name="play_audio_stream" type="play_audio_stream.py" respawn="true" if="$(arg launch_play_audio_stream)" />
```
になっていたら，テストを行わない．

- naoqi_appsに関しては，実は途中から`<arg name="autonomous_enabled" default="false" />`にしたのでテストを行わなくなって，問題なくなった．

- naoqi_dashboardに関しては，引数をfalseにすればテストを行わなくなるが，デフォルトではtrueにしたい．
そこで，テストに使うファイルを別に用意し，そのファイルでの`launch_dashboard`引数はfalseにすればテストを行わなくなって，エラーは出なくなるはず・・・ => https://github.com/jsk-ros-pkg/jsk_robot/pull/1074/commits/d5e78195ffc871c7c8af44c16e9d26d11f2f0e39

kochigami, 2019-05-11:
```
Traceback (most recent call last):
                                                                          
  File "/home/travis/ros/ws_jsk_robot/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/script/roslaunch-check", line 150, in <module>
                                                                                
    error_msg = check_roslaunch_file(roslaunch_path)
                                                                                
  File "/home/travis/ros/ws_jsk_robot/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/script/roslaunch-check", line 110, in check_roslaunch_file
                                                                                
    error_msg = check_roslaunch(roslaunch_file)
                                                                                
  File "/home/travis/ros/ws_jsk_robot/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/script/roslaunch-check", line 63, in check_roslaunch
                                                                                
    base_pkg, file_deps, missing = roslaunch.depends.roslaunch_deps([f])
                                                                                
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/depends.py", line 316, in roslaunch_deps

  rl_file_deps(file_deps, launch_file, verbose)
                                                                               
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/depends.py", line 217, in rl_file_deps
                                                                                
    parse_launch(launch_file, file_deps, verbose)
                                                                                
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/depends.py", line 202, in parse_launch
                                                                                
    _parse_launch(launch_tag.childNodes, launch_file, file_deps, verbose, context)
                                                                                
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/depends.py", line 172, in _parse_launch
                                                                                
    _parse_launch(launch_tag.childNodes, sub_launch_file, file_deps, verbose, sub_context)
                                                                                
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/depends.py", line 137, in _parse_launch
                                                                                
    _parse_launch(tag.childNodes, launch_file, file_deps, verbose, context)
                                                                                
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/depends.py", line 174, in _parse_launch
                                                                                
    raise RoslaunchDepsException("Cannot load roslaunch include '%s' in '%s'"%(sub_launch_file, launch_file))

 roslaunch.depends.RoslaunchDepsException: Cannot load roslaunch include '/opt/ros/indigo/share/naoqi_apps/launch/basic_awareness.launch' in '/home/travis/ros/ws_jsk_robot/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/launch/jsk_pepper_startup.launch'

 Cannot find results, writing failure results to '/home/travis/ros/ws_jsk_robot/build/jsk_pepper_startup/test_results/jsk_pepper_startup/MISSING-roslaunch-check_roslaunch-check-pepper-startup__ROS_IP_10.68.0.2 NAO_IP_10.68.0.3.xml.xml'
```

k-okada, 2019-05-12:
１）https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/jsk_pepper_startup/CMakeLists.txt#L16
を完全にコメントアウトする
２）jsk_robot/.travis.rosinstall.indigo
に必要なバージョン／ブランチを書いておく（travisテスト時にこれがソースでインストールされる容認なる）（参考：https://github.com/jsk-ros-pkg/jsk_robot/blob/master/.travis.rosinstall.kinetic）
３）  jsk_pepper_startup.launchの`<arg name="launch_joy" default="true" />`
みたいにlaunch_gui default=trueみたいにしておいて，CMakeLists.txtのなかで
  `roslaunch_add_file_check(launch/jsk_pepper_startup.launch "ROS_IP=10.68.0.2 NAO_IP=10.68.0.3" launch_gui=false)`
みたいにする

kochigami, 2019-05-13:
ありがとうございます．
2) の形で一旦やってみました：https://github.com/jsk-ros-pkg/jsk_robot/pull/1074/commits/7df220310eff61476dc42011232353c4ba05553e

> ２）jsk_robot/.travis.rosinstall.indigo
に必要なバージョン／ブランチを書いておく（travisテスト時にこれがソースでインストールされる容認なる）（参考

について，
- .travis.ymlの`allow_failures`に入っていない，`ROS_DISTRO=indigo  USE_DEB=true`でテストを通さないといけないから，2が必要だと分かりました．
- .travis/travis.shを実行して，https://github.com/jsk-ros-pkg/jsk_travis/blob/3651899d45e151f47bea20cd0e88d1abb12941b3/travis.sh#L234 で2の.rosinstallを読み込むようになると分かりました．

kochigami, 2019-05-13:
すみません，

> .travis/travis.shを実行して，https://github.com/jsk-ros-pkg/jsk_travis/blob/3651899d45e151f47bea20cd0e88d1abb12941b3/travis.sh#L234 で2の.rosinstallを読み込むようになると分かりました．

はUSE_DEB=falseの時だけでした．https://github.com/jsk-ros-pkg/jsk_travis/blob/3651899d45e151f47bea20cd0e88d1abb12941b3/travis.sh#L227
3の作戦で行こうと思います．

エラー
```
`unused args [launch_gui] for include of [/opt/ros/indigo/share/naoqi_dashboard/launch/naoqi_dashboard.launch]`
```

kochigami, 2019-05-13:
ググり方が悪いのか，引数の与え方を調べても分からない
多分下のような書き方だと引数を渡せていない？
```
roslaunch_add_file_check(launch/jsk_pepper_startup.launch "ROS_IP=10.68.0.2 NAO_IP=10.68.0.3" launch_dashboard=false)
```
=>
```
[/home/travis/ros/ws_jsk_robot/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/launch/jsk_pepper_startup.launch]:

	unused args [launch_gui] for include of [/opt/ros/indigo/share/naoqi_dashboard/launch/naoqi_dashboard.launch]
```

kochigami, 2019-05-13:
```
roslaunch_add_file_check(launch/jsk_pepper_startup.launch "ROS_IP=10.68.0.2 NAO_IP=10.68.0.3 launch_dashboard=false" )
```
=> 
```
unused args [launch_gui] for include of [/opt/ros/indigo/share/naoqi_dashboard/launch/naoqi_dashboard.launch]
```

kochigami, 2019-05-14:
CMakeLists.txt
```
roslaunch_add_file_check(test/jsk_pepper_startup.test.xml "ROS_IP=10.68.0.2 NAO_IP=10.68.0.3")
```
jsk_naoqi_robot/jsk_pepper_startup/test/jsk_pepper_startup.test.xml
```
<launch>
  <include file="$(find jsk_pepper_startup)/launch/jsk_pepper_startup.launch">
    <arg name="launch_dashboard" value="false" />
  </include>
</launch>
```
=>
```
Cannot load roslaunch include '/opt/ros/indigo/share/naoqi_apps/launch/basic_awareness.launch' in '/home/travis/ros/ws_jsk_robot/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/launch/jsk_pepper_startup.launch'
```

kochigami, 2019-05-14:
naoqi_appsを使わないで，テスト用のlaunchファイルを作った方が良いかもしれない

```
<launch>
  <include file="$(find jsk_pepper_startup)/launch/jsk_pepper_startup.launch">
    <arg name="launch_dashboard" value="false" />
    <arg name="autonomous_enabled" value="false" />
  </include>
</launch>
```
=>
```
roslaunch.depends.RoslaunchDepsException: Cannot load roslaunch include '/opt/ros/indigo/share/naoqi_apps/launch/basic_awareness.launch' in '/home/travis/ros/ws_jsk_robot/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/launch/jsk_pepper_startup.launch'
```

kochigami, 2019-05-15:
結局，テスト用のlaunchファイルを作った：
- CMakeLists.txt
```
roslaunch_add_file_check(test/jsk_pepper_startup.launch "ROS_IP=10.68.0.2 NAO_IP=10.68.0.3")
```
- test/jsk_pepper_startup.launch

naoqi_dashboardは起動しない設定に
（起動する設定(true)にすると，naoqi_dashboardもテストされる．
naoqi_dashboard.launchはdebにも存在するので，「ファイルがない」というエラーは出ない．
しかし，`launch_gui`という引数はkochigami-developにしかないのでエラーが出る．）
```
<arg name="launch_dashboard" default="false" />
```
naoqi_apps/basic_awareness/ background_movement.launchをコメントアウト
naoqi_appsのdebは存在するが，これらのlaunchは含まれないのでテストで「ファイルがない」といエラーが出る．起動しない設定(false)にしても，このエラーは出る．
```
  <!-- <group if="$(arg autonomous_enabled)"> -->
  <!--   <include file="$(find naoqi_apps)/launch/basic_awareness.launch" /> -->
  <!--   <node pkg="jsk_pepper_startup" type="start_basic_awareness.py" name="start_basic_awareness" /> -->
  <!--   <include file="$(find naoqi_apps)/launch/background_movement.launch" /> -->
  <!--   <node pkg="jsk_pepper_startup" type="start_background_movement.py" name="start_background_movement" /> -->
  <!-- </group> -->
```

closed 2019-05-15 by kochigami