jsk-ros-pkg/jsk_robot #1154: [fetcheus] go-velocity does not work

MiyabiTane, 2019-10-21:
```
(send *ri* :go-velocity 0.1 0 0.5)
[ERROR] [1571647540.324682228]: Setting velocity in y-axis is not valid for non-holonomic robot
nil
```

cc @Affonso-Gui  @YutoUchimi

knorth55, 2019-10-21:
FYI: PR2 is holonomic robot and fetch is non-holonomic robot.

pazeshun, 2019-10-22:
I understand what is happening.
@MiyabiTane I'll fix the critical issue, but please be careful when using `:go-velocity` even after it is fixed, because this method doesn't seem used in real robot.
Sorry for your trouble...

The following source code is for catching a case in which second argument (velocity in y direction) **is not 0**:
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_fetch_robot/fetcheus/fetch-interface.l#L212-L215
This is because fetch cannot move in y direction (non-holonomic).
However, the above code is wrong as it works when second argument **is 0** (`unless (not (zerop y)`).

PR #1156 (2019-10-23): [fetcheus] Fix :go-velocity to reject non-zero y correctly
Includes #1155 
Fixes #1154
closed 2019-10-23 by k-okada