jsk-ros-pkg/jsk_robot #1211: nao.lが生成されない

Kanazawanaoaki, 2020-04-04:
https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot に従って進めていってnaoeusを使ってみようと思い、roseusで  
```lisp
(load "package://naoeus/nao-interface.l")
```
をしたところ、  
```
/opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error:  file #P"/home/kanazawa/nao_ws/src/jsk_robot/jsk_naoqi_robot/naoeus/nao.l" not found in (apply #'ros::load-org-for-ros ros::fullname args)
```
のエラーが出て、naoeusにnao.lが無いと怒られました。  

naoeusの[CMakeList.txt](https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/naoeus/CMakeLists.txt)にnao.l generationの項目があるので、catkin buildが上手くいっていないせいなのだと考えました。

catkin build --continue-on-failure
をした時に、
naoqi_driverのおそらくこの部分https://github.com/kochigami/naoqi_driver/blob/kochigami-develop/src/helpers/driver_helpers.hpp が呼ばれている時に、  
#include <nao_interaction_msgs/AudioPlayback.h>がないと怒られていました。
nao_interaction_msgs https://github.com/ros-naoqi/nao_interaction のmsgを見ると確かにAudioPlayback.hは無いので、  
ひとまずはnaoqi_driverのブランチをmasterに変更してもう一度catkin buildをしました。  
catkin buildはエラーなく成功しましたが、まだnao.lは生成されていなくてnaoeusを使おうとすると同じエラーが出ます。  

sudo apt-get install ros-melodic-nao-meshesはしましたし、
CMakeListsにかかれているcompile_naoqi_model(nao naoV50_generated_urdf)のnaoV50_generated_urdfを探すと、nao_descriptionパッケージのurdfというフォルダの中にnaoV50_generated_urdfはありました。
それが上手く読まれていないのが問題なのでしょうか？

pazeshun, 2020-04-04:
nao_interaction_msgs/AudioPlayback.hは、nao_interaction_msgsをビルドしたときに、AudioPlayback.srvから自動生成されるファイルです。
nao_interaction_msgsがビルドされていないのではないでしょうか。
もう一度、ワークスペース構成などをデフォルトに戻して、catkin buildした時のログを全て見せてください。

Kanazawanaoaki, 2020-04-04:
なるほど。ありがとうございます。
naoqi_driverのブランチをkochigami-developに戻して、  
catkin buildをしたら次のようになりました。一回buildしてしまったせいかエラーが変わってしまっているのでもう一度最初からやった方が良いですかね？

kanazawa@kanazawa-ThinkPad-T495s:~/nao_ws
$ catkin build --continue-on-failure
----------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/kanazawa/nao_ws
----------------------------------------------------------
Build Space:        [exists] /home/kanazawa/nao_ws/build
Devel Space:        [exists] /home/kanazawa/nao_ws/devel
Install Space:      [unused] /home/kanazawa/nao_ws/install
Log Space:          [exists] /home/kanazawa/nao_ws/logs
Source Space:       [exists] /home/kanazawa/nao_ws/src
DESTDIR:            [unused] None
----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------
[build] Found '46' packages in 0.0 seconds.                                                         
[build] Package table is up to date.                                                                
Starting  >>> aeroeus                                                                               
Starting  >>> fetcheus                                                                              
Starting  >>> jsk_baxter_startup                                                                    
Starting  >>> jsk_baxter_web                                                                        
Starting  >>> jsk_fetch_diagnosis                                                                   
Starting  >>> jsk_pr2_calibration                                                                   
Starting  >>> jsk_robot_startup                                                                     
Starting  >>> jsk_robot_utils                                                                       
Finished  <<< jsk_robot_utils                           [ 0.2 seconds ]                             
Starting  >>> nao_description                                                                       
Finished  <<< jsk_baxter_web                            [ 0.2 seconds ]                             
Starting  >>> nao_interaction_msgs                                                                  
Finished  <<< jsk_robot_startup                         [ 0.4 seconds ]                             
Starting  >>> naoqi_apps                                                                            
Finished  <<< fetcheus                                  [ 0.3 seconds ]                             
Starting  >>> naoqi_bridge_msgs                                                                     
Finished  <<< jsk_fetch_diagnosis                       [ 0.7 seconds ]                             
Starting  >>> naoqi_tools                                                                           
Finished  <<< jsk_baxter_startup                        [ 0.2 seconds ]                             
Starting  >>> pepper_description                                                                    
Finished  <<< aeroeus                                   [ 0.2 seconds ]                             
Starting  >>> pr2_base_trajectory_action                                                            
Finished  <<< jsk_pr2_calibration                       [ 0.3 seconds ]                             
Starting  >>> roseus_remote                                                                         
Finished  <<< nao_description                           [ 0.3 seconds ]                             
Starting  >>> jsk_fetch_startup                                                                     
Finished  <<< naoqi_apps                                [ 0.2 seconds ]                             
Finished  <<< naoqi_tools                               [ 0.2 seconds ]                             
Finished  <<< pepper_description                        [ 0.2 seconds ]                             
Finished  <<< pr2_base_trajectory_action                [ 0.3 seconds ]                             
Finished  <<< jsk_fetch_startup                         [ 0.3 seconds ]                             
Finished  <<< roseus_remote                             [ 0.5 seconds ]                             
Finished  <<< nao_interaction_msgs                      [ 1.6 seconds ]                             
Starting  >>> baxtereus                                                                             
Starting  >>> jsk_baxter_desktop                                                                    
Starting  >>> jsk_aero_startup                                                                      
[... 473 lines cut ...]
/opt/ros/melodic/include/qi/detail/future_fwd.hpp:750:5: note: declared here
     operator const typename Future<T>::ValueTypeCast&() const          { _sync = false; return _future.value(); }
[... cut ...]
[build] Summary: 32 of 41 packages succeeded.                                                       
[build]   Ignored:   5 packages were skipped or are blacklisted.                                    
[build]   Warnings:  None.                                                                          
[build]   Abandoned: 8 packages were abandoned.                                                     
[build]   Failed:    1 packages failed.                                                             
[build] Runtime: 46.8 seconds total.

kochigami, 2020-04-04:
`nao.l`が生成されないのは，`naoqi_driver`のコンパイルが失敗して，それに依存するとされている`naoeus`のコンパイルが行われていないからかなと思います．
（`Abandoned <<< naoeus                                    [ Depends on failed job naoqi_driver ] `）
（`naoeus`だけコンパイルすると，`nao.l`は出来るかもしれないです．）

`naoqi_driver`のコンパイルが失敗するのは，
```
Errors << naoqi_driver:make /home/kanazawa/nao_ws/logs/naoqi_driver/build.make.011.log
/home/kanazawa/nao_ws/src/naoqi_driver/src/helpers/driver_helpers.cpp: In function ‘naoqi_bridge_msgs::RobotInfo& naoqi::helpers::driver::getRobotInfoLocal(const SessionPtr&)’:
/home/kanazawa/nao_ws/src/naoqi_driver/src/helpers/driver_helpers.cpp:43:55: error: ‘qi::FutureSync::operator const typename qi::Future::ValueTypeCast&() const [with T = qi::Objectqi::Empty; typename qi::Future::ValueTypeCast = qi::Objectqi::Empty]’ is deprecated: "Use either then, andThen, value or wait functions instead." [-Werror=deprecated-declarations]
qi::AnyObject p_memory = session->service("ALMemory");
```
などのdeprecatedと言われている所が怪しいのかなと思いました．
お手数をお掛けしますが，`nao_ws/build`, `nao_ws/devel` を消してもう一度 `catkin build --continue-on-failure`した時のログが貼って頂いたものと同じか確認してくださいませんか．

私はmelodicで環境構築したことがなく，もしこの辺りに詳しい方がいらっしゃったら助けて頂きたいです．

Kanazawanaoaki, 2020-04-04:
ありがとうございます。  
nao_ws/build,nao_ws/develを消してもう一度catkin build --continue-on-failureをしたところ、
```
Errors     << naoqi_driver:make /home/kanazawa/nao_ws/logs/naoqi_driver/build.make.000.log          
/home/kanazawa/nao_ws/src/naoqi_driver/src/event/touch.cpp: In instantiation of ‘naoqi::TouchEventRegister<T>::TouchEventRegister(const string&, std::vector<std::__cxx11::basic_string<char> >, const float&, const SessionPtr&) [with T = naoqi_bridge_msgs::Bumper_<std::allocator<void> >; std::__cxx11::string = std::__cxx11::basic_string<char>; qi::SessionPtr = boost::shared_ptr<qi::Session>]’:
/home/kanazawa/nao_ws/src/naoqi_driver/src/event/touch.hpp:119:220:   required from here
/home/kanazawa/nao_ws/src/naoqi_driver/src/event/touch.cpp:48:21: error: ‘qi::FutureSync<T>::operator const typename qi::Future<T>::ValueTypeCast&() const [with T = qi::Object<qi::Empty>; typename qi::Future<T>::ValueTypeCast = qi::Object<qi::Empty>]’ is deprecated: "Use either `then`, `andThen`, `value` or `wait` functions instead." [-Werror=deprecated-declarations]
     isDumping_(false)
```
から始まるようになりましたが、出ているエラーの順番が変わっただけで、出ているエラーは同じようです。  

上のエラーは一度無理やりコンパイルしてしまったために起こっているエラーな気もするので、別にワークスペースを作ってもう一度最初から実行してみることにしました。[setup-environment](https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot#setup-environment)のFinally, please compile them.の部分で、srcの下にjsk_robotとnaoqi_dashboardのみの状況でcatkin build --continue-on-failureをして、
```
Warnings   << naoeus:cmake /home/kanazawa/naoqi_ws/logs/naoeus/build.cmake.000.log                  
-- set EUSDIR to /opt/ros/melodic/share/euslisp/jskeus/eus
-- set ARCHDIR to Linux64
CMake Warning at /home/kanazawa/naoqi_ws/src/jsk_robot/jsk_naoqi_robot/naoeus/CMakeLists.txt:5 (find_package):
  By not providing "Findnao_description.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "nao_description", but CMake did not find one.

  Could not find a package configuration file provided by "nao_description"
  with any of the following names:

    nao_descriptionConfig.cmake
    nao_description-config.cmake

  Add the installation prefix of "nao_description" to CMAKE_PREFIX_PATH or
  set "nao_description_DIR" to a directory containing one of the above files.
  If "nao_description" provides a separate development package or SDK, be
  sure it has been installed.

CMake Warning at /home/kanazawa/naoqi_ws/src/jsk_robot/jsk_naoqi_robot/naoqieus/cmake/compile_naoqi_model.cmake:8 (message):
  Could not found nao.urdf in
  /share/nao_description/urdf/naoV50_generated_urdf/nao.urdf and
  /urdf/naoV50_generated_urdf/nao.urdf
Call Stack (most recent call first):
  CMakeLists.txt:12 (compile_naoqi_model)

cd /home/kanazawa/naoqi_ws/build/naoeus; catkin build --get-env naoeus | catkin env -si  /usr/bin/cmake /home/kanazawa/naoqi_ws/src/jsk_robot/jsk_naoqi_robot/naoeus --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/kanazawa/naoqi_ws/devel/.private/naoeus -DCMAKE_INSTALL_PREFIX=/home/kanazawa/naoqi_ws/install; cd -
```
というWarningsがnaoeusの部分で出ていて、Warningsは沢山出ていますがcatkin build自体はエラーが出ずに終わりました。  

wstool set nao_robot --git https://github.com/ros-naoqi/nao_robot など、続きも進めてからもう一度コンパイルをしてみます。

Kanazawanaoaki, 2020-04-04:
新しく作ったワークスペースでも、貼ってあるものと同じエラーが出ました。  
最初に出たAudioPlayback.hは何かの原因で置きたnao_interaction_msgsのコンパイル失敗によるものだったのだと思います。  
今でいるエラーについては、この[issue](https://github.com/ros-naoqi/naoqi_driver/issues/114)を見るとmelodicであるから起こるエラーで、少し前にROS melodic版が出たようなので明日見てみようと思います。

closed 2020-04-04 by Kanazawanaoaki

pazeshun, 2020-04-05:
@Kanazawanaoaki 加えてくれたcommitが君のブランチにしか入ってないと思うので、kochigami-developブランチにPull Requestを出してみて下さい。
後、該当commitが君の名義になってる気がするけど、オリジナルのcommitをgit cherry-pickで取ってこれないかな？

Kanazawanaoaki, 2020-04-05:
git cherry-pickをしてPull Requestを出してみました。
https://github.com/kochigami/naoqi_driver/pull/10

kochigami, 2020-04-06:
ありがとうございます．
ros-naoqiのnaoqi_driverパッケージは最近色々と改良が進んでいるようなのですが，私はあまり追えていないので，他にも何かあったらおっしゃってください．
（masterブランチを使うのが一番望ましいと思うので，時間が出来たら独自ブランチの解消に取り組もうと思います．）