jsk-ros-pkg/jsk_robot #1216: run fetch-interface.l without roscore

takayuki5168, 2020-04-18:
There is an initialization of a ros node in [`fetch-interface.l`](https://github.com/jsk-ros-pkg/jsk_robot/blob/95959dc105b2d2c90d1d5d1bad49cd2a6560aa21/jsk_fetch_robot/fetcheus/fetch-interface.l#L1), so we cannot run `fetch-interface.l` without `roscore`.
Are there any reasons to initialize the ros node?

In `pr2-interface.l`, there is no initialization of ros node, which I think much more convenient.

knorth55, 2020-04-18:
maybe not.
can you run fetch in kinematics simulator without ros?
if so, it is better to remove the initialization.

takayuki5168, 2020-04-18:
@knorth55 
Kinematics simulator means using `robot-interface` right?
We cannot generate an instance of `robot-interface` without ros because it uses ros publisher or subscriber. So when we use `robot-interface`, we have to run roscore.

knorth55, 2020-04-19:
so, why do you want to remove the line?

takayuki5168, 2020-04-19:
It is convenient when we only generate motion in euslisp, which is independent of ros,..., but I found that when we want to do that, we could realize by loading not "pr2-interface.l" or "fetch-interface.l" but "pr2.l" or "fetch.l".

However, we should unify whether we initialize the ros node or not in "pr2-interface.l" and "fetch-interface.l". I think both is fine.

Affonso-Gui, 2020-04-19:
I do agree on moving the node initialization to `fetch-init` in resemble to the pr2-interface

Not very relevant, but this can cause misleading behavior in cases like the below example, in which the node would be initialized as "fetch" rather than as "my_node"
```
(load "package://fetcheus/fetch-robot.l")
(ros::roseus "my_node")
(fetch-init)
```

Also, pr2 uses "pr2_eus_interface" and fetch uses "fetch". We could unify those as well.

takayuki5168, 2020-04-21:
@Affonso-Gui 
Thanks for your favorable comment.

In `fetch-interface.l` the ros node is about to be initialized again as `fetch_eus_interface`, but ignored because the ros node has been already initialized as `fetch` in the first line.

I made a simple PR for this. https://github.com/jsk-ros-pkg/jsk_robot/pull/1218

knorth55, 2020-04-21:
IMO, if it causes some error, we need to remove it, but it not, it is personal preference.
I prefer removing the line, but I don't strongly recommend it.

Affonso-Gui, 2020-04-21:
Closing this issue and moving the discussion to https://github.com/jsk-ros-pkg/jsk_robot/pull/1218

closed 2020-04-21 by Affonso-Gui