jsk-ros-pkg/jsk_robot #1221: Fetch15でformat 3のpackage.xmlがbuildできない

knorth55, 2020-05-27:
Fetch15でformat 3のpackage.xmlが含まれたパッケージをビルドすると以下のようなエラーがでる。

```
fetch@fetch15:~/ros/indigo/src/ros-drivers/audio_common$ catkin b audio_common
==> Expanding alias 'b' from 'catkin b audio_common' to 'catkin build audio_common'
-----------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/fetch/ros/indigo
-----------------------------------------------------------
Source Space:       [exists] /home/fetch/ros/indigo/src
Log Space:          [exists] /home/fetch/ros/indigo/logs
Build Space:        [exists] /home/fetch/ros/indigo/build
Devel Space:        [exists] /home/fetch/ros/indigo/devel
Install Space:      [unused] /home/fetch/ros/indigo/install
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
Traceback (most recent call last):
  File "/usr/bin/catkin", line 9, in <module>
    load_entry_point('catkin-tools==0.4.4', 'console_scripts', 'catkin')()
  File "/usr/lib/python2.7/dist-packages/catkin_tools/commands/catkin.py", line 267, in main
    catkin_main(sysargs)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/commands/catkin.py", line 262, in catkin_main
    sys.exit(args.main(args) or 0)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/cli.py", line 420, in main
    summarize_build=opts.summarize  # Can be True, False, or None
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/build.py", line 283, in build_isolated_workspace
    workspace_packages = find_packages(context.source_space_abs, exclude_subspaces=True, warnings=[])
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 86, in find_packages
    packages = find_packages_allowing_duplicates(basepath, exclude_paths=exclude_paths, exclude_subspaces=exclude_subspaces, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 153, in find_packages_allowing_duplicates
    path_parsed_packages, warnings_lists = zip(*pool.map(parser, data))
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 251, in map
    return self.map_async(func, iterable, chunksize).get()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 558, in get
    raise self._value
AssertionError: Unable to handle package.xml format version '3', please update catkin_pkg (e.g. on Ubuntu/Debian use: sudo apt-get update && sudo apt-get install --only-upgrade python-catkin-pkg)
```

原因としてはfetchroboticsのpackage.fetchrobotics.comが更新されていないため、`python-catkin-pkg`が`0.3.9`のままなことが原因と思われる。
http://packages.fetchrobotics.com/ros/ubuntu/pool/main/p/python-catkin-pkg/
参考: http://packages.ros.org/ros/ubuntu/pool/main/p/python-catkin-pkg/

```
fetch@fetch15:~$ cat /etc/apt/sources.list.d/ros-latest.list
deb http://packages.fetchrobotics.com/ros/ubuntu/ trusty main
# deb http://packages.ros.org/ros/ubuntu trusty main
```

解決策：
1. たえる
2. aptではなくpipで最新のcatkin-pkgをいれる
3. `/etc/apt/sources.list.d/ros-latest.list`を`packages.ros.org`に変更する

@k-okada どれがいいとおもいますか？

cc. @708yamaguchi

knorth55, 2020-05-27:
関連PR: https://github.com/fetchrobotics/fetch_ros/pull/120

> 4. なんとかして`packages.fetchrobotics.com`を更新してもらう

https://github.com/fetchrobotics/fetch_ros/pull/120#issuecomment-485516897
このコメントで`packages.fetchrobotics.com`を更新するつもりがないと言及されているので厳しいか……？

708yamaguchi, 2020-05-27:
良いタイミングだからmelodicに上げてみたいなぁとも思いますが、テンポラリな解決策は、
> 4. なんとかしてpackage.fetchrobotics.comを更新してもらう

に淡い期待をいだきつつ、

> 2. aptではなくpipで最新のcatkin-pkgをいれる

をしてjsk_fetch_startupのREADMEに書いておくのが良いのでしょうか。

以下のfetch roboticsのマニュアルを見るに、現段階では18.04のfetchは独自のミラーサーバーを見るのではなくpackage.ros.orgを参照しているようです。
https://docs.fetchrobotics.com/indigo_to_melodic.html#manual-approach-18-04-install-and-installing-ros-fetch-packages
なので、package.ros.orgにかかれているpython-catkin-pkgのバージョンを確認して、そのバージョンをpipでインストールするのが良いのかも知れません。

k-okada, 2020-05-27:
>    1. なんとかしてpackage.fetchrobotics.comを更新してもらう
>    1. aptではなくpipで最新のcatkin-pkgをいれる

pip でもいいけど，fetchの中でpython-catkin-pkg
のdebを作り直すんじゃないでしょうか．PR2ではなんかのパッケージでそれをやっていたんだけど，どこに書いてあったか．．．

wget
http://packages.ros.org/ros/ubuntu/pool/main/p/python-catkin-pkg-modules/python-catkin-pkg-modules_0.4.8-1.debian.tar.xz
wget
http://packages.ros.org/ros/ubuntu/pool/main/p/python-catkin-pkg-modules/python-catkin-pkg-modules_0.4.8.orig.tar.gz
 tar -xvzf python-catkin-pkg-modules_0.4.8.orig.tar.gz
 cd catkin_pkg_modules-0.4.8/
emacs -nw debian/control # emove python-all ... from build-depend
xzcat  ../python-catkin-pkg-modules_0.4.8-1.debian.tar.xz | tar xfv -
dpkg-buildpackage  -uc -us

ただ，今やってみたら，普通にインストールできそう．
$ wget
http://packages.ros.org/ros/ubuntu/pool/main/p/python-catkin-pkg-modules/python-catkin-pkg-modules_0.4.8-1_all.deb
$ dpkg -i python-catkin-pkg-modules_0.4.8-1_all.deb
$ sudo dpkg -i python-catkin-pkg-modules_0.4.8-1_all.deb

万が一間違えたときにapt remove したら，かならず消せるという利点がある．
pipはもう既に入れちゃっているだろうか，pipはcatkin-pkgだけ，とかいうルールだと後戻りできやすいけど，もしあれやこれや入っていって，面倒なこｔに成ると，面倒か．
--
◉ Kei Okada

knorth55, 2020-05-27:
@k-okada @708yamaguchi 
いえ、北川はまだ作業は着手していないので、山口くんがやってないのであれば元のままです。
debをダウンロードしてきて入れる方法ですか、確かにそれがよさそうですね。

>以下のfetch roboticsのマニュアルを見るに、現段階では18.04のfetchは独自のdebianサーバーを見るのではなくpackage.ros.orgを参照しているようです。
docs.fetchrobotics.com/indigo_to_melodic.html#manual-approach-18-04-install-and-installing-ros-fetch-packages
なので、18.04用のpackage.ros.orgにかかれているpython-catkin-pkgのバージョンを確認して、そのバージョンをpipでインストールするのが良いのかも知れません。

これも少し気になりますね。
indigoでも`packages.ros.org`でいいのであれば、そっちにスイッチしたい気もします。
（けどmelodicにアップデートしてしまいたいという気分もありますが面倒です）

708yamaguchi, 2020-05-27:
山口もまだ作業していません．

また今更ですが，`catkin_pkg`のソースを`jsk_fetch.rosinstall`経由でインストールするという手段もアリかなと思いました．

catkin_pkg
https://github.com/ros-infrastructure/catkin_pkg
jsk_fetch_.rosinstall
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_fetch_robot/jsk_fetch.rosinstall

708yamaguchi, 2020-05-27:
アリかなと書きましたが，catkin_pkgはrosのパッケージではないので，.rosinstallで管理する旨味がない気がしてきました．

knorth55, 2020-05-27:
@k-okada @708yamaguchi 

いま作業を行いました。
Fetch15の体内の`python-catkin-pkg`と`python-catkin-pkg-modules`を`0.4.13`にしました。
`python-catkin-pkg`: http://packages.ros.org/ros/ubuntu/pool/main/p/python-catkin-pkg/python-catkin-pkg_0.4.13-100_all.deb
`python-catkin-pkg-modules`: http://packages.ros.org/ros/ubuntu/pool/main/p/python-catkin-pkg-modules/python-catkin-pkg-modules_0.4.13-1_all.deb

```
fetch@fetch15:~$ sudo dpkg -l python-catkin-pkg*
[sudo] password for fetch: 
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                          Version                     Architecture                Description
+++-=============================================-===========================-===========================-===============================================================================================
ii  python-catkin-pkg                             0.4.13-100                  all                         catkin package library
ii  python-catkin-pkg-modules                     0.4.13-1                    all                         catkin package library
```

package.xml format=3のパッケージをビルドできることを確認しました。

knorth55, 2020-05-28:
これで様子をみようとおもいます
問題が起こったらissueを再度openしてください

closed 2020-05-28 by knorth55